openrazer/openrazer #2271: RazerGenie does not detect Ornata V3

RobQnadqs, 2024-06-29:
### Distribution

Linux Mint 21.3 Cinnamon

### Support

I have a Basilisk V3 mouse, and an Ornata V3 keyboard. RazerGenie does detect my mouse, but not my keyboard. I looked through the troubleshooting and ran `lsusb | grep 1532` in my terminal. This gave the following output:
`Bus 001 Device 007: ID 1532:028f Razer USA, Ltd Razer Ornata V3`
`Bus 001 Device 006: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3`

However, in this list (https://github.com/openrazer/openrazer/tree/stable#device-support), the Ornata V3 is listed to have ID 1532:02A1.

lah7, 2024-06-29:
Yep, same name but different PID. Fortunately, it looks like support is in the works:

* https://github.com/openrazer/openrazer/issues/1925
* https://github.com/openrazer/openrazer/pull/2233

You can help test it by [downloading that branch](https://github.com/tulser/openrazer/tree/ornata-v3-support), building the package with:

    ./scripts/build_debs.sh

and then install, reboot, etc. Might need to use `apt` to install a thing or two for building.

RobQnadqs, 2024-06-30:
Thanks for your reply! I didn't notice there were two versions of the Ornata V3

closed 2024-06-30 by RobQnadqs

RobQnadqs, 2024-08-24:
I'm trying to download the branch to test, but I'm not sure how to do it exactly. I'm quite new to Linux and not very experienced with this kind of stuff. Looking for some help...

lah7, 2024-08-24:
No problem. It [isn't](https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint) the most obvious but it's a few commands away in a Terminal.

For your convenience, I've already built the packages if you wanted to jump to step 4: [openrazer-3.7.0-ornata-v3-028f.zip](https://github.com/user-attachments/files/16737529/openrazer-3.7.0-ornata-v3-028f.zip)

1. Install some dependencies in order to build the packages:

       sudo apt install git build-essential debhelper dh-python python3-setuptools

2. Download the branch with the proposed device support (in your case, it's [#2233](https://github.com/openrazer/openrazer/pull/2233)):

       git clone https://github.com/tulser/openrazer.git -b ornata-v3-support
       cd openrazer

3. Build the package:

       ./scripts/build_debs.sh

4. Install the built packages:

       sudo apt install ./dist/*.deb

5. Reboot, and it should start working in RazerGenie.

 _If you like to keep things tidy, you could uninstall the packages from step 1 by reversing the command with `autoremove`._

I did notice this branch is based on v3.7.0, so if you try updating the system, the v3.8.0 'update' will lose these changes. For now, you could disable the `openrazer/stable` PPA in **Software Sources** program (as it's called in Linux Mint) until you see a future OpenRazer release adding support for the device.

RobQnadqs, 2024-08-24:
Thank you very much for your detailed explanation! It is working now. I'm slowly learning and getting used to Linux...

lah7, 2024-08-24:
Glad to hear! If you run into an issue, or find it works perfectly, please do leave feedback on the _pull request_ so this keyboard model can get merged sooner:

* https://github.com/openrazer/openrazer/pull/2233

RobQnadqs, 2024-08-24:
Will do!