openrazer/openrazer #2273: DKMS + Secure Boot documentation

z3ntu, 2024-07-03:
Apparently it's quite doable to use DKMS modules with Secure Boot, on Ubuntu at least.

Try to put this into our documentation:

> Official doc here: https://wiki.ubuntu.com/UEFI/SecureBoot
> 
> usually all you need to do if it does not work is this:
> `sudo update-secureboot-policy --enroll-key`
> 
> if the user has never created keys, first this:
> `sudo update-secureboot-policy --new-key`
> then this:
> `sudo update-secureboot-policy --enroll-key`
> upon reboot, user will be asked to enroll key, then enter the password they selected

Thanks @ilvipero!

z3ntu, 2024-08-31:
@zjDrummond Thanks for confirming!

lah7, 2024-08-31:
I created a new wiki page documenting the secure boot key enrolment process. Mainly targets Ubuntu, which is actually easier then I thought.

https://github.com/openrazer/openrazer/wiki/Secure-Boot

closed 2024-08-31 by lah7