openrazer/openrazer #2274: Failed to start Daemon to manage razer devices in userspace. A start job for unit UNIT has failed

Draconicrose, 2024-07-08:
### Distribution
Linux Mint 21.3 Cinnamon

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

While openrazor works (with Polychromatic), the following error message appears in my system logs:
```
Failed to start Daemon to manage razer devices in userspace.
A start job for unit UNIT has finished with a failure.

The job identifier is 33 and the job result is failed.
```

I've tried reinstalling the openrazer package, rebooting, (re)adding the user to plugdev, dkms install, and checking modprobe. To be clear, nothing seems to be failing but I would like confirmation and potentially a way to stop the message from appearing in my logs.

z3ntu, 2024-07-08:
What's the output of `systemctl --user status openrazer-daemon.service` ?

Draconicrose, 2024-07-08:
```
draconicrose@alexDesktop:~$ systemctl --user status openrazer-daemon.service
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-07-08 14:27:07 WEST; 7min ago
       Docs: man:openrazer-daemon(8)
   Main PID: 1393 (openrazer-daemo)
      Tasks: 4 (limit: 38093)
     Memory: 14.6M
        CPU: 148ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/openrazer-daemon.service
             └─1393 openrazer-daemon "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" "" ">

jul 08 14:27:07 alexDesktop systemd[1385]: Starting Daemon to manage razer devices in userspace...
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: Starting daemon.
jul 08 14:27:07 alexDesktop systemd[1385]: Started Daemon to manage razer devices in userspace.
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: 2024-07-08 14:27:07 | razer                          | INFO     | I>
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: 2024-07-08 14:27:07 | razer.screensaver              | INFO     | I>
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: 2024-07-08 14:27:07 | razer                          | INFO     | F>
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: 2024-07-08 14:27:07 | razer.device0                  | INFO     | I>
jul 08 14:27:07 alexDesktop openrazer-daemon[1393]: 2024-07-08 14:27:07 | razer                          | INFO     | S>

```

z3ntu, 2024-07-08:
It seems to be running fine there, even though you cut off the actual log messages on the right.

Did you do something weird to your dbus setup maybe? Can you run `qdbus --session org.razer` (assuming qdbus or some variant of qdbus is installed?)

Draconicrose, 2024-07-08:
Sorry, my bad. I updated my comment with the correct log messages. Everything seems to be fine. 

I installed qdbus (I don't think Mint comes with an equivalent) and this is the output:
```
draconicrose@alexDesktop:~$ qdbus --session org.razer
/
/org
/org/razer
/org/razer/device
/org/razer/device/PM2112H13000617
```

z3ntu, 2024-07-08:
Looks sane, the device is detected, so there's no problem from what I can tell.

So what's the problem exactly?

Draconicrose, 2024-07-08:
I wanted confirmation that it wasn't actually a problem. Now I'd like to know how to make that message not appear in my logs, if possible!

z3ntu, 2024-07-08:
You could check `journalctl --user-unit openrazer-daemon.service` and see if there's some errors in there but otherwise I can't really tell you much

Draconicrose, 2024-07-08:
Hm. That says -- No entries --. Is it supposed to?

z3ntu, 2024-07-08:
For me it has all the entries in there. I guess you could configure systemd not to put logs into the journal but I doubt you did this - or Linux Mint? But also since you showed that `systemctl --user status openrazer-daemon.service` has logs, they must be somewhere... But I guess if there are really none we can also ignore this.

So if you don't see any actual issues apart from some logs somewhere can we close this?

Draconicrose, 2024-07-08:
Yes, there's no actual issues. Thank you for your time!

closed 2024-07-08 by Draconicrose