openrazer/openrazer #2276: mouse not showing

Totallynotmwa, 2024-07-13:
### Distribution
Bazzite (installed on fedora 40 kinoite)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

So my mouse is not showing in open razer it's a razer cobra and is supported when I tried polychromatic it said razer daemon not detected I hit Troubleshoot it it said smth about dvdk or whatever it's called

z3ntu, 2024-07-13:
Could you run `sudo dkms install openrazer-driver/3.8.0` ?

Totallynotmwa, 2024-07-13:
> Could you run `sudo dkms install openrazer-driver/3.8.0` ?

[Dkms command not found](sudo: dkms: command not found)

z3ntu, 2024-07-14:
It seems you haven't installed openrazer then?

Totallynotmwa, 2024-07-14:
> It seems you haven't installed openrazer then?

Maybe it was installed incorrectly

closed 2024-07-15 by Totallynotmwa