openrazer/openrazer #2277: Light up keyboard when using touchpad

robinmayol, 2024-07-14:
### Distribution

Arch (btw)
Hyprland

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

On a Razer Blade 13" (Late 2020), I would like my keyboard to light up when moving the cursor. I have tried to implement this myself via python-openrazer but it doesn't seem that it provides a method to actually request the keyboard to light up.
I have also tried to simulate keypresses using ydotool (the wayland equivalent of xdotool) but nothing happens. I understand from this, that lighting up the keyboard happens deep in the firmware and maybe impossible to implement but I thought I'd ask :)

Thank you for your time working on open-source software.

z3ntu, 2024-08-31:
You mean light up like when the ripple effect is active and you press a key? Or with another effect like spectrum?

robinmayol, 2024-08-31:
Hey z3ntu!
My laptop doesn't have addressable leds so I only have access to the "static" effect.
I would just like the keyboard lighting to be triggered by activity on other input devices. For example, when I'm browsing the web, I'm mostly use my touchpad, I'd like the keyboard to be lit when I'm about to type something in the address bar (because I've clicked on it). It's also easier to swipe across the touchpad to light up the keyboard rather than find and hit say, the shift key in the dark just so I can see the keyboard.
Hope it makes sense! 

P.S. I also recently realised that I can't adjust how long the light stays on after inactivity. It's very short curently, like 1s.

z3ntu, 2024-08-31:
I'm not sure I understand correctly. When you have selected static effect for the keyboard, then the light only turns on when you touch the keyboard, and it turns off again automatically after a second?

robinmayol, 2024-08-31:
Yes, that's the behaviour. OK I actually counted about 5 seconds from the key press. The light fades in for about a second, stays on for about 3 and fades out over a second. This does not happen if I touch my touchpad. Should it???

z3ntu, 2024-08-31:
This behavior is definitely not from OpenRazer. The only thing where we get the keyboard input is for the ripple effect, but that's not applicable to your keyboard if there's only static backlight and no 'matrix' support.

Maybe check in the BIOS settings if you find something there?

robinmayol, 2024-08-31:
Oh I did :) So OpenRazer only sets the led effect, it doesn't handle actual triggers and timeouts? How do other Linux people set this kind of stuff on their Razer? I used to run Arch on a MacBook (I was still on Xorg back then), I'd made a script that just wrote to a file in /sys/ to set the led brightness value depending on the output of xprintidle.

z3ntu, 2024-08-31:
> So OpenRazer only sets the led effect, it doesn't handle actual triggers and timeouts?

Correct. On non-laptop devices this is just not a thing either.

> How do other Linux people set this kind of stuff on their Razer?

That I do not know, I don't own a Razer laptop. I'd guess maybe the Arch wiki or some forums may have some info around that?

> I used to run Arch on a MacBook (I was still on Xorg back then), I'd made a script that just wrote to a file in /sys/ to set the led brightness value depending on the output of xprintidle.

On my work laptop (Dell XPS) the backlight on/off is also handled by the BIOS/UEFI - I guess handled using ACPI or some other controller in the laptop. Not sure how that actually works there.

robinmayol, 2024-08-31:
I did read the wiki back when I installed Arch on this machine, I'm pretty sure I came across OpenRazer that way!

I just found out that [razer-laptop-control-no-dkms](https://github.com/Razer-Linux/razer-laptop-control-no-dkms/tree/main) does what I want. It comes with a cli which can directly control the brightness from the command line so I can script from that.

Razer Laptop Control is mentioned on the Arch Wiki but only in regards to setting power modes.

Thank you very much for taking the time.

closed 2024-08-31 by robinmayol

robinmayol, 2024-09-01:
I have just made an edit to the [Razer Blade](https://wiki.archlinux.org/title/Razer_Blade) page on the Arch wiki to explain the different behaviours of openrazer and razer-laptop-control. I just thought I'd mention it here in case someone more knowledgeable than me would like to clarify what I explain there.