openrazer/openrazer #2279: Lights on devices not turning off when screen turns off

NulliferBones, 2024-07-18:
### Distribution Arch linux

### Support

- [x ] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello,

My problem is as title states, the lights on my devices do not turn off when my screen turns off.
Devices are
Razer Naga Pro
Razer Blackwidow V3 mini hyperspeed

i checked the config and the values are set to true.

lah7, 2024-07-20:
Please could you share your log file? `~/.local/share/razer.log`

Another recent report (#2280) might suggest there's a race condition. Does it work if you run `openrazer-daemon -r` manually?

NulliferBones, 2024-07-20:
I'm using KDE, and I have lockscreen disabled (i tested and the lights do turn off when going on lockscreen)
however there is some bug with KDE lockscreen and multimonitors i'm unable to recover on lock.
So i was hoping it was possible with just screen turned off.

And yes I do need to manually start openrazer.
[razer.log](https://github.com/user-attachments/files/16321546/razer.log)

lah7, 2024-08-11:
Sorry for the delay. That feature in OpenRazer is designed to listen and react to the "screensaver" events. I guess this would be a feature request. However, I'm not sure if there are events for the monitor power state.

Ideally, the bug in KDE lock screen bug gets fixed, assuming its reported to [bugs.kde.org](https://bugs.kde.org).

If you manually turn off the monitor using its power button, a potential hardware workaround could be to plug the devices into your monitor's USB hub (if it has one) and have them powered off when the monitor is off.

closed 2025-03-16 by github-actions[bot]