openrazer/openrazer #2280: Race condition for battery notification script upon login

steveyh25, 2024-07-19:
### Distribution

Linux Mint 21.3

### Support

- [Yes] Have you tried rebooting?
- [Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Battery notification error in logs upon login:

Failed to init notification daemon, err: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files

Running "openrazer-daemon -r" after I have logged in then produces no error, and the notifications start working as normal, so it seems like a race condition with login

steveyh25, 2024-07-21:
https://wiki.archlinux.org/title/Desktop_notifications mentions that cinnamon and KDE (#2279) start their notification server up on user login

`notify-send 'test'` works fine for me when I'm logged in

Does the systemd service not need something like `Requires=org.freedesktop.Notifications`?

For now I have done:

`systemctl --user mask openrazer-daemon.service`

and made it start up on login with a 10s delay (~/.config/autostart/openrazer.desktop):

`[Desktop Entry]
Type=Application
Exec=openrazer-daemon
X-GNOME-Autostart-enabled=true
NoDisplay=false
Hidden=false
Name[en_GB]=openrazer daemon
Comment[en_GB]=No description
X-GNOME-Autostart-Delay=10`

z3ntu, 2024-08-31:
Hi! Could you please try #2296 if that solves the issue for you?

steveyh25, 2024-09-01:
@z3ntu It mostly works but there is still a race condition on the 1st invocation of show_notification that causes the script to crash (and hence still requires a daemon restart) unless I modify it with:

`    def show_notification(self, summary: str, message: str, icon: str) -> None:
        try:
            subprocess.check_call(["notify-send", "-a", "OpenRazer", "-i", icon, "-t", "4000", summary, message])
        except Exception as err:
            self._logger.warning("Failed to call show_notification, err: {0}".format(err))
`

The above amendment returns the error below on the first attempt to call show_notification, however, subsequent attempts work fine without a daemon restart because the exception is being caught:

2024-09-01 07:40:56 | razer.device0.batterynotifier  | WARNING  | Failed to call show_notification, err: Command '['notify-send', '-a', 'OpenRazer', '-i', 'battery-full', '-t', '4000', 'Razer Viper Ultimate (Wireless)', 'Battery is 74%']' returned non-zero exit status 1.

I appreciate that the error isn't very helpful - is there a way for me to get something more detailed for you? Thanks

z3ntu, 2024-09-01:
Great, thanks for testing!

I did notice that when testing and calling a non-existent executable instead of notify-send that the thread would crash and stop (the rest of the daemon would continue working), but I guess it makes sense to try every time again to send the notification, especially for your case it's still needed.

z3ntu, 2024-09-01:
I've added error handling now, see https://github.com/openrazer/openrazer/compare/d9ebae899b9c3c281f930a6cf25cf7b898ed4bc2..c93c3cfe3dbda2c120ce455a43ee4395098f65e8 for the diff between the previous version and this.

Could you please try the newest version?

steveyh25, 2024-09-01:
It's the same error, but now it's in "runtime" form:

2024-09-01 19:14:02 | razer.device0.batterynotifier  | WARNING  | Failed to show notification: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files

z3ntu, 2024-09-02:
Ok sounds good then!

commit d9ebae89 (2024-08-31): Replace notify2 dependency used in battery notifier

The website https://notify2.readthedocs.io/en/latest/ states "notify2 is
deprecated". Furthermore there seems to be a race condition during
startup if the notification dbus service is not yet available which can
happen during login when everything is starting up.

Replace the code, as suggested by notify2 docs by just simply calling
notify-send (from libnotify package) to achieve the same functionality.

Fixes #2280
commit c93c3cfe (2024-09-01): Replace notify2 dependency used in battery notifier

The website https://notify2.readthedocs.io/en/latest/ states "notify2 is
deprecated". Furthermore there seems to be a race condition during
startup if the notification dbus service is not yet available which can
happen during login when everything is starting up.

Replace the code, as suggested by notify2 docs by just simply calling
notify-send (from libnotify package) to achieve the same functionality.

Fixes #2280
PR #2296 (2024-09-03): Replace notify2 dependency used in battery notifier
The website https://notify2.readthedocs.io/en/latest/ states "notify2 is deprecated". Furthermore there seems to be a race condition during startup if the notification dbus service is not yet available which can happen during login when everything is starting up.

Replace the code, as suggested by notify2 docs by just simply calling notify-send (from libnotify package) to achieve the same functionality.

Fixes #2280
closed 2024-09-03 by z3ntu