openrazer/openrazer #2283: Snap Tap support for Razer Huntsman Pro (Mini) V3

darix, 2024-07-21:
Details:

https://www.youtube.com/watch?v=Feny5bs2JCg

Kyr4l, 2024-07-25:
is this feature possible to implement on "unsupported keyboards" via a software workaround ?

z3ntu, 2024-08-31:
If something there is enabled in the keyboard firmware via Synapse then we can probably add a toggle. If it's just a software concept, then it's out of scope for OpenRazer and a generic tool should be able to do the same.

If you can grab packet captures from Razer Synapse of doing this, I can take a look.

But also seems from the video that this is more a software thing of how the analog key values are interpreted - #1868

darix, 2024-08-31:
but it is interpreted by the firmware not the driver.

ByteDrummer, 2024-12-22:
Yeah, at least for the Huntsman V3 Pro line, it's built into the firmware and Synapse doesn't need to be running. They have since rolled out the feature for other keyboards, and it's probably done software side, since you need Synapse to be running.

z3ntu, 2025-01-02:
Then we really need USB packet captures of this to be able to implement it.

closed 2025-06-02 by github-actions[bot]