openrazer/openrazer #2284: depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razerkbd.ko.zst: Invalid argument

kaykhan, 2024-07-26:
This morning i went to upgrade using `sudo apt-get upgrade` and eventually getting depmod errors related to razer.

```                       
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```

I ran the following command 

```
sudo dpkg --configure -a   
```

I can see the following errors aftger running the above

> depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razeraccessory.ko.zst: Invalid argument
> depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razermouse.ko.zst: Invalid argument
> depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razerkraken.ko.zst: Invalid argument
> depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razerkbd.ko.zst: Invalid argument
> 

```
Setting up linux-image-6.8.0-39-generic (6.8.0-39.39) ...
depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razeraccessory.ko.zst: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razermouse.ko.zst: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razerkraken.ko.zst: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/6.8.0-39-generic/updates/dkms/razerkbd.ko.zst: Invalid argument
Processing triggers for linux-image-6.8.0-39-generic (6.8.0-39.39) ...
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 6.8.0-39-generic
 * dkms: autoinstall for kernel 6.8.0-39-generic
   ...done.
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.8.0-39-generic
/etc/kernel/postinst.d/zz-update-grub:
Sourcing file `/etc/default/grub'
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.0-39-generic
Found initrd image: /boot/initrd.img-6.8.0-39-generic
Found linux image: /boot/vmlinuz-6.8.0-38-generic
Found initrd image: /boot/initrd.img-6.8.0-38-generic
Found linux image: /boot/vmlinuz-6.8.0-36-generic
Found initrd image: /boot/initrd.img-6.8.0-36-generic
Found memtest86+ 64bit EFI image: /boot/memtest86+x64.efi
Warning: os-prober will be executed to detect other bootable partitions.
Its output will be used to detect bootable binaries on them and create new boot entries.
Found Windows Boot Manager on /dev/nvme0n1p2@/efi/Microsoft/Boot/bootmgfw.efi
Adding boot menu entry for UEFI Firmware Settings ...
done
```

**sudo dkms status** 

```
openrazer-driver/3.8.0, 6.8.0-36-generic, x86_64: installed
openrazer-driver/3.8.0, 6.8.0-38-generic, x86_64: installed
openrazer-driver/3.8.0, 6.8.0-39-generic, x86_64: installed
```

**lsb_release -a**                                  

```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
```

lah7, 2024-07-26:
`Invalid argument` and `dpkg was interrupted` suggests the modules wasn't rebuilt for the latest kernel correctly. Which one are you running anyway?

    uname -r

I'd try these commands to remove the old versions, ensure the system is fully up-to-date, and then rebuild the driver:
    
    sudo dkms remove openrazer-driver/3.8.0 --all
    sudo apt update
    sudo apt full-upgrade
    sudo dkms install openrazer-driver/3.8.0

Then try this to test it worked:

    sudo modprobe razerkbd

If there's no output from `modprobe` command, then it's ready to use after a reboot... unless secure boot will prevent it from loading (as seen in the troubleshooter, it appears to be on)

kaykhan, 2024-07-26:
> ```
> uname -r
> ```

```
6.8.0-39-generic
```

kaykhan, 2024-07-26:
yes secure boot is on, is there anything i need to do? not sure i want to turn that off

lah7, 2024-07-26:
Did the commands work? What's the output of `sudo modprobe razerkbd` now?

Was secure boot on before?  "Secure boot" just prevents unsigned drivers from loading. It's simpler to turn it off in the BIOS settings, but if left on, the modules can be signed - I don't know how to do this, but there should be instructions for Ubuntu.

z3ntu, 2024-07-27:
For secure boot on Ubuntu: #2273

kaykhan, 2024-07-29:
> sudo dkms remove openrazer-driver/3.8.0 --all
> sudo apt update
> sudo apt full-upgrade
> sudo dkms install openrazer-driver/3.8.0

I ran 
```

sudo dkms remove openrazer-driver/3.8.0 --all
sudo apt update
sudo dkms install openrazer-driver/3.8.0
```

`sudo modprobe razerkbd -> empty`

rebooted and reran

`sudo modprobe razerkbd -> empty.`

secure boot was enabled already but did not have to touch it to fix the issue. 

Seems like uninstalling and reinstalling works in this scenario.

closed 2024-07-29 by kaykhan

lah7, 2024-07-29:
Glad to hear! Thanks for noting that secure boot still says it fails in the troubleshooter, I'll get that improved.