openrazer/openrazer #2285: Support for huntsman v3 tkl

human664, 2024-08-04:
### Device Info

* Name: huntsman v3  pro tkl
* Product Number: RZ03-04980100-R3M1
* Model Number: rz03-0498

### Device Information
```

```
Bus 001 Device 004: ID 1532:02a7 Razer USA, Ltd Razer Huntsman V3 Pro TKL
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a7 Razer Huntsman V3 Pro TKL
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman V3 Pro TKL
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00a6
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 230 lines cut ...]
 00 C0 05 01 09 00 A1 01 85 0B 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 2F 81 00 C0

001:004:000:DESCRIPTOR         1722804504.272025
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
</details>

### Packet Captures
i have captures in correct file format however unsure how to upload.
captures of:
rapid trigger actuation toggling
snap tap toggling
static to breathing rgb settings

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
[image]
[image]
[image]

lah7, 2024-08-05:
> i have captures in correct file format however unsure how to upload.

Thanks. You should be able to put them in a ZIP and drag-and-drop them when editing the issue.

human664, 2024-08-05:
> > i have captures in correct file format however unsure how to upload.
> 
> Thanks. You should be able to put them in a ZIP and drag-and-drop them when editing the issue.

can do

z3ntu, 2024-08-31:
Hi! A few things I see from the packet captures.

This keyboard uses `MATRIX_DIMS = [6, 22]`.
The transaction_id in driver code should be 0x1f 

Probably you can take some existing Huntsman support and copy paste most things for your keyboard. See also https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md and view recent commits in razerkbd driver for inspiration.

z3ntu, 2025-02-15:
@TheDarkUndoing Somebody with the device needs to do the work to add support for this device to OpenRazer.

closed 2025-08-31 by z3ntu

Med0-n, 2025-10-22:
so it's working ?

PR #2462 (2025-08-31): Add support for Razer Huntsman V3 Pro TKL
Copied from Add support for Razer Huntsman V3 Pro #2365

Did test on lights and gamemode.

resolve #2285