openrazer/openrazer #2286: My razer viper mini is not recognized by OpenRazer, but Fedora is.

Warillon, 2024-08-10:
### Affected Device / PID

Bus 001 Device 005: ID 1532:008a Razer USA, Ltd RZ01-0325, Gaming Mouse [Viper Mini]

### Operating System / Version

Fedora 40

### Kernel Version

6.10.3-200.fc40.x86_64

### OpenRazer Version

latest version

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

[image]

`warillon@localhost-live:~$ lsusb | grep '1532:'
Bus 001 Device 005: ID 1532:008a Razer USA, Ltd RZ01-0325, Gaming Mouse [Viper Mini]`

I did everything I needed to do, including adding to the plugdev, rebooting, unplugging and reconnecting the mouse, etc.

lah7, 2024-08-11:
Which version of OpenRazer do you have installed? This will also give a clue about the driver status.

    openrazer-daemon --version
    dkms status | grep openrazer

What does your daemon log say?

    cat ~/.local/share/openrazer/logs/razer.log

Warillon, 2024-08-11:
```
warillon@desktop-f89dv3d:~$ openrazer-daemon --version
dkms status | grep openrazer
openrazer-daemon 3.8.0
```
And

```
warillon@desktop-f89dv3d:~$ cat ~/.local/share/openrazer/logs/razer.log
2024-08-10 09:56:23 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:56:23 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:56:23 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:56:23 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:56:30 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:56:30 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:56:58 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:56:58 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:56:59 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:56:59 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:58:17 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-08-10 09:58:17 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-08-10 09:59:30 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 3267
2024-08-10 09:59:30 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-10 09:59:30 | razer                          | INFO     | Serving DBus
2024-08-10 10:15:07 | razer                          | INFO     | Stopping daemon on signal 1
2024-08-10 10:15:08 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 6330
2024-08-10 10:15:08 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWired ----------------- (1532:00AA)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWireless -------------- (1532:00AB)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3XHyperSpeed -------------- (1532:00B9)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2024-08-10 10:15:08 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
[... cut ...]
2024-08-10 10:15:57 | razer                          | DEBUG    | Writing persistence config
2024-08-10 10:16:53 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 3293
2024-08-10 10:16:53 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-10 10:16:53 | razer                          | INFO     | Serving DBus

```

After I saw "User is not a member of the plugdev group |Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot !", I did it but nothing changed.

lah7, 2024-08-11:
Daemon doesn't see the devices, let's check the driver side is installed correctly. Please run this command to check this:

    sudo dkms status

If it's not installed or it's installed for a different kernel version, this command should fix it:

    sudo dkms remove openrazer-driver/3.8.0 --all
    sudo dkms install openrazer-driver/3.8.0

(Then reboot)

Otherwise, if it is installed, what is output when running this?

    sudo modprobe razermouse

Warillon, 2024-08-13:
Perfect it's work, I wanted to install the drivers because I'd seen they were missing but I couldn't find the command for fedora 40 on the Internet.