openrazer/openrazer #2289: 'This supported device wasn't detected by the OpenRazer daemon (v3.5.1)' on polychromatic

airbutbackwards, 2024-08-16:
### Affected Device / PID

1532:02A2

### Operating System / Version

Debian 12

### Kernel Version

6.9.11-1-liquorix-amd64

### OpenRazer Version

3.5.1

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

the daemon is not detecting my razer ornata v3 x, simple as that

lsmod:
`razeraccessory        118784  0
razerkraken            28672  0
razermouse            176128  0
razerkbd              131072  0
`
dmesg:
`[  511.274981] razerkbd: loading out-of-tree module taints kernel.
`
ls -al:
`ls: cannot access '/sys/bus/hid/drivers/razer*/*1532*/': No such file or directory
`
all other commands give something entirely blank

the troubleshoot thing in polychromatic shows everything as good, except for:
Unknown- Kernel module can be probed
and
Failed- OpenRazer is the latest version

i tried re-installing openrazer with the instructions given through apt, but it never gives me the latest version (yes, even after apt / apt-get update)

lah7, 2024-08-16:
The version of OpenRazer in [Debian's official repositories](https://packages.debian.org/search?keywords=openrazer) can be a few versions behind. Support for your device [was added in v3.7.0](https://github.com/openrazer/openrazer/releases/tag/v3.7.0). To make it work, please add OpenRazer's Debian repository; update the system and then reboot:

https://openrazer.github.io/#debian

airbutbackwards, 2024-08-16:
hm, i thought i installed it like that to begin with... must've made a mistake somewhere.
this did work though!!!

airbutbackwards, 2024-08-16:
oops forgot to close

closed 2024-08-16 by airbutbackwards