openrazer/openrazer #2290: Unable to Retrieve Battery Level (Always Returns 0%) for Razer Basilisk Ultimate Mouse

Riebeck-one, 2024-08-22:
### Affected Device / PID

Razer Basilisk Ultimate / 1532:0088

### Operating System / Version

Arch Linux

### Kernel Version

6.10.6-2-cachyos

### OpenRazer Version

3.8.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [X] My device is missing a feature.

### Please describe the issue

I can't get the battery level of my Razer Basilisk Ultimate mouse. Running `cat /sys/bus/hid/drivers/razermouse/0003:1532:0088.0009/charge_status` always shows `0`, no matter what I do. I tried restarting the daemon, rebooting the PC, and plugging and unplugging the mouse plug.

Logs :

```
sudo journalctl -b | grep -i razer_mount
août 22 17:59:12 archlinux razer_mount[665]: Driver razermouse
août 22 17:59:12 archlinux razer_mount[666]: Device_ID 5-4.1
août 22 17:59:12 archlinux razer_mount[667]: Modprobing razermouse
août 22 17:59:13 archlinux razer_mount[704]: Modprobed razermouse
août 22 17:59:13 archlinux razer_mount[708]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[709]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[710]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[711]: Device_ID 5-4.1:1.0
août 22 17:59:13 archlinux razer_mount[712]: Device_ID 5-4.1:1.1
août 22 17:59:13 archlinux razer_mount[713]: Device_ID 5-4.1:1.2
août 22 17:59:13 archlinux razer_mount[717]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[718]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[719]: Device_ID 0003:1532:0088.0006
août 22 17:59:13 archlinux razer_mount[720]: Driver razermouse
août 22 17:59:13 archlinux razer_mount[721]: Device_ID 0003:1532:0088.0008
août 22 17:59:13 archlinux razer_mount[722]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0006/
août 22 17:59:13 archlinux razer_mount[723]: Device_ID 0003:1532:0088.0007
août 22 17:59:13 archlinux razer_mount[724]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0008/
août 22 17:59:13 archlinux razer_mount[725]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0007/
août 22 17:59:13 archlinux razer_mount[729]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0008/
août 22 17:59:13 archlinux razer_mount[730]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0007/
août 22 17:59:13 archlinux razer_mount[731]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0006/
août 22 18:04:11 archlinux razer_mount[8267]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8268]: Device_ID 5-4.1
août 22 18:04:11 archlinux razer_mount[8274]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8276]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8275]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8277]: Device_ID 5-4.1:1.0
août 22 18:04:11 archlinux razer_mount[8278]: Device_ID 5-4.1:1.1
août 22 18:04:11 archlinux razer_mount[8279]: Device_ID 5-4.1:1.2
août 22 18:04:11 archlinux razer_mount[8283]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8284]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8285]: Driver razermouse
août 22 18:04:11 archlinux razer_mount[8286]: Device_ID 0003:1532:0088.0009
août 22 18:04:11 archlinux razer_mount[8287]: Device_ID 0003:1532:0088.000A
août 22 18:04:11 archlinux razer_mount[8288]: Device_ID 0003:1532:0088.000B
août 22 18:04:11 archlinux razer_mount[8289]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0009/
août 22 18:04:11 archlinux razer_mount[8290]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.000A/
août 22 18:04:11 archlinux razer_mount[8291]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0088.000B/
août 22 18:04:11 archlinux razer_mount[8295]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.000A/
août 22 18:04:11 archlinux razer_mount[8296]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.0009/
août 22 18:04:11 archlinux razer_mount[8297]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0088.000B/
```
```
cat ~/.local/share/openrazer/logs/razer.log
2024-08-21 11:40:12 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 13260
2024-08-21 11:40:12 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-21 11:40:12 | razer                          | INFO     | Found device.0: 0003:1532:0088.0002
2024-08-21 11:40:12 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-21 11:40:13 | razer                          | INFO     | Serving DBus
2024-08-21 11:56:47 | razer                          | INFO     | Stopping daemon on signal 15
2024-08-21 11:56:48 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 32830
2024-08-21 11:56:48 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-21 11:56:48 | razer                          | INFO     | Found device.0: 0003:1532:0088.0002
2024-08-21 11:56:48 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-21 11:56:49 | razer                          | INFO     | Serving DBus
2024-08-22 17:40:59 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 1166
2024-08-22 17:40:59 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-22 17:40:59 | razer                          | INFO     | Found device.0: 0003:1532:0088.0006
2024-08-22 17:40:59 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-22 17:41:00 | razer                          | INFO     | Serving DBus
2024-08-22 17:58:50 | razer                          | INFO     | Stopping daemon on signal 15
2024-08-22 17:59:20 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 1212
2024-08-22 17:59:20 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-22 17:59:20 | razer                          | INFO     | Found device.0: 0003:1532:0088.0006
2024-08-22 17:59:20 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-22 17:59:21 | razer                          | INFO     | Serving DBus
2024-08-22 18:04:01 | razer                          | WARNING  | Removing 0003:1532:0088.0006
2024-08-22 18:04:13 | razer                          | INFO     | Found valid device.0: 0003:1532:0088.0009
2024-08-22 18:04:13 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-22 18:05:10 | razer                          | INFO     | Stopping daemon on signal 15
2024-08-22 18:05:10 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 9206
2024-08-22 18:05:10 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-08-22 18:05:10 | razer                          | INFO     | Found device.0: 0003:1532:0088.0009
2024-08-22 18:05:10 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskUltimateReceiver
2024-08-22 18:05:11 | razer                          | INFO     | Serving DBus
```

lah7, 2024-08-23:
> Running cat /sys/bus/hid/drivers/razermouse/0003:1532:0088.0009/charge_status always shows 0

Did you mean to read [`charge_level`](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#charge_level---gets-the-battery-level-from-the-mouse-read-only)? [`charge_status`](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#charge_status---gets-the-charging-status-read-only) is whether the device says it's charging (0 = no, 1 = yes). Not sure if this only applies to the device PID (`0086`, the non receiver PID)

It could be similar to #2122 if you see `Command timed out` in `dmesg` when trying to get the charge level.

Riebeck-one, 2024-08-23:
Yeah, I meant I read the charge level.
I think I found the culprit: it's OpenRGB. I use OpenRGB to control every LED on my PC, including the mouse. Disabling OpenRGB instantly resolves the issue, and enabling it again instantly recreates the problem.

Maybe it's worth mentioning this incompatibility somewhere?

closed 2025-01-22 by github-actions[bot]

lah7, 2025-06-30:
If OpenRGB is constantly sending frames to the device, I wonder if there's a lot of `razermouse: Command timed out` in the kernel log. It would appear here:

    sudo dmesg -w

If so, it would be related to #2122 where requests like "what's your serial, battery status?" are not being fulfilled because the firmware is busy / ignoring them.