openrazer/openrazer #2297: Support for Razer Naga V2 Pro

omarluq, 2024-09-03:
### Device Name

Naga V2 Pro

### Product Number

RZ01-04400100

### Model Number

RZ01-0440

### VID:PID

1532:00a7

### USB Details

```shell
❯ lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:00a7 Razer USA, Ltd Razer Naga V2 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a7 Razer Naga V2 Pro
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga V2 Pro
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     100
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
lrwxrwxrwx 1 root root  10 Sep  2 20:50 usb-Razer_Razer_Naga_V2_Pro-event-if01 -> ../event22
lrwxrwxrwx 1 root root  10 Sep  2 20:50 usb-Razer_Razer_Naga_V2_Pro-event-mouse -> ../event19
lrwxrwxrwx 1 root root  10 Sep  2 20:50 usb-Razer_Razer_Naga_V2_Pro-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Sep  2 20:50 usb-Razer_Razer_Naga_V2_Pro-if02-event-kbd -> ../event20
lrwxrwxrwx 1 root root   9 Sep  2 20:50 usb-Razer_Razer_Naga_V2_Pro-mouse -> ../mouse2
```

### USB HID Dump

```shell
001:003:001:DESCRIPTOR         1725328763.205891
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:003:002:DESCRIPTOR         1725328763.208569
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:003:000:DESCRIPTOR         1725328763.212571
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 75 01 95 08 81 03 05 0C 0A 38 02 15 81 25 7F 75
 08 95 01 81 06 05 01 09 38 15 81 25 7F 75 08 95
 01 81 06 09 30 09 31 16 00 80 26 FF 7F 75 10 95
 02 81 06 C0 06 00 FF 09 02 15 00 25 01 75 08 95
 5A B1 01 C0
```

z3ntu, 2024-09-03:
Duplicate of #1952

But to confirm, I assume lsusb shows `1532:00a7` when connected wired, and `1532:00a8` when connected wireless?

closed 2024-09-03 by z3ntu

omarluq, 2024-09-08:
> But to confirm, I assume lsusb shows `1532:00a7` when connected wired, and `1532:00a8` when connected wireless?

Did not see the existing issue, my bad. And yes, you are right, the bus ID is 00a8 when connected wireless and 00a7 when wired.

z3ntu, 2024-09-09:
Thanks for confirming!