openrazer/openrazer #2298: Support for Razer Kraken V4

FuzzyExpress, 2024-09-03:
### Device Name

Kraken V4

### Product Number

RZ04-05170100-R3U1

### Model Number

 RZ04-0517

### VID:PID

1532:056c

### USB Details

```shell
Bus 003 Device 009: ID 1532:056c Razer USA, Ltd Razer Kraken V4 2.4
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x056c Razer Kraken V4 2.4
  bcdDevice            1.00
  iManufacturer           1 MediaTek Inc
  iProduct                2 Razer Kraken V4 2.4
  iSerial                 3 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e1
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              4 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004e
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
[... 254 lines cut ...]
 09 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02 09
 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54 09
 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54 09
 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54 09
 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54 15
 00 26 FF 00 75 08 95 20 B1 02 C0 85 35 09 60 A1
 02 09 55 15 00 25 08 75 08 95 01 B1 02 09 61 09
 62 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09 51
 09 52 09 53 09 54 15 00 26 FF 00 75 08 95 04 B1
 02 C0 85 36 09 70 A1 02 09 71 15 00 25 01 75 08
 95 01 B1 02 C0 C0
```

### Packet Captures

I found and installed The New Razer Synapse, it updated my headset, and then after a VM reboot could no longer find it. 💀  

Old: Razer Synapse does not recognize the headset in my Windows 10 VM. It should be working, 2 of my 3 SteelSeries devises work. (MSI Raider Light Bar & Aerox 5 WL, but not the laptop keyboard) 
Windows can see the device, but only the Chat output. Linux does not see it at all with the VM open, but could see the Game output before I fired up the VM. (Might be a PipeWire config thing, like the Nari had issues with)

I'll upload these once I have more time.

### Synapse Screenshots

Same as above.

z3ntu, 2024-09-03:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

virnik0, 2024-12-27:
I do face the same experience with this headset. Windows 10 and Windows 11 does see the headset and audio also works. But Device Manager lists an exclamation mark to "Razer Kraken V4 - Game" driver, which have "I/O error, driver couldn't start" in the event log. Looks like Razer issue, Razer Synapse V3 and V4 behaves the same in this regard - doesn't see the headset. You can still enable Spatial Audio without Razer Synapse, but then you can't set RGB or any other feature. In Linux, it works fine, but Polychromatic depends on openrazer, which as of now doesn't recognize the device.

I am using Linux as a primary OS. But I do have beefed-up KVM available with both Win10 and Win11 in it. If there's a way to capture packets you need to be able to look at possible support of the headset, please share - and I'd try to follow it and provide the packet capture.

virnik0, 2025-01-20:
OK, got it working. Razer released new firmware for this headset, so it started to work in Windows as well. I didn't changed anything on my side other than downloading separate firmware installer from razer site (version 1.2.3).  Both Razer Synapse V3 and V4 works fine now. The problem is, that the Synapse doesn't store any color to on-board memory. It uploads it to cloud, and restores it from cloud (or cache, if previously set and networking is off). OpenRGB can replace this, and information on what to trigger on the USB device  should be in the packet capture.

**As for packet capture of USB traffic, here it is:**
https://drive.google.com/file/d/1436Ha-qDRHYNjMt1_a1rtgwJd_2Yg-P-/view?usp=drive_link

**What I did: Turned the headset, and then wait for Windows 11 to finish booting.**
a) Set the RGB color to Purple
b) Set the RGB color to Blue
c) Import color profile (Blue) and apply
d) exit Razer Synapse/Chroma apps (lights reset to defaults)
e) started Razer Synapse (lights restored to Blue color)
d) disconnected USB buss from the VM (the lights stays on and with correct color)
e) turned off the headset
f) turned it on (RGB lights flash, but then turns off)

virnik0, 2025-02-11:
Bump! How else can I help with resolution? I can't code mysel, not in required quality and capacity. pcap provided in above post.

virnik0, 2025-05-18:
Bump! Like - really, Razer Cobra Wireless Pro is well fine, while Kraken V4 and Kraken V4 Pro are ignored? I get it that the load could be overwhelming, but at least "got all required info, we'll get to that once ready" or "you have to provide this and that" would do. But ignorance? Why?

z3ntu, 2025-07-06:
@virnik0 This is an open source project, with nobody paying anyone to work on it.

Device support in OpenRazer generally works that somebody who owns the device works on adding support for it. Afterwards the maintainer (me) can review the changes and then finally integrate it into the main code base.

virnik0, 2025-07-06:
> [@virnik0](https://github.com/virnik0) This is an open source project, with nobody paying anyone to work on it.
>
> Device support in OpenRazer generally works that somebody who owns the device works [...]

Beg me a pardon, I know how OSS works. But nobody could be a dev. I am very good platform engineer myself, and yet - I do not demand others to debug the kernel, or customize distro for specific needs, as I know how to do it, so I do it myself. With that approach, I provided data/details devs requested, to help finding out the solution. I do not demand it! I politely asked, and provided anything viable which could help. But please do not ask me to code - I can passively update/edit the code, I do understand what is happening in it (eg: I am mostly capable to read it), but I am not good enough to author new code.

z3ntu, 2025-07-06:
Whether you mean it or not, the way you phrased especially the comment https://github.com/openrazer/openrazer/issues/2298#issuecomment-2889285942 is not "I politely asked", at least it's not how I read it.

That being said, hopefully somebody will come along soon who can add support for this device.

CalcProgrammer1, 2025-10-29:
I did some Wireshark sleuthing to get the Kraken V4 working in OpenRGB and decoded some of the non-RGB packets as well.  This was done with the headset plugged in with a USB cable in wired mode, the wireless dongle may be different (my test code that works in OpenRGB works in wired mode but not with the wireless dongle).  I also just updated my headset and dongle to the latest 1.5.0 firmware.

Here is the OpenRGB merge request:
https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/3042

Here is what I've learned:

```
Razer Kraken V4

Main packet format

SZ: Size
CM: Command
CS: Checksum
DD: Data

Total size: 64 bytes

02 00 60 00 00 00 SZ 00 00 CM 00 DD ... DD CS 00

Commands

00: Get Serial Number
01: ???
02: Get Firmware Info
21: ???
2A: ???

93: Set Active Audio Equalizer Preset
    Data: 5 bytes
    01 MM 00 00 00

    MM: Preset Mode
    07: Game
    08: Music
    09: Movie
    FF: Custom

95: Edit Active Audio Equalizer Preset
    Data: 14 bytes
    0A VV VV VV VV VV VV VV VV VV VV 00 00 00

    VV: Equalizer value -12 to +12, signed 8 bit

    There are 10 values, corresponding to 31Hz, 63Hz, 125Hz, 250Hz, 500Hz, 1kHz, 2kHz, 4kHz, 8kHz, and 16kHz respectively

96: Set Active Microphone Equalizer Preset
    Data: 5 bytes
    01 MM 00 00 00

    MM: Preset Mode
    01: Mic Boost
    02: Broadcast
    03: Conference

98: Set Mic Monitoring (Sidetone) Enable
    Data: 5 bytes
    01 EN 00 00 00

    EN: Sidetone Enable Flag

99: Set Mic Monitoring (Sidetone) Level
    Data: 5 bytes
    01 LV 00 00 00

    LV: Sidetone Level value 0 to 15

AC: Set Wireless Power Saving Timeout
    Data: 5 bytes
    01 TO 00 00 00

    TO: Power Saving Timeout value, 0 for off or between 15 to 60 minutes

C0: Set RGB Effect
    Data: 5 bytes
    01 MM 00 00 00

    MM: Effect Mode value?

    There only appears to be one mode, value 04, which is the default rainbow wave

    All other modes are done through Synapse's effects engine and direct mode

    When Synapse closes, it goes back to rainbow wave at the given brightness

C1: Set Brightness (not including Direct mode)
    Data: 5 bytes
    01 BR 00 00 00

    BR: Brightness value 0 (off) to 255

E8: ???

RGB packet format - seems to be more like Razer's conventional matrix RGB packet

SZ: Size
DD: Data

There is no checksum

Total size: 64 bytes

02 00 06 00 00 00 SZ 0F 03 00 00 00 DD ... DD

Data: 33 bytes

00 00 08 RR GG BB RR GG BB RR GG BB RR GG BB RR GG BB RR GG BB RR GG BB RR GG BB RR GG BB 00 00 00

RR/GG/BB: RGB color, for 9 LEDs total

When sent this command appears to put the device in direct mode if it isn't already

Stays in direct mode until 0xC0 (Set RGB Effect) is sent, which Synapse sends when closed to put it back to default rainbow wave
```

CalcProgrammer1, 2025-11-01:
Looks like if you do fill in the CRC on the matrix custom frame RGB packet it still accepts it.  I've updated my OpenRGB merge request with a packet structure that lines up with the standard Razer packet and also added Kraken Kitty V3 Pro which appears to use this same protocol at least for RGB control (except for an additional LED for the ears).