openrazer/openrazer #2299: Razer BlackWidow V4 75% Fn+KEY combinations not working

Flexksx, 2024-09-06:
### Affected Device / PID

Razer BlackWidow V4 75%, 1532:02A5

### Operating System / Version

Arch Linux x86_64

### Kernel Version

6.10.8-arch1-1

### OpenRazer Version

3.8.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [X] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [X] It worked before a system/kernel update.
- [X] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [X] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [X] My device is missing a feature.

### Please describe the issue

After latest patch that added support for BlackWidow V4 75%, the Fn+KEY combinations stopped working.
Specifically:
Fn+P -> PrtScr
Fn+del -> Hybernate
Fn+PgUp -> Home
Fn+PgDn -> End
Fn+Ins -> Pause

The function key row is working as normal.
Fn+ F11 is decreasing brightness, Fn+F12 is increasing it
Fn+F10 turns game mode on
Fn+F9 I can't understand how do I create a macro.

Flexksx, 2024-09-06:
@Oery opened the issue

Oery, 2024-09-06:
Okay so it seems the mapping for those keys must be manually added in ```driver/razerkbd_driver.c```. I'm going to do that.

My only issue is that Fn+Del reboot my PC so I can't really check what key it's supposed to be. If you can check what key is produced using something like ```evtest``` it would be super useful.

PR #2300 (2024-09-09): Fix Razer BlackWidow V4 75% Fn+KEY combinations not working
Fixes #2299 

When adding support by observing other commits, I missed Fn keys and used the same translation map as other keyboards.

This pull request fixes that by adding another key translation map, dedicated to the BlackWidow V4 75%.

Everything should work but there are still some cases I haven't been able to test, Brightness+/- works regardless of loaded driver. Macros are too confusing to test. Pause doesn't show up as anything in evtest, so not sure about this one.
closed 2024-09-09 by z3ntu