openrazer/openrazer #2302: Support for Razer Hammerhead Hyperspeed

evll, 2024-09-12:
### Device Name

Razer Hammerhead Hyperspeed

### Product Number

RZ12-03820200

### Model Number

RZ12-R3G1

### VID:PID

1532:0a3a

### USB Details

```shell
Bus 001 Device 011: ID 1532:0a3a Razer USA, Ltd Razer Hammerhead HyperSpeed 2.4
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a3a Razer Hammerhead HyperSpeed 2.4
  bcdDevice            2.00
  iManufacturer           0 
  iProduct                2 Razer Hammerhead HyperSpeed 2.4
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e2
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0048
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
[... 199 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
usb-1532_Razer_Hammerhead_HyperSpeed_2.4-event-if03 -> ../event15
```

z3ntu, 2024-09-24:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

farion, 2024-10-14:
Maybe I have a similar device (I added my device already to this issue: https://github.com/openrazer/openrazer/issues/1569#issuecomment-2387900836)

z3ntu, 2024-10-20:
@farion -> https://github.com/openrazer/openrazer/issues/1569