openrazer/openrazer #2307: Support for Razer [name]

Tyler-Posella, 2024-09-19:
### Device Name

Razer Viper V3 Pro

### Product Number

RZ01-05120100

### Model Number

RZ01-0512

### VID:PID

1532:00c0

### USB Details

```shell
Bus 001 Device 009: ID 1532:00c0 Razer USA, Ltd Razer Viper V3 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00c0 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                2 Razer Viper V3 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
drwxr-xr-x 2 root root 400 Sep 19 12:45 .
drwxr-xr-x 4 root root 700 Sep 19 12:45 ..
lrwxrwxrwx 1 root root  10 Sep 19 12:45 usb-1532_Razer_Viper_V3_Pro_000000000000-event-if01 -> ../event17
lrwxrwxrwx 1 root root  10 Sep 19 12:45 usb-1532_Razer_Viper_V3_Pro_000000000000-event-mouse -> ../event15
lrwxrwxrwx 1 root root  10 Sep 19 12:45 usb-1532_Razer_Viper_V3_Pro_000000000000-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Sep 19 12:45 usb-1532_Razer_Viper_V3_Pro_000000000000-if02-event-kbd -> ../event18
lrwxrwxrwx 1 root root   9 Sep 19 12:45 usb-1532_Razer_Viper_V3_Pro_000000000000-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Sep 19 12:39 usb-Blue_Microphones_Yeti_Stereo_Microphone_REV8-event-if03 -> ../event14
lrwxrwxrwx 1 root root  10 Sep 19 12:39 usb-KEYBOARD_USB_KEYBOARD-event-if02 -> ../event10
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-KEYBOARD_USB_KEYBOARD-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Sep 19 12:39 usb-KEYBOARD_USB_KEYBOARD-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Sep 19 12:39 usb-KEYBOARD_USB_KEYBOARD-if02-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-KEYBOARD_USB_KEYBOARD-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-Logitech_USB_Receiver-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-Logitech_USB_Receiver-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-Logitech_USB_Receiver-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-UBEST_zoom75_wireless_061DB7FA3C18-event-if02 -> ../event4
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-UBEST_zoom75_wireless_061DB7FA3C18-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-UBEST_zoom75_wireless_061DB7FA3C18-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Sep 19 12:39 usb-UBEST_zoom75_wireless_061DB7FA3C18-if02-mouse -> ../mouse0
```

### USB HID Dump

```shell
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/009, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
Failed to find and open the devices: Access denied (insufficient permissions)
```

Tyler-Posella, 2024-09-19:
Device ID 1532:00c0 doesn't appear in the supported devices even though there is a completed issue with this device in the past.

lah7, 2024-09-19:
Support was added in #2266 for `1532:00C0` (wired) and `1532:00C1` (wireless), but it's not in any stable release yet.

Depending on your distro, you can install the equivalent `-git` packages (for Arch), use the daily PPA (for Ubuntu) or [build the package](https://github.com/openrazer/openrazer/wiki/Building-a-package) for others using the `master` branch.

Tyler-Posella, 2024-09-19:
> Support was added in #2266 for `1532:00C0` (wired) and `1532:00C1` (wireless), but it's not in any stable release yet.
> 
> Depending on your distro, you can install the equivalent `-git` packages (for Arch), use the daily PPA (for Ubuntu) or [build the package](https://github.com/openrazer/openrazer/wiki/Building-a-package) for others using the `master` branch in the meantime.

I've been trying all morning but I keep getting device not detected

lah7, 2024-09-19:
What distro are you using? Which frontend are you using (Polychromatic, RazerGenie)?

Tyler-Posella, 2024-09-19:
> What distro and frontend are you using (Polychromatic, RazerGenie)?
> 
> If the device isn't listed, sounds like not all of the `master` equivalent packages are installed (e.g. `openrazer-daemon-git`, `python-openrazer-git`, `openrazer-driver-dkms-git`). There is the usual [troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting) in case it's something as simple as a missing `plugdev` user group.

I've already gone through the troubleshooting and everything ran properly. For somereason though none of the "supported apps" are showing the device as supported still but it detects the product code properly

Tyler-Posella, 2024-09-19:
I am currently running on PopOS version 22.04

lah7, 2024-09-19:
Since it's based on Ubuntu, add the ["daily" PPA](https://launchpad.net/~openrazer/+archive/ubuntu/daily) and update the system:

```
sudo add-apt-repository ppa:openrazer/daily
sudo apt update
sudo apt dist-upgrade
```

Then reboot. The device should be recognised in both apps afterwards.

Once your device is in a stable release (next one: v3.9.0), then you can remove the daily PPA, or remove it immediately now if you want.

lah7, 2024-10-29:
@jtsec92 That's strange. Presumably you've rebooted? What's in the daemon's log file: `~/.local/share/openrazer/logs/razer.log`?

(Try `sudo modprobe razermouse` instead)

lah7, 2024-10-29:
Daemon doesn't see the device, driver appears to be OK. `1532:00C1` is definitely supported in 3.9.0. What are the permissions for the mouse in the sysfs?

    ls -al /sys/bus/hid/drivers/razer*/*1532*/

Did you add your user to `plugdev` group? (The daemon typically refuses to start if it wasn't)

    sudo gpasswd -a $USER plugdev

Also for good measure, what does the kernel say?

```
dmesg | grep razer
```
```
sudo journalctl -b | grep -i razer_mount
```

lah7, 2024-10-29:
I'm not sure what's happening here. There's typically three directories: yours is missing `0003:1532:00C1.000D`, and one of the three should have files starting with `matrix_`, `device_`, etc, which are missing. Driver appears to look like it's installed and loaded though.

* What distro/version are you using?
* Does it work if you unplug it, wait, then plug it in?  _(In case device is in a weird state)_

lah7, 2024-10-29:
`00C1` is the wireless one - let's try device PID `00C0`.

* Does it work when the mouse is wired directly?
* Does having the wireless dongle plugged in at the same time make any difference?  
* _If necessary, press CTRL+F5 to force refresh Polychromatic, or restart the daemon._

Puzzling one, because it looks right, except the driver obviously isn't creating device files, but there's no error. 1bbbe473382a8f9bda1e7a33a85be575a343c0e7 looks sane and other users reported it working (#2266). Unless there is an odd chance that it isn't actually v3.9.0 (e.g. mismatching files). You could check that:

    cat /usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py | grep 'class RazerViperV3Pro'
    cat /usr/src/openrazer-driver-3.9.0/driver/razermouse_driver.c | grep RAZER_VIPER_V3_PRO

It should output lines showing this device.

Did you always use the stable PPA, or did you try daily PPA, building packages, or manual installation, etc? 

Lastly, _because of a recent nuisance with **secure boot**_ - is this turned on/off in your BIOS/UEFI?

lah7, 2024-10-30:
Good to know. Thanks for going through the troubleshooting. It was head scratching since everything seems in place. Polychromatic saying it's "not supported" will be addressed in its next update. The PPA SSL shouldn't matter since you have the repository now, and presumably it's the release version of 3.9.0:

    apt-cache show openrazer-daemon | grep Version

_(Can check which repository in **Software & Updates** → Other Software tab)_

Regarding secure boot, did you do anything special to [sign the modules](https://github.com/openrazer/openrazer/wiki/Secure-Boot)? It looks like they're loaded (no message from `modprobe`) but it wouldn't surprise me if _nothing happens_ still because it's still unsigned/untrusted. **That's problem A.**

---

> It's the same problem where clicks work and the scroll wheel works but it only moves along the x-axis.

**For problem B:** To rule things out further, does the mouse work fine if you enter the UEFI settings in your BIOS? (assuming it has cursor support)

I would guess there's a stray system configuration file that's dictating the mouse behaviour. First, is it Wayland or X11?

    env | grep SESSION_TYPE

If it's X11, check `/etc/X11/xorg.conf.d/` - by default, it's empty in 22.04. In case of a stray local configuration file, there could be something in your home folder (`~/.config` or `~/.local/share`), try creating a new user, and see if it happens in a fresh profile.

lah7, 2024-10-31:
Problem solved then! See our [secure boot wiki page ](https://github.com/openrazer/openrazer/wiki/Secure-Boot) for details on that screen. Could it be that it was already set up before?

closed 2024-10-31 by lah7