openrazer/openrazer #2308: Connecting a razer mouse model to a different usb dongle

AnasL2, 2024-09-22:
Hello everyone.
I'm new to the programing world so please don't be harsh on me XD. and **I'm a windows user**.
My issue is that i have a **Razer Deathadder V2 Pro** that came with non functional usb dongle, and i got my hands on a functioning **Viper V2 pro usb dongle**. My question is , can i use the Viper V2 Pro usb dongle to connect my Razer Deathadder V2 Pro using the hyperspeed wireless function?
Thanks in advance.
Product Number: **RZ01-03350400**
Model Number: **RZ01-0335**

z3ntu, 2024-09-23:
If you can achieve this using Razer Synapse then we can probably also build this functionality into OpenRazer. If not, then not.

For the HyperPolling Wireless Dongle there's the sysfs files `hyperpolling_wireless_dongle_pair` and `hyperpolling_wireless_dongle_unpair` which do this, but as I said, I don't know if the same works for other dongles also.

AnasL2, 2024-09-26:
hi , 
[image]
I'm connecting my mouse through Bluetooth and the viper v2 pro usb dongle 
as you can see i can accesses to my mouse features but not to the other one as i don't have it .
I'm not sure if i completely understood what you meant.

KohlerVG, 2024-09-27:
How can I pair the Razer Hyper Polling Wireless Dongle to my Razer Basilisk V3 Pro using OpenRazer?

z3ntu, 2024-10-02:
@KohlerVG Something like `echo $pid_of_new_mouse > /sys/bus/hid/drivers/razermouse/0*/hyperpolling_wireless_dongle_pair` so e.g. if you want to pair DeathAdder V2 Pro it would be `echo 007D > [...]` I think, or 0x007D, not sure actually.

Currently we don't have a better way for that, I don't even know which mice can be paired like that.

KohlerVG, 2024-10-02:
@z3ntu - It worked! I used the command: echo 0x00AB > /sys/bus/hid/drivers/razermouse/0*/hyperpolling_wireless_dongle_pair

Is there a way to pair it with the Razer Mouse Dock Pro too?

z3ntu, 2024-10-06:
If that `hyperpolling_wireless_dongle_pair` file exists then you could also use that. If not, then either it's not implemented in OpenRazer for that device, or the device just doesn't support this functionality. Does the Windows driver support doing what you want to do?

KohlerVG, 2024-10-06:
@z3ntu - Yes, Razer Synapse can pair it, so I'm wondering if OpenRazer can pair it too and if it used the same protocol for the Razer Dock Pro?

z3ntu, 2024-10-20:
If you can take USB packet captures of Synapse doing this, we can try and implement it in OpenRazer also. https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Augvoltz, 2025-02-10:
can someone make a tutorial on how to d this

gucan98, 2025-07-18:
Hello , i have the same problem as OP but i have a adapter of VIPER V3 PRO that i would like to work on a V2 PRO is that possible and please if you have ideas make it easy i'm not into coding at all