openrazer/openrazer #2310: Fedora 41

svenjacobs, 2024-09-24:
[Fedora 41 Beta](https://fedoramagazine.org/announcing-fedora-linux-41-beta/) has been released recently. The stable version is expected by end of October. I kindly ask you to provide packages for Fedora 41 🙏🏼 Thank you very much! 😃

z3ntu, 2024-09-24:
Unfortunately it looks like Fedora 41 is not yet available on build.opensuse.org. See https://build.opensuse.org/project and type "Fedora:" in the search.

Perhaps the Rawhide packages will work on 41 right now? Or you can build it from source: https://github.com/openrazer/openrazer/wiki/Building-a-package

svenjacobs, 2024-09-24:
Thanks for the information. Unfortunately I'm not familiar with OBS and don't know where to request a new distribution version. The Rawhide packages seem to work, though.

z3ntu, 2024-09-24:
We'll just have to wait until closer to the release date until Fedora 41 will be available on OBS, nothing much to do until then.

But I'm glad the Rawhide packages work for now!

andrewschott, 2024-10-15:
The instructions are also a bit off due to the switch to dnf5.  The correct repo add command is something akin to:

```
sudo dnf config-manager addrepo --from-repofile=https://download.opensuse.org/repositories/hardware:/razer/Fedora_Rawhide/hardware:razer.repo --overwrite
```

Take note that the _**--add-repo**_ is changed to _**addrepo**_ and the url requires _**--from-repofile=**_

lah7, 2024-10-15:
Thanks for the correction. I've updated instructions on the website for Fedora Rawhide.

Still looks like OpenSUSE Build Service hasn't added Fedora 41 yet!

z3ntu, 2024-10-17:
Awesome!

closed 2024-10-17 by z3ntu