openrazer/openrazer #2313: This supported device wasn't detected by the OpenRazer daemon (v3.5.1)

xr34p, 2024-10-02:
### Affected Device / PID

Razer Ornata V2, 1532:025d

### Operating System / Version

Parrot OS 6.2

### Kernel Version

6.9.7-amd64

### OpenRazer Version

3.5.1

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I've been looking through several issues but nothing solved the problem. Rebooting or unplugging the keyboard doesn't fix the problem.
[image]

lah7, 2024-10-02:
What do you see when you click "Troubleshoot"? It'll run through some checks and let you know what's not properly installed (most likely the driver, or user `plugdev` group).

It seems Parrot OS 6.2 is based on Debian 12.

xr34p, 2024-10-02:
got fixed by itself after coming back home

closed 2024-10-02 by xr34p