openrazer/openrazer #2314: Support for Razer Barracuda

powersemmi, 2024-10-04:
### Device Name

Razer Barracuda

### Product Number

RZ04-03790200

### Model Number

RZ04-0379

### VID:PID

1532:053c

### USB Details

```shell
Bus 005 Device 003: ID 1532:053c Razer USA, Ltd Razer Barracuda 2.4
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x053c Razer Barracuda 2.4
  bcdDevice            0.01
  iManufacturer           1 Macronix
  iProduct                2 Razer Barracuda 2.4
  iSerial                 3 1234
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e0
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
[... 254 lines cut ...]
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0

005:002:000:DESCRIPTOR         1728014219.990081
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 01 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

powersemmi, 2024-10-04:
I will try to add pcap and screenshots this weekend.

z3ntu, 2024-10-04:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

Menta14, 2025-01-01:
This is the hid dump i get when filtering by both vid and pid
```shell
001:008:003:DESCRIPTOR         1735764657.914263
 06 00 FF 09 00 A1 01 85 01 19 00 29 3F 15 00 26
 FF 00 75 08 95 3F 81 00 19 00 29 3F 15 00 26 FF
 00 75 08 95 3F 91 00 C0 05 0C 09 01 A1 01 85 02
 09 E9 09 EA 09 E2 75 01 95 03 15 00 25 01 81 02
 75 05 95 01 81 03 09 B6 09 CD 09 B5 75 01 95 03
 81 02 75 05 95 01 81 03 09 CF 75 01 95 01 81 02
 75 0F 95 01 81 03 C0
```
Here is a packet capture too
[baracuda_1532_053c.pcapng.zip](https://github.com/user-attachments/files/18286581/baracuda_1532_053c.pcapng.zip)

and synapse screenshots
[image]
[image]
[image]
[image]

I intend to open a branch for this issue, if that's ok with @powersemmi

Menta14, 2025-01-01:
I will come back with a test fork after i get familiar with the codebase.

z3ntu, 2025-01-02:
FWIW I don't see any relevant packets in the pcap, but maybe just the capture didn't work correctly or my script doesn't.

But e.g. bass boost or microphone settings are all software side most likely and not hardware.

Menta14, 2025-01-02:
> FWIW I don't see any relevant packets in the pcap, but maybe just the capture didn't work correctly or my script doesn't.
> 
> But e.g. bass boost or microphone settings are all software side most likely and not hardware.

I tried to capture more packets but nothing shows up. I think most features are strictly software related like you said, which kinda makes sense because this headset is quite basic. I will just add it in the source code for the sake of UI tho.

Menta14, 2025-01-02:
Ok I got something, but i get an internal error in Ubuntu.
`hostname kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 000000000000000000000000000000000000 .`
Obviously not related to the newly created razerbarracuda, so I don't know if it is just from not booting the VM correctly or a legit issue.

maxreitner, 2025-09-08:
> Ok I got something, but i get an internal error in Ubuntu. `hostname kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, [...]

is there progress on the barracuda?