openrazer/openrazer #2315: Cant get Ripple to work

Azer0s, 2024-10-06:
### Affected Device / PID

Razer Deathstalker V2 Pro TKL

### Operating System / Version

Arch Linux x86_64

### Kernel Version

6.11.2-arch1-1

### OpenRazer Version

3.8.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [X] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [X] My device is missing a feature.

### Please describe the issue

Title says it all. Ripple works under Windows, but not on OpenRazer. I tried updating the firmware to 1.0.7

lah7, 2024-10-09:
Is the ripple effect simply not working at all, or only at startup?

Do you have a frontend (like Polychromatic or RazerGenie) installed to try and create a custom effect? For example, lighting all the keys up in different colours and applying that? Ripple is just a software effect, but handled by the daemon, so it behaves the same as a "custom" effect would.

If it's only at startup, check `~/.config/openrazer/razer.conf` for:
```ini
[Startup]
restore_persistence = True
```

Azer0s, 2024-10-11:
I'm using Polychromatic. Ripple is not working at all :/

lah7, 2024-10-12:
In Polychromatic, go to "Device Info" then "Inspect Matrix", if you select the grid, do the keys light up? I ask, because ripple uses the same _protocol_ (it's a custom effect), but it's handled by `openrazer-daemon`.

Also, please could you paste/upload the log for `~/.local/share/openrazer/logs/razer.log`?

Azer0s, 2024-10-21:
> Please could you paste/upload the log for `~/.local/share/openrazer/logs/razer.log`?

<details>
<summary><h2>~/.local/share/openrazer/logs/razer.log content</h2></summary>

```
2024-10-07 01:21:47 | razer                          | CRITICAL | User is not a member of the plugdev group
2024-10-07 01:21:47 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2024-10-07 01:22:43 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2033
2024-10-07 01:22:43 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-07 01:22:43 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-07 01:22:43 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-07 01:22:43 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-07 01:22:43 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-07 01:22:43 | razer                          | INFO     | Serving DBus
2024-10-07 01:31:05 | razer.device0                  | INFO     | Suspending RazerDeathStalkerV2ProTKLWireless
2024-10-08 00:12:14 | razer.device0                  | INFO     | Resuming RazerDeathStalkerV2ProTKLWireless
2024-10-08 00:12:14 | razer.device0                  | INFO     | Putting device back into driver mode.
2024-10-08 13:33:29 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2190
2024-10-08 13:33:29 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-08 13:33:29 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-08 13:33:29 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-08 13:33:29 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-08 13:33:29 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-08 13:33:29 | razer                          | INFO     | Serving DBus
2024-10-08 13:33:56 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 3226
2024-10-08 13:33:56 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-08 13:33:56 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-08 13:33:56 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-08 13:33:56 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-08 13:33:56 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-08 13:33:56 | razer                          | INFO     | Serving DBus
2024-10-08 23:25:22 | razer.device0                  | INFO     | Suspending RazerDeathStalkerV2ProTKLWireless
2024-10-09 00:20:56 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2222
2024-10-09 00:20:56 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-09 00:20:56 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-09 00:20:56 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-09 00:20:56 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-09 00:20:56 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-09 00:20:56 | razer                          | INFO     | Serving DBus
2024-10-09 13:30:02 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2042
2024-10-09 13:30:02 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-09 13:30:02 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-09 13:30:02 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-09 13:30:02 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-09 13:30:02 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-09 13:30:02 | razer                          | INFO     | Serving DBus
2024-10-09 23:20:25 | razer.device0                  | INFO     | Suspending RazerDeathStalkerV2ProTKLWireless
2024-10-10 10:36:34 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 2165
2024-10-10 10:36:34 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-10 10:36:34 | razer                          | INFO     | Found device.0: 0003:1532:0296.0006
2024-10-10 10:36:34 | razer.device0                  | INFO     | Initialising device.0 RazerDeathStalkerV2ProTKLWireless
2024-10-10 10:36:34 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-10 10:36:34 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-10 10:36:34 | razer                          | INFO     | Serving DBus
[... cut ...]
2024-10-21 16:25:00 | razer                          | INFO     | Found device.1: 0003:1532:0298.000B
2024-10-21 16:25:00 | razer.device1                  | INFO     | Initialising device.1 RazerDeathStalkerV2ProTKLWired
2024-10-21 16:25:00 | razer.device1.keymanager       | WARNING  | No event files for KeyWatcher
2024-10-21 16:25:00 | razer.device1                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-10-21 16:25:00 | razer                          | INFO     | Serving DBus
```

</details>

Azer0s, 2024-10-21:
```
> ls -lah /dev/input/by-id/

drwxr-xr-x 2 root root 220 Oct 21 16:24 ./
drwxr-xr-x 4 root root 700 Oct 21 16:25 ../
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL_000000000000-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL_000000000000-if01-event-mouse -> ../event17
lrwxrwxrwx 1 root root   9 Oct 21 16:24 usb-Razer_DSV2Pro_TKL_000000000000-if01-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL_000000000000-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 Oct 21 16:24 usb-Razer_DSV2Pro_TKL_000000000000-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL-if01-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 Oct 21 16:24 usb-Razer_DSV2Pro_TKL-if02-event-mouse -> ../event21
lrwxrwxrwx 1 root root   9 Oct 21 16:24 usb-Razer_DSV2Pro_TKL-if02-mouse -> ../mouse1
```

Azer0s, 2024-10-24:
I applied the patch and started openrazer like so `PYTHONPATH="pylib:daemon" /usr/bin/python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf` but it still didn't work. I have also attached the log.

```
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_current_effect
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_brightness
2024-10-24 15:52:48 | razer.device0                  | DEBUG    | DBus call get_game_mode
2024-10-24 15:52:54 | razer                          | DEBUG    | State recently changed, writing to disk
2024-10-24 15:52:54 | razer                          | DEBUG    | Writing persistence config
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call set_reactive_effect
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerDeathStalkerV2ProTKLWireless:PM2330F67100293, 'setReactive', dbus.Byte(0), dbus.Byte(255), dbus.Byte(0), dbus.Byte(2))
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, reactive)
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | Set persistence (backlight, speed, 2)
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_brightness
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_game_mode
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call set_ripple_effect
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerDeathStalkerV2ProTKLWireless:PM2330F67100293, 'setRipple', dbus.Byte(0), dbus.Byte(255), dbus.Byte(0), dbus.Double(0.05))
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, ripple)
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_wave_dir
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_speed
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_brightness
2024-10-24 15:53:00 | razer.device0                  | DEBUG    | DBus call get_game_mode
2024-10-24 15:53:04 | razer                          | DEBUG    | State recently changed, writing to disk
2024-10-24 15:53:04 | razer                          | DEBUG    | Writing persistence config
```

z3ntu, 2024-12-07:
Can you do this part from my previous comment:

> If this doesn't work, please run `cat /dev/input/by-id/usb-Razer_DSV2Pro_TKL_000000000000-event-kbd` and on the other files with `usb-Razer_DSV2Pro_TKL` and check if any output appears when you press something on the keyboard - and if yes **which** file gives some output.

colepdn, 2025-02-28:
Hi! I've looked into this myself, and found that the correct files are `DSV2Pro_TKL_000000000000-if01-event-mouse` when wireless and `DSV2Pro_TKL-if01-event-kbd` when wired.

I've made PR #2404, which resolves this issue on my machine.

PR #2404 (2025-03-15): daemon: Fix filename for Deathstalker V2 Pro TKL
Closes #2315

The daemon was reading from the wrong file for keypress events, this resolves that problem, and the ripple effect now works on my keyboard.
closed 2025-03-15 by z3ntu