openrazer/openrazer #2316: Support for Razer [BlackShark V2 HS (enter connection type here)]

konn-neko, 2024-10-07:
### Device Name

Razer BlackShark V2 HS (enter connection type here)

### Product Number

RZ04-04960100-R3U1

### Model Number

RZ04-0496

### VID:PID

1532:0565

### USB Details

```shell
Bus 001 Device 005: ID 1532:0565 Razer USA, Ltd Razer BlackShark V2 HS 2.4
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0565 Razer BlackShark V2 HS 2.4
  bcdDevice            1.00
  iManufacturer           1 MediaTek Inc
  iProduct                2 Razer BlackShark V2 HS 2.4
  iSerial                 3 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e1
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              4 Razer BlackShark V2 HS 2.4
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004e
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
[... 235 lines cut ...]
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-event-kbd -> ../event13
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-event-mouse -> ../event9
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-mouse -> ../mouse0
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-mouse -> ../mouse1
lrwxrwxrwx     - root  6 Oct 21:24   usb-Logitech_G935_Gaming_Headset-event-if03 -> ../event7
lrwxrwxrwx     - root  6 Oct 21:24   usb-MediaTek_Inc_Razer_BlackShark_V2_HS_2.4_0000000000000000-event-if03 -> ../event3
```

### USB HID Dump

```shell
001:005:003:DESCRIPTOR         1728269065.509888
 06 13 FF 09 01 A1 01 15 00 26 FF 00 85 06 09 00
 75 08 95 3D 91 02 85 07 09 00 75 08 95 3D 81 02
 C0 05 0C 09 01 A1 01 85 0C 15 00 25 01 09 E9 09
 EA 09 E2 09 CD 09 B5 09 B6 75 01 95 06 81 02 09
 00 95 02 81 02 C0 05 0B 09 05 A1 01 15 00 25 01
 75 01 85 05 09 20 95 01 81 22 09 2F 95 01 81 06
 09 24 09 21 09 97 09 2A 09 50 95 05 81 06 09 07
 05 09 09 01 75 01 95 01 81 02 05 08 85 05 09 17
 09 09 09 18 09 20 09 21 09 2A 95 06 91 22 95 02
 91 01 C0 06 14 FF 09 01 A1 01 15 00 26 FF 00 85
 02 09 00 75 08 95 08 B2 02 01 85 02 09 00 75 08
 95 3F 91 02 85 02 09 00 75 08 95 3F 81 02 C0
```

konn-neko, 2024-10-07:
about to add the direct usb connection too

konn-neko, 2024-10-07:
Vendor UID
```1532:056e```

USB details
```

Bus 001 Device 016: ID 1532:056e Razer USA, Ltd Razer BlackShark V2 HS USB
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x056e Razer BlackShark V2 HS USB
  bcdDevice            1.00
  iManufacturer           1 MediaTek Inc
  iProduct                2 Razer BlackShark V2 HS USB
  iSerial                 3 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e1
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              4 Razer BlackShark V2 HS USB
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004e
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 2
        bSourceID               1
[... 220 lines cut ...]
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-event-mouse -> ../event12
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-event-kbd -> ../event13
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-event-mouse -> ../event9
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-if01-mouse -> ../mouse0
lrwxrwxrwx     - root  6 Oct 21:24   usb-Compx_2.4G_Wireless_Receiver-mouse -> ../mouse1
lrwxrwxrwx     - root  6 Oct 21:24   usb-Logitech_G935_Gaming_Headset-event-if03 -> ../event7
lrwxrwxrwx     - root  6 Oct 21:49   usb-MediaTek_Inc_Razer_BlackShark_V2_HS_USB_0000000000000000-event-if03 -> ../event4
```
USB HID Dump
```
001:016:003:DESCRIPTOR         1728269535.102230
 06 13 FF 09 01 A1 01 15 00 26 FF 00 85 06 09 00
 75 08 95 3D 91 02 85 07 09 00 75 08 95 3D 81 02
 C0 05 0C 09 01 A1 01 85 0C 15 00 25 01 09 E9 09
 EA 09 E2 09 CD 09 B5 09 B6 75 01 95 06 81 02 09
 00 95 02 81 02 C0 05 0B 09 05 A1 01 15 00 25 01
 75 01 85 05 09 20 95 01 81 22 09 2F 95 01 81 06
 09 24 09 21 09 97 09 2A 09 50 95 05 81 06 09 07
 05 09 09 01 75 01 95 01 81 02 05 08 85 05 09 17
 09 09 09 18 09 20 09 21 09 2A 95 06 91 22 95 02
 91 01 C0 06 14 FF 09 01 A1 01 15 00 26 FF 00 85
 02 09 00 75 08 95 08 B2 02 01 85 02 09 00 75 08
 95 3F 91 02 85 02 09 00 75 08 95 3F 81 02 C0

```

konn-neko, 2024-10-07:
There is also bluetooth capability but im unsure how to read its info, this headset has no rgb, just a battery indicator and the thx spatial surround

z3ntu, 2024-10-20:
Would be interesting to see what functionality is actually done via the regular Razer protocols, spatial sound etc is done in software normally as far as I know.

sevy2011, 2025-03-22:
I would love to see this as well
this headset also has a few equalizer settings you can save on device afaik
also connecting to the dongle is way faster on windows with razer synapse than on linux don't know if thats synapse or windows tho

Holcane1337, 2025-04-09:
Does anyone know which Profile this headset should use?

[image]

If I put on 5.1 it sends a horrible loud shrill into the headset.

GegudeBR, 2025-05-15:
I just did a bunch of captures.
This is the first time I'm doing this, so haven't done any protocol decoding yet.
[RazerBlackSharkV2HyperSpeed.zip](https://github.com/user-attachments/files/20234756/RazerBlackSharkV2HyperSpeed.zip)