openrazer/openrazer #2317: Battery notifications does not dissapear on GNOME

HenrikHolst, 2024-10-07:
### Affected Device / PID

Razer DeathAdder V3 PRO

### Operating System / Version

Ubuntu 24.04

### Kernel Version

6.8.0-45-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [X] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

The battery notification adds new rows/events and does not close the previous one on GNOME (v46). Unsure if this started in 3.9.0 or if it simply us more prominent in 3.9.0 since in previous versions the notifications was very rare.
[image]

lah7, 2024-10-09:
It's very likely this started in 3.9.0, since OpenRazer switched to using the [`notify-send` command](https://man.archlinux.org/man/notify-send.1.en) — https://github.com/openrazer/openrazer/commit/c93c3cfe3dbda2c120ce455a43ee4395098f65e8

I tried experimenting with the `notify-send` command under GNOME Shell 46 / Ubuntu 24.04, but the other combinations I tried didn't seem to clear past notifications.

We could add:

* `-e` to be a transient notification, meaning, as soon as it disappears, it's not in the history.
* `-u low` for a low priority notification which doesn't show a pop up, but it still stacks up in the history, even with `-e` strangely.
* `-p` and `-r ID` is supposed to give notifications an ID, so it can replace the last one, but that doesn't seem to work in GNOME.

It could be possible to include `-e` to simply not add it history, if you wanted a 4 second window to read the notification before it's gone forever until the next arrives.

IMO, long term, notifications are not really the correct UI. The superior approach would be to integrate with the desktop's power/battery settings - still a work in progress though.
* #87 
* #2160

HenrikHolst, 2024-10-09:
Interesting that -p and -r doesn't work in GNOME because (but I guess they use a different API perhaps) other applications like Rhyhmbox and Firefox (when playing a video, Firefox adds it as a notification) can add and remove notifications.  Ofc upower looks to be the proper way.