openrazer/openrazer #2319: Razer Basilisk V3 Different Profiles Not Saving Colours

AlexanderTheGr3at, 2024-10-08:
### Affected Device / PID

Basilisk V3 1532:0099 

### Operating System / Version

Ubuntu 24.04

### Kernel Version

6.8.0-45-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I only seem to be able to save a colour on one profile of the mouse. I have three set (one for media, work, gaming). When I have it set to media it stays the colour I want. When I switch via either the dedicated button or the one I set up in synapse (before I uninstalled it) it reverts back to multi-colour flashing.

lah7, 2024-10-08:
Sadly, that is expected at the moment - only the primary profile is supported. OpenRazer communicates with the firmware more directly. The "profiles" feature (also known as "onboard memory") wasn't implemented nor [understood](https://github.com/openrazer/openrazer/wiki/Unknown-commands#on-board-memory). 

The hardware is mostly controlled by a driver (software), so it was most likely Razer Synapse that was handling the effect/colours per profile. When the firmware's on its own (or under OpenRazer), it doesn't have anything in its memory so spectrum is usually the default.

Similar issue to https://github.com/openrazer/openrazer/issues/1630

As an alternate, you could rig up some scripts to some global/keyboard shortcuts (depending on your desktop environment), or see if there is a keypress emitted when that profile button is clicked.

AlexanderTheGr3at, 2024-10-09:
> Sadly, that is expected at the moment - only the primary profile is supported. OpenRazer communicates with the firmware more directly. The "profiles" feature (also known as "onboard memory") wasn't [...]

Any idea how I could go about checking if there is a keypress when the profile button is clicked? And if so what could I do with that information? I really want to use this mouse but I can't justify their awful awful software aha.

lah7, 2024-10-09:
You can try the command line tools `xev` and/or `evtest`. If a click is registered when you press the button, you're in luck.

If not, it's likely the daemon  _swallows_ the input because the device is in "driver mode" but hasn't implemented it (#2265). You could try switching it back to "normal mode" using the command in https://github.com/openrazer/openrazer/issues/2265#issuecomment-2204523897 to test but try `\x00\x00` instead of `\x03\x00`, and restart `xev`/`evtest`. Note that the device _might not_ change effects in normal mode.

If that doesn't work either, then the mouse might not emit a keypress. In which case, even if OpenRazer wanted to, we couldn't use that button.

If there is a keypress, then you could write a Python script to interface with the OpenRazer library to implement whatever logic you want. In GNOME/KDE there's usually a way to set a global shortcut to run an action.

closed 2024-10-20 by AlexanderTheGr3at