openrazer/openrazer #2321: Blackwidow V3 Pro (Wireless) still not working latest update

plexecutor, 2024-10-10:
### Affected Device / PID

1532:025c

### Operating System / Version

Bazzite 3.7.0

### Kernel Version

6.9.12-205.fsync.fc40.x86_64

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

#2231 which was merged back in July was supposed to have added support for the Blackwidow V3 Pro (Wireless). OpenRazer 3.9.0 which just released a few days ago includes this merge and lists the keyboard as now supported in the changelog, however it is not working for me.

`~$ lsusb | grep 1532
Bus 001 Device 006: ID 1532:025c Razer USA, Ltd Razer BlackWidow V3 Pro
Bus 003 Device 003: ID 1532:00a4 Razer USA, Ltd Razer Mouse Dock Pro`

I have a Basilisk V3 Pro connected to a Mouse Dock Pro and the Blackwidow V3 Pro is connected via 2.4ghz to the dongle it came with and the keyboard shows up in RazerGenie when wired.

z3ntu, 2024-10-20:
Can you please run this and post the log? `systemctl --user stop openrazer-daemon.service` and then `openrazer-daemon -Fv` while the keyboard is connected wireless?

Toixinhell, 2024-10-24:
``
openrazer-daemon -Fv
Starting daemon.
2024-10-24 07:44:28 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 150870
2024-10-24 07:44:28 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeedWireless - (1532:0271)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3ProWired --------------- (1532:025A)
2024-10-24 07:44:28 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3TK --------------------- (1532:0A24)
[... cut ...]
^C2024-10-24 07:44:57 | razer                          | INFO     | Stopping daemon on signal 2
2024-10-24 07:44:57 | razer.device0.keymanager       | DEBUG    | Stopping key manager
2024-10-24 07:44:57 | razer.device0.ripplemanager    | DEBUG    | Closing Ripple Manager
2024-10-24 07:44:57 | razer.device0.batterymanager   | DEBUG    | Closing Battery Manager
2024-10-24 07:44:57 | razer.device0.batterynotifier  | DEBUG    | Shutting down battery notifier
2024-10-24 07:44:57 | razer                          | DEBUG    | Writing persistence config
Stopping daemon.
``

Toixinhell, 2024-10-24:
Volume mute(bcd), forward(bcd), backward(bcd), volume wheel all not working. In brackets is what the keyboard types if I press it)

closed 2025-03-24 by github-actions[bot]