openrazer/openrazer #2323: Polling rate issues; is it the dongle?

rlind3r, 2024-10-13:
### Affected Device / PID

Razer DeathAdder V3 Pro, Hyperpolling Wireless Dongle

### Operating System / Version

Pop!_OS noble 24.04

### Kernel Version

6.9.3-76060903-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

[image]
The hyperpolling wireless dongle is supported and appears to be working but won't be detected by Polychromatic, RazerGenie, etc. and it doesn't appear in lsusb either.

rlind3r, 2024-10-15:
Doesn't show up in a Windows 10 VM either, just shows up as the DeathAdder V3 Pro and not its own device so it seems like there might be other variations of the Hyperpolling Wireless Dongle that differ from the one that's actually supported by OpenRazer?

z3ntu, 2024-10-20:
If the device doesn't show up in `lsusb` then OpenRazer also can't communicate with the device. Then it's general USB problems or something.

rlind3r, 2024-10-27:
> If the device doesn't show up in `lsusb` then OpenRazer also can't communicate with the device. Then it's general USB problems or something.

Apparently there's more than one "version" of the Hyperpolling dongle, the first one I had came with a DA V3 and only works with it, doesn't work with other mice which is why it never showed up even in Synapse when I ran Win10 through a VM. I got a separate Hyperpolling dongle after the fact and it works fine.

closed 2024-10-27 by rlind3r