openrazer/openrazer #2324: Support for Razer [name]

narzdavid, 2024-10-14:
### Device Name

Razer Core X

### Product Number

RC21-01311101100

### Model Number

RC21-0131

### VID:PID

1532:02a5

### USB Details

```shell
$ lsusb -d 1532:02a5 -v

Bus 005 Device 004: ID 1532:02a5 Razer USA, Ltd Razer BlackWidow V4 75%
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a5 Razer BlackWidow V4 75%
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V4 75%
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
[... 207 lines cut ...]
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:005:001:DESCRIPTOR         1728896061.426759
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

003:005:000:DESCRIPTOR         1728896061.429749
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

z3ntu, 2024-10-20:
The information you posted doesn't contain anything from Razer Core X. Does that thing not show up over USB? What of it can you control in Razer Synapse on Windows?

turbos-and-tacos, 2025-01-27:
@narzdavid What are you wanting to control on the Core X? IIRC the Core X doesn't have any LEDs. @z3ntu I have some time to help with things and I'm looking to pitch in.

narzdavid, 2025-02-03:
Hi there, thanks for taking a look at my support request. Apologies for not replying back sooner! I looked into my hardware and turns out my ThinkPad P12s Gen2 does not support egpus since my port is only USB-C 3.2 instead of Thunderbolt. For that reason, it does not show the GPU at all. It doesn't matter though since I don't game on this laptop. I'll be closing this issue now 🙇‍

closed 2025-02-03 by narzdavid