openrazer/openrazer #2326: Support for Razer [Razer Huntsman V3 X TKL]

julesartd, 2024-10-17:
### Device Name

Razer Huntsman V3 X TKL

### Product Number

RZ03-050220500-R3F1

### Model Number

RZ03-0502

### VID:PID

1532:02b1

### USB Details

```shell
Bus 003 Device 011: ID 1532:02b1 Razer USA, Ltd Razer Huntsman V3 X Tenkeyless
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02b1 Razer Huntsman V3 X Tenkeyless
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman V3 X Tenkeyless
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     327
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```

### Device Input Paths

```shell
total 0
drwxr-xr-x. 2 root root 200 17 oct.  09:25 .
drwxr-xr-x. 4 root root 780 17 oct.  09:26 ..
lrwxrwxrwx. 1 root root   9 17 oct.  09:25 usb-MOSART_Semi._2.4G_Wireless_Mouse-event-if00 -> ../event6
lrwxrwxrwx. 1 root root   9 17 oct.  09:25 usb-MOSART_Semi._2.4G_Wireless_Mouse-event-mouse -> ../event5
lrwxrwxrwx. 1 root root   9 17 oct.  09:25 usb-MOSART_Semi._2.4G_Wireless_Mouse-mouse -> ../mouse1
lrwxrwxrwx. 1 root root  10 17 oct.  09:25 usb-Razer_Razer_Huntsman_V3_X_Tenkeyless-event-if01 -> ../event26
lrwxrwxrwx. 1 root root  10 17 oct.  09:25 usb-Razer_Razer_Huntsman_V3_X_Tenkeyless-event-kbd -> ../event24
lrwxrwxrwx. 1 root root  10 17 oct.  09:25 usb-Razer_Razer_Huntsman_V3_X_Tenkeyless-if01-event-kbd -> ../event25
lrwxrwxrwx. 1 root root  10 17 oct.  09:25 usb-Razer_Razer_Huntsman_V3_X_Tenkeyless-if02-event-mouse -> ../event27
lrwxrwxrwx. 1 root root   9 17 oct.  09:25 usb-Razer_Razer_Huntsman_V3_X_Tenkeyless-if02-mouse -> ../mouse4
```

### USB HID Dump

```shell
003:011:003:DESCRIPTOR         1729151004.395649
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2B 09 2C 09
 2D 15 00 26 FF 00 75 08 95 06 B1 02 C0 85 04 09
 50 A1 02 09 03 09 55 15 00 25 08 75 08 95 02 B1
 02 09 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 09 51 09 52 09 53 09 54 09 51 09 52 09 53 09 54
 15 00 26 FF 00 75 08 95 20 B1 02 C0 85 05 09 60
 A1 02 09 55 15 00 25 08 75 08 95 01 B1 02 09 61
 09 62 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09
 51 09 52 09 53 09 54 15 00 26 FF 00 75 08 95 04
 B1 02 C0 85 06 09 70 A1 02 09 71 15 00 25 01 75
 08 95 01 B1 02 C0 C0

003:011:002:DESCRIPTOR         1729151004.396453
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

003:011:001:DESCRIPTOR         1729151004.396994
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

003:011:000:DESCRIPTOR         1729151004.397661
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

### Synapse Screenshots

[image]

z3ntu, 2024-10-20:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

TheDarkUndoing, 2025-01-02:
I have this keyboard as well, I can get the PCAP created if need be. Is there a standard process for getting the capture?

julesartd, 2025-11-24:
> any idea how to get started on this? i just purchased one and would love to customize it

Hey!
I haven’t started anything on this yet.
If you decide to work on it or make any progress, I’d love it if you could keep me updated!