openrazer/openrazer #2327: Support for Razer [name]

EndoliteMatrix, 2024-10-18:
### Device Name

Razer Thunderbolt 4 Dock Mercury (NASA EDITION)

### Product Number

Impossible to find

### Model Number

Incorrect links

### VID:PID

Bus 001 Device 010: ID 1532:122e Razer USA, Ltd Thunderbolt 4 Dock

### USB Details

```shell
Bus 004 Device 003: ID 1532:0f30 Razer USA, Ltd USB3.1 Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f30 
  bcdDevice           30.60
  iManufacturer           1 Razer
  iProduct                2 USB3.1 Hub
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0

Bus 001 Device 010: ID 1532:122e Razer USA, Ltd Thunderbolt 4 Dock
[... 295 lines cut ...]
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:004:001:DESCRIPTOR         1729273840.123719
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:004:000:DESCRIPTOR         1729273840.124219
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

┌─[puser@parrot]─[~]
```

z3ntu, 2024-10-20:
Is there anything that Razer Synapse on Windows can control on this device?

EndoliteMatrix, 2025-01-03:
> Is there anything that Razer Synapse on Windows can control on this device?

Sorry for the delay in response. For some reason this model dock does not interact with Razer synapse, and I have NO idea where to find any drivers or firmware for this. Upon looking deeper into what the dock was called in my Razer account - it says Razer Thunderbolt 4 Dock Mercury - NASA Packaging