openrazer/openrazer #2333: Support for Razer [name]

yonizosII, 2024-10-22:
### Device Name

Razer Leviathan V2

### Product Number

RZ05-03920100

### Model Number

RC30-039201

### VID:PID

1532:0532

### USB Details

```shell
Bus 003 Device 014: ID 1532:0532 Razer USA, Ltd Razer Leviathan V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0532 
  bcdDevice           10.41
  iManufacturer           1 Razer
  iProduct                2 Razer Leviathan V2
  iSerial                 3 0000000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00b9
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        bCategory              10
[... 239 lines cut ...]
 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95
 01 75 10 81 00 75 08 95 13 81 01 C0 05 01 09 80
 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95 03
 81 02 95 05 81 01 75 08 95 14 81 01 C0 05 01 09
 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15
 81 00 C0

003:007:000:DESCRIPTOR         1729616524.350578
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:014:002:DESCRIPTOR         1729616524.351272
 05 0C 09 01 A1 01 15 00 25 01 85 01 09 E9 09 EA
 75 01 95 02 81 02 09 E2 09 00 81 06 05 0C 09 CD
 09 B7 09 B5 09 B6 95 04 81 06 26 FF 00 09 00 75
 08 95 0E 81 02 09 00 95 0F 91 02 85 FF 75 08 95
 0F 81 03 09 00 95 0F 91 02 85 02 09 00 75 08 95
 01 81 02 85 03 09 00 75 08 95 02 91 02 85 05 09
 00 75 08 95 0F 81 02 85 08 09 00 75 08 95 3F B1
 01 85 07 09 00 75 08 95 5A B1 01 C0
```

z3ntu, 2024-10-22:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

Zarf42, 2025-02-24:
[leviathan_v2.zip](https://github.com/user-attachments/files/18945334/leviathan_v2.zip)

z3ntu, 2025-02-24:
I do see some data in the captures but it doesn't seem to match the standard protocol that Razer uses.

But I think if you make Wireshark show `usb.data_fragment` field, and correlate a bit more your actions with what's shown, I think you could figure out what the bits mean.

[image]

But another question for the device, what currently works out of the box for this device? I assume audio things (incl. volume control) but not the RGB?

Dylon0107, 2025-04-05:
Any chance this will get added soon?

z3ntu, 2025-04-19:
@Dylon0107 Somebody owning the device needs to add support for it.

terricain, 2025-04-20:
```
Razer Leviathan V2
 
not red 080003000000200f030000010008ed0000ed0000ed0000ed0000ed0000ed0000ed0000ed0000ed000000000000000000000000000000000000000000000000c8
    red 080011000000200f030000000008ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000db
    red 080012000000200f030000010008ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000da
    
    
    red 080019000000200f030000000008ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000db
    red 08001a000000200f030000010008ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000da
              000000200f030000010008ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000 checksum xor after frame counter
                              ^^ varstore?
                                  ^^ LED Count, 0 based?
 
brightness   0  08001a000000030f0401000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000009
brightness  71  08001b000000030f0401008c00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000085
brightness 100  08001c000000030f040100ff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000f6
                                      ^^ brightness
                                ^^ command id?
                              ^^ command class
                            ^^ command arg byte count
```
Rough guess based on some pcaps I have

[caps.zip](https://github.com/user-attachments/files/19825690/caps.zip)

terricain, 2025-04-20:
Pretty sure right near the beginning is a "sequence number" of sorts, perhaps something to do with "frames stored internally" As whilst the traffic looks like synapse is constantly spamming commands to change leds for a spectrum effect, the soundbar is still doing spectrum without synapse

z3ntu, 2025-08-08:
@sourkyle https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

aloerch, 2026-01-07:
This is a different device than the Leviathan V2 X. I think the Leviathan V2 is larger and has a subwoofer. 

It's possible the PR I submitted for the Leviathan V2 X may have methods that would be helpful in supporting the Leviathan V2, but I don't know.