openrazer/openrazer #2334: razercfg reports no devices with Razer Cobra

bbogart, 2024-10-23:
### Affected Device / PID

Razer Cobra

### Operating System / Version

Ubuntu 20.04

### Kernel Version

5.15.0-122-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Installed openRazer from official ubuntu repo first (version 2.5.0). I saw the Cobra was not supported, so I uninstalled and autoremoved and then following the instructions for the OpenRazer PPA and installed v 3.9.0. I confirm no version 2.5.0 packages are left:

ii  openrazer-daemon              3.9.0~ubuntu20.04.1 all          OpenRazer pe>
un  openrazer-doc                 <none>              <none>       (no descript>
ii  openrazer-driver-dkms         3.9.0~ubuntu20.04.1 all          OpenRazer pe>
un  openrazer-kernel-modules-dkms <none>              <none>       (no descript>
ii  openrazer-meta                3.9.0~ubuntu20.04.1 all          OpenRazer pe>
ii  python3-openrazer             3.9.0~ubuntu20.04.1 all          OpenRazer pe>
un  python3-razer                 <none>              <none>       (no descript>
ii  qrazercfg                     0.41+ds-1build1     all          Graphical Ra>
ii  qrazercfg-applet              0.41+ds-1build1     all          Graphical Ra>
un  razer-chroma-driver           <none>              <none>       (no descript>
un  razer-daemon                  <none>              <none>       (no descript>
un  razer-kernel-modules-dkms     <none>              <none>       (no descript>
ii  razercfg        

Went through the whole troubleshooting guide, everything acts fine, it should just be working, but no device found:

$ razercfg -V
No Razer device found in the system

*Trouble Shooting*
$ lsusb | grep 1532
Bus 001 Device 004: ID 1532:00a3 Razer USA, Ltd Razer Cobra

$ openrazer-daemon --version
openrazer-daemon 3.9.0

$ sudo dkms status | grep openrazer
openrazer-driver, 3.9.0, 5.15.0-122-generic, x86_64: installed

$ sudo dkms install openrazer-driver/3.9.0
Module openrazer-driver/3.9.0 already installed on kernel 5.15.0-122-generic/x86_64

$ lsmod | grep razer
razermouse            135168  0
hid                   147456  3 usbhid,hid_generic,razermouse

$ groups
... adm dialout cdrom sudo dip plugdev lpadmin sambashare

$ 
ps aux | grep openrazer-daemon
...      986  0.0  0.1 267088 25732 ?        Ssl  17:01   0:00 openrazer-daemon

[razer.log](https://github.com/user-attachments/files/17483391/razer.log)

$ dmesg | grep razer
[    4.130466] razermouse: loading out-of-tree module taints kernel.
[    4.130803] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[    4.167178] razermouse 0003:1532:00A3.0003: input,hidraw2: USB HID v1.11 Mouse [Razer Razer Cobra] on usb-0000:00:14.0-4/input0
[    4.341137] razermouse 0003:1532:00A3.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Cobra] on usb-0000:00:14.0-4/input1
[    4.613345] razermouse 0003:1532:00A3.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Cobra] on usb-0000:00:14.0-4/input2

$ sudo journalctl -b | grep -i razer_mount
Oct 22 17:01:02 domi razer_mount[544]: Driver razermouse
Oct 22 17:01:02 domi razer_mount[547]: Device_ID 1-4
Oct 22 17:01:02 domi razer_mount[552]: Modprobing razermouse
Oct 22 17:01:02 domi razer_mount[741]: Modprobed razermouse
Oct 22 17:01:02 domi razer_mount[806]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[820]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[821]: Device_ID 1-4:1.2
Oct 22 17:01:03 domi razer_mount[831]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[832]: Device_ID 0003:1532:00A3.0005
Oct 22 17:01:03 domi razer_mount[834]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0005/
Oct 22 17:01:03 domi razer_mount[813]: Device_ID 1-4:1.1
Oct 22 17:01:03 domi razer_mount[861]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[863]: Device_ID 1-4:1.0
Oct 22 17:01:03 domi razer_mount[869]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[870]: Device_ID 0003:1532:00A3.0004
Oct 22 17:01:03 domi razer_mount[871]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0004/
Oct 22 17:01:03 domi razer_mount[858]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0005/
Oct 22 17:01:03 domi razer_mount[873]: Driver razermouse
Oct 22 17:01:03 domi razer_mount[878]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0004/
Oct 22 17:01:03 domi razer_mount[875]: Device_ID 0003:1532:00A3.0003
Oct 22 17:01:03 domi razer_mount[881]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0003/
Oct 22 17:01:03 domi razer_mount[894]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:00A3.0003/

$ ls -al /sys/bus/hid/drivers/razer*/*1532*/
'/sys/bus/hid/drivers/razermouse/0003:1532:00A3.0003/':
total 0
drwxr-xr-x 5 root plugdev    0 Oct 22 17:01 .
drwxr-xr-x 5 root root       0 Oct 22 17:01 ..
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 country
-rw-rw---- 1 root plugdev 4096 Oct 22 17:01 device_mode
-r--r----- 1 root plugdev 4096 Oct 22 17:01 device_serial
-r--r----- 1 root plugdev 4096 Oct 22 17:01 device_type
-rw-rw---- 1 root plugdev 4096 Oct 22 17:01 dpi
-rw-rw---- 1 root plugdev 4096 Oct 22 17:01 dpi_stages
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 Oct 22 17:01 firmware_version
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 hidraw
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 input
-rw-rw---- 1 root plugdev 4096 Oct 22 17:01 logo_led_brightness
--w--w---- 1 root plugdev 4096 Oct 22 17:01 logo_matrix_effect_breath
--w--w---- 1 root plugdev 4096 Oct 22 17:01 logo_matrix_effect_none
--w--w---- 1 root plugdev 4096 Oct 22 17:01 logo_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Oct 22 17:01 logo_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Oct 22 17:01 logo_matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 modalias
-rw-rw---- 1 root plugdev 4096 Oct 22 17:01 poll_rate
drwxr-xr-x 2 root plugdev    0 Oct 22 17:01 power
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 Oct 22 17:01 test
-rw-r--r-- 1 root plugdev 4096 Oct 22 17:01 uevent
-r--r----- 1 root plugdev 4096 Oct 22 17:01 version

'/sys/bus/hid/drivers/razermouse/0003:1532:00A3.0004/':
total 0
drwxr-xr-x 5 root plugdev    0 Oct 22 17:01 .
drwxr-xr-x 5 root root       0 Oct 22 17:01 ..
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 country
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 driver -> ../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 hidraw
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 input
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 modalias
drwxr-xr-x 2 root plugdev    0 Oct 22 17:01 power
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Oct 22 17:01 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:00A3.0005/':
total 0
drwxr-xr-x 5 root plugdev    0 Oct 22 17:01 .
drwxr-xr-x 5 root root       0 Oct 22 17:01 ..
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 country
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 driver -> ../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 hidraw
drwxr-xr-x 3 root plugdev    0 Oct 22 17:01 input
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 modalias
drwxr-xr-x 2 root plugdev    0 Oct 22 17:01 power
-r--r--r-- 1 root plugdev 4096 Oct 22 17:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Oct 22 17:01 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Oct 22 17:01 uevent

What am I missing?

bbogart, 2024-10-23:
I manually edited ~/.config/openrazer/persistence.conf and indeed, that gets updated on the mouse, so everything is working except for razercfg (v 0.41).

z3ntu, 2024-10-23:
razercfg is completely unrelated to OpenRazer, what are you trying to do?

bbogart, 2024-10-23:
Thanks @z3ntu , understood. I was just looking for a gui because I could not easily find a guide to how to manually edit the config file, i.e. what "effects" are available, what they mean, how the colors work, what the units of speed are, etc. 

All is possible through guess and test and looking at the code directly, but would be easier if I could figure out how to get the daemon to reload the persistence.conf file. I saw the package includes a systemd file /usr/lib/systemd/user/openrazer-daemon.service but systemctl does not find a corresponding service:

$ systemctl restart openrazer-daemon.service
Failed to restart openrazer-daemon.service: Unit openrazer-daemon.service not found.
$ systemctl restart openrazer-daemon
Failed to restart openrazer-daemon.service: Unit openrazer-daemon.service not found.
$ systemctl restart openrazer
Failed to restart openrazer.service: Unit openrazer.service not found.

kill -HUP also did not work.

z3ntu, 2024-10-23:
Please check the "Apps" section on https://openrazer.github.io/#apps

For systemctl you need to use `--user` since it's a user service, not a system service. So e.g. `systemctl --user restart openrazer-daemon.service`.

Also the persistence file is not meant to be edited by hand, it's just a state file for the daemon. So I recommend not touching that.

bbogart, 2024-10-23:
Ah thanks! 

Which file should I be editing then? Seemingly not razer.conf in the same directory?

lah7, 2024-10-23:
OpenRazer is mainly a driver/daemon stack. It's geared towards a GUI/CLI program that would interact with it, rather then editing configuration files. [Polychromatic](https://polychromatic.app/) and [RazerGenie](https://github.com/z3ntu/RazerGenie) (both Qt based) are popular GUI choices. Or try [razer-cli](https://github.com/lolei/razer-cli).

The _"none of the above"_ options would be to talk to D-Bus directly (session bus, `org.razer`) using a tool like [D-Feet](https://wiki.gnome.org/Apps/DFeet) or Qt QDBusViewer. Or use the Python library directly [with these examples.](https://github.com/openrazer/openrazer/tree/master/examples)

I mean, you could get away with editing `persistence.conf`, as long as the daemon isn't running, and the correct variables are used.

bbogart, 2024-10-23:
Ah! Understood; I had assumed that razercfg was exactly that since openrazer was available in official repos and I did not see any other GUIs for openrazer in official repos. I'll check out those projects separately and that closes this issue.