openrazer/openrazer #2339: dmesg ring buffer completely clobbered with "Command timed out" for wireless mice when computer left unattended

steveyh25, 2024-11-02:
### Affected Device / PID

Razer Viper Ultimate

### Operating System / Version

Mint 22

### Kernel Version

6.8.0-48-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [X] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Hi,

Something I have noticed when leaving my computer on over night is that my dmesg buffer is entirely filled with messages like below. I presume this is just because my mouse has a 15 minute timeout before sleeping, and so becomes unresponsive - however it is a bit frustrating that after approx 3 hours of being away from the computer I have lost anything that may have been output to dmesg.

`[70660.186803] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70670.350869] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70680.514821] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70690.677831] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70700.841838] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70711.005848] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70721.169856] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70731.333866] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70741.497875] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70751.661884] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70761.825893] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70771.989902] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70782.153913] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70792.317895] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
[70795.788895] razermouse: Command timed out. status: 04 transaction_id.id: 3f remaining_packets: 00 protocol_type: 00 data_size: 03, command_class: 0f, command_id.id: 04 Params: 0104bf00000000000000000000000000 .
`

edt-xx, 2024-11-08:
Same device, same problem.  It would be VERY nice to see this fixed.

edt-xx, 2024-11-08:
A work around that does the trick here: https://github.com/openrazer/openrazer/issues/2203#issuecomment-1925411515

dm17, 2024-12-10:
> A work around that does the trick here: [#2203 (comment)](https://github.com/openrazer/openrazer/issues/2203#issuecomment-1925411515)

Doesn't fit is for me. I get `razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 40 Params: 01040000000000000000000000000000` every time I change the polling rate (which is the only thing the software can do for a razer mouse anyway). I also get stuttering if I set it over 1k polling, which it should go up to 8k and it is on a USB 3.2 port.

jkcdarunday, 2025-02-17:
Happens with my Razer Viper V3 Pro too

nikandlv, 2025-05-23:
Same issue

closed 2026-08-18 by github-actions[bot]

sigmonsays, 2026-11-03:
There is quite a few people with this problem. why is this the resolution to close it?

I've since uninstalled this daemon, it has done very little but cause issues, but i'd like to fix it..

z3ntu, 2026-11-14:
I think we can track this in #2122