openrazer/openrazer #2340: Cant install openrazer after system update

melodeu, 2024-11-04:
### Affected Device / PID

Razer deathadder v2

### Operating System / Version

Fedora 41

### Kernel Version

6.6.58-200.fc41.x86_64

### OpenRazer Version

lastest

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [X] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Hey so basically after i updated to fedora 41 everytime i try to install it says:

**Failed to resolve the transaction:
Problem: package openrazer-meta-3.9.0-1.1.noarch from hardware_razer requires python3-openrazer, but none of the providers can be installed
  - conflicting requests
  - nothing provides python(abi) = 3.12 needed by python3-openrazer-3.9.0-1.1.noarch from hardware_razer
  - nothing provides python3.12dist(dbus-python) >= 1.2 needed by python3-openrazer-3.9.0-1.1.noarch from hardware_razer
  - nothing provides python3.12dist(numpy) >= 1.11 needed by python3-openrazer-3.9.0-1.1.noarch from hardware_razer
You can try to add to command line:
  --skip-broken to skip uninstallable packages**

z3ntu, 2024-11-05:
Can you check in `/etc/yum.repos.d/` whether the entry for the hardware:razer repo points to Fedora 41 and not some older Fedora version?

melodeu, 2024-11-06:
how can i check that?

lah7, 2024-11-06:
Assuming GNOME desktop:
1. Open the "Files" app.
1. Press CTRL+L and go to `/etc/yum.repos.d/` 
2. Open the file `hardware:razer.repo`
3. Replace all text that says `Fedora_40` with `Fedora_$releasever`
4. Save (should prompt for authentication) and try updating the system again.

alichtman, 2024-11-09:
```
$ sudo dnf config-manager addrepo --from-repofile=https://openrazer.github.io/hardware:razer.repo

$ cat /etc/yum.repos.d/hardware:razer.repo
[hardware_razer]
name=hardware:razer (Fedora_40)
type=rpm-md
baseurl=https://download.opensuse.org/repositories/hardware:/razer/Fedora_40/
gpgcheck=1
gpgkey=https://download.opensuse.org/repositories/hardware:/razer/Fedora_40/repodata/repomd.xml.key
enabled=1
```

I'm running Fedora 41, so this is incorrect. 

To fix locally:

```
$ sudo sed -i s/40/41/ /etc/yum.repos.d/hardware:razer.repo

$ sudo dnf install openrazer-meta
Updating and loading repositories:
 hardware:razer (Fedora_41)                                                                                                                           100% |   4.8 KiB/s |   6.3 KiB |  00m01s
Repositories loaded.
Package                                                          Arch           Version                                                          Repository                               Size
Installing:
 openrazer-meta                                                  noarch         3.9.0-1.1                                                        hardware_razer                        0.0   B
Installing dependencies:
 openrazer-daemon                                                noarch         3.9.0-1.1                                                        hardware_razer                      974.2 KiB
 openrazer-kernel-modules-dkms                                   noarch         3.9.0-1.1                                                        hardware_razer                      760.3 KiB
 python3-daemonize                                               noarch         2.5.0-21.fc41                                                    fedora                               25.0 KiB
 python3-numpy                                                   x86_64         1:1.26.4-8.fc41                                                  fedora                               43.8 MiB
 python3-openrazer                                               noarch         3.9.0-1.1                                                        hardware_razer                      262.3 KiB
 python3-setproctitle                                            x86_64         1.3.3-2.fc41                                                     fedora                               39.3 KiB
 xautomation                                                     x86_64         1.09-11.fc41                                                     fedora                              183.5 KiB

Transaction Summary:
 Installing:         8 packages

Total size of inbound packages is 8 MiB. Need to download 8 MiB.
After this operation, 46 MiB extra will be used (install 46 MiB, remove 0 B).
Is this ok [y/N]: 
```

I'm not sure exactly what the long-term solution is though (i.e. something is reporting the wrong Fedora version number, but I'm not sure what)

z3ntu, 2024-11-09:
Did you remove the `.repo` file again before running the addrepo command? I assume if it already exists then it won't completely replace that file?

alichtman, 2024-11-09:
I didn't. The docs should likely mention that.

z3ntu, 2024-11-09:
The text on the website assumes you want to install openrazer when you haven't installed it before.

I'm actually more surprised that apparently `dnf config-manager addrepo` won't complain then if a different repo config has already been installed than what it's trying to install.

melodeu, 2024-11-10:
> Assuming GNOME desktop:
> 
> 1. Open the "Files" app.
> 2. Press CTRL+L and go to `/etc/yum.repos.d/`
> 3. Open the file `hardware:razer.repo`
> 4. Replace all text that says `Fedora_40` with `Fedora_$releasever`
> 5. Save (should prompt for authentication) and try updating the system again.
> 
> With `Fedora_$releasever`, you won't need to worry about this again on the next upgrade (#1981)

Thanks this worked!

closed 2024-11-17 by melodeu