openrazer/openrazer #2342: Support for Razer Basilisk V3 35K

ThePyromage, 2024-11-13:
### Device Name

Razer Basilisk V3 35K

### Product Number

RZ01-05230100

### Model Number

RZ01-0523

### VID:PID

1532:00cb

### USB Details

```shell
Bus 007 Device 003: ID 1532:00cb Razer USA, Ltd Razer Basilisk V3 35K
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00cb Razer Basilisk V3 35K
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk V3 35K
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

### Device Input Paths

```shell
lrwxrwxrwx 1 root root  10 Nov 13 13:02 usb-Razer_Razer_Basilisk_V3_35K-event-if01 -> ../event25
lrwxrwxrwx 1 root root   9 Nov 13 13:02 usb-Razer_Razer_Basilisk_V3_35K-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Nov 13 13:02 usb-Razer_Razer_Basilisk_V3_35K-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root  10 Nov 13 13:02 usb-Razer_Razer_Basilisk_V3_35K-if02-event-kbd -> ../event26
lrwxrwxrwx 1 root root   9 Nov 13 13:02 usb-Razer_Razer_Basilisk_V3_35K-mouse -> ../mouse0
```

z3ntu, 2024-12-07:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

ThePyromage, 2024-12-11:
This is for the non-pro version, not a duplicate.

lexicalbit, 2024-12-25:
I opened a PR intended to add support for this: #2363

PR #2363 (2025-02-16): Feature: Add support for Razer Basilisk V3 35K (non-Pro version)
Note that this is a different device from the "Pro" version, as it is only available as a wired mouse and carries a different PID.

Fixes #2342
closed 2025-02-16 by z3ntu