openrazer/openrazer #2344: openSUSE Tumbleweed: dkms not auto installing after kernel update.

ad-on-is, 2024-11-15:
Each time there's a kernel update in openSUSE Tumbleweed, I have to run `sudo dkms install openrazer-driver/<version> manually.

IIRC, when I was on Fedora, this used to be automatically.

z3ntu, 2024-12-07:
The .spec is the same for both and for dkms we don't have different codepaths there. We only have a "dkms install" and "dkms remove" on install and uninstall but nothing for upgrades.
https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec#L154-L171

As I'm not really familiar with openSUSE, do you know of a different package which handles upgrades correctly? Would be great to have a reference how that's supposed to work.

ad-on-is, 2024-12-07:
Hmmm... If there was never an upgrade script, then my perception of how things behaved in Fedora might be twisted, probably due to less frequent kernel updates. 

I'm not familiar with OpenSUSE either. But one thing that comes to mind, maybe VirtualBox. Again, IIRC, when there are updates I see some postscript stuff regarding virtualbox. But that's just out of my head, haven't done any research.

z3ntu, 2024-12-07:
Some distributions, like Arch Linux have a trigger to rebuild DKMS modules without the package having to do anything which works pretty reliably on my experience https://gitlab.archlinux.org/archlinux/packaging/packages/dkms/-/blob/main/hook.install?ref_type=heads

Many other distros are a random mess with dkms modules.

also zfs-on-linux packaging should be some reference if you can find something.

hannemann, 2025-02-01:
I once had a similar problem after switching to Tumbleweed and found out that i had to enable the dkms.service. Just asking... ;)

Asherathe, 2025-02-02:
> I once had a similar problem after switching to Tumbleweed and found out that i had to enable the dkms.service. Just asking... ;)

I never knew such a thing existed. Amazing.

z3ntu, 2025-02-15:
Is this resolved by enabling `dkms.service`?

ad-on-is, 2025-02-15:
> Is this resolved by enabling `dkms.service`?

I think it did, as since I haven't had any issues yet.

closed 2025-02-15 by ad-on-is

z3ntu, 2025-02-15:
Ok, good to hear! I guess it's a good idea to add to our installation page that people run `sudo systemctl enable --now dkms.service` open openSUSE, I imagine that command should be sufficient?

ad-on-is, 2025-02-15:
yes... that's the command. But, before you add it to the docs, do you maybe want me to make absolutely sure, that it works as expected?

I recall, after enabling the service, running an update, and rebooting, I had issues where it didn't work. So I ran dkms install, but it showed that it's already installed. I then rebooted again, and it worked.

However, a recent update came with a new kernel, and everything worked right after a single reboot.

So just to make sure that it was a single edge case where it didn't work, I'd like to keep an eye on it and report back, once I'm 100% certain.

z3ntu, 2025-02-15:
@ad-on-is Ok, I've opened a Draft PR for the website so we don't completely forget. Let me know what you find out!

ad-on-is, 2025-02-20:
@z3ntu There was a kernel update yesterday and I did some observation.

while the dkms.service build for openrazer worked automatically, I had to restart my PC twice for it to take effect.

z3ntu, 2025-02-20:
Okay, good to hear that it's working for the most part at least!

Maybe you could check with the openSUSE people if they can improve the service to make it work without the extra round of reboot that's needed now.