openrazer/openrazer #2345: Layout request for Swiss-French keyboard

FairplexVR, 2024-11-16:
### Affected Device / PID

Razer BlackWidow V3 Pro (Wired) / 1532:025A

### Operating System / Version

Arch Linux

### Kernel Version

6.11.8-arch1-2

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I have this error when I try to set a custom layout:

You are using a keyboard layout that is not recognized by the daemon. Please help us by visiting https://github.com/openrazer/openrazer/wiki/Keyboard-layouts. For now, a generic device will be used.

My physical keyboard layout is 0F. While it's the same number as for swiss-german, the keys are not the same on both.

z3ntu, 2024-12-07:
First, it's interesting that you get the message because since commit 2b88ad654e90d9b8af05e8d70d5e6ef9d001f7f7 `0F` is recognized

Could you run the following commands, to figure out what the daemon is reporting in the first place?
```
# get the serial number
qdbus org.razer /org/razer razer.devices.getDevices
# replace "XX0000000203" with your serial number
qdbus org.razer /org/razer/device/XX0000000203 razer.device.misc.getKeyboardLayout
```

Also if the keyboard is reporting the same layout as Swiss German, then apart from if you find a different property on the device that can differentiate Swiss French with Swiss German I'm not sure there's too much we can do.

FairplexVR, 2024-12-08:
Apparently its unknown

`[fairplex@archlinux ~]$ qdbus org.razer /org/razer razer.devices.getDevices
IO2048F47000530
[fairplex@archlinux ~]$ qdbus org.razer /org/razer/device/IO2048F47000530 razer.device.misc.getKeyboardLayout
unknown
[fairplex@archlinux ~]$ 
`

> Also if the keyboard is reporting the same layout as Swiss German, then apart from if you find a different property on the device that can differentiate Swiss French with Swiss German I'm not sure there's too much we can do.

I think they use the same keys but there is a difference on some characters french vs german for the same keys: è > ë , é > ö, à >ä.

z3ntu, 2024-12-08:
What's the output of `cat /sys/bus/hid/drivers/razerkbd/0*/kbd_layout` then?

FairplexVR, 2024-12-08:
```
[fairplex@archlinux ~]$ cat /sys/bus/hid/drivers/razerkbd/0*/kbd_layout
0f
```

z3ntu, 2024-12-08:
Oh, hm lowercases 0f?
I wonder how this worked for Turkish and Japanese ever, can't imagine we changed something there in the kernel driver?

z3ntu, 2024-12-09:
Could you check with #2355 if at least the lookup start working?

FairplexVR, 2024-12-13:
I am not sure how to proceed. I tried to switch to the branch layoutids-hex and used "make driver", then I used  
`rmmod razerkbd && insmod razerkbd.ko`
But openrazer fails to load after this. If I reboot I still have the same issue. Can you guide me how can I test it correctly?

z3ntu, 2024-12-14:
See https://github.com/openrazer/openrazer/wiki/Building-a-package , does that help?

FairplexVR, 2025-02-15:
https://github.com/user-attachments/assets/ca4a65cd-ea43-428c-ad01-3e158fbdff54

Here is the problem in video. The keys seem to have an offset, and some do not light up.

z3ntu, 2025-02-15:
I think that's just a Razer BlackWidow V3 Pro quirk then compared to other keyboards with the same "matrix dimensions" - which is the only metric currently used in RazerGenie to display different layouts.

For now you could hold Shift (or Ctrl, I don't remember) while clicking "Open custom editor" which should give you a view of all the possible LED locations.

closed 2025-07-16 by github-actions[bot]