openrazer/openrazer #2349: Razer Basilisk V3 - Mousewheel suddenly disappeared

Malaccoda, 2024-11-23:
### Affected Device / PID

Razer Basilisk V3 1532:0099 Firmware 1.20

### Operating System / Version

Arch / Wayland / Hyprland

### Kernel Version

6.11.9-arch1-1

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

About a month ago I bought the Basilisk V3. There were already from start some minor issues with a lagging mouse wheel and debugging showed a massive latency up to several seconds.  After I updated the mouse to the latest firmware 1.20 the lags didnt occure anymore. I played a bit with polychromatic but did not change anything permanently.  After that everything seemed to work fine until two days ago when suddenly the mousewheel stopped working. 

Because I'm currently migrating my desktop environment to hyprland I expected anyway some instabilities and my first hope was that a reboot would may be fix the issue. After that didnt work I checked if there was a some hidding binding I accidentaly trigger or if polychromatic / openrazor provides a function to deactivate the function of the mouse wheel. It seemed at least possible because the buttons, LEDs and even tactility settings just worked fine and the logs seemed not  to indicate a major problem. But I couldnt find a function like "deactivate mousewheel" and my next try was to reset the mouse via the windows synapse software. 

So I plugged the mouse on the windows machine and was surprised to see that the wheel after "factory resetting" and running on windows didnt reactivate itself. Synapse meanwhile just tells me that the mouse works fine and because the firmware is already up to date the razer update exe refused to install the latest firmware. More out of curiosity I tried to map some makros and functions but synapse was still conviced that nothing was wrong. Unfortunately that makes myself the more concerned about the security concept. On an Input device capable of holding enough memory for makros I assumed at least an immutable firmware part which can be veriefed with a checksum and a writable section for the config which I can inspect or eraze myself.  A wizard lying about a factory reset while incapable of tracing persisting code changes isn't really helpful at all. 

Anyway, after switching back to linux and began to dig a bit deeper. A test via libinput confirmed, that wayland doesnt receive any input from the mouse wheel.  And after consulting lsusb a partial failure to initialize all "subdevices" of the "mousehub" and the openrazer wiki brought some insight about the driver settings. Not brave enough to root cat globs through /sys/ i cowardly decided to check the settings via (g)dbus. Altought not hundred percent sure if that checks the driver or device setting, because the blackwidow keyboard returns to getDeviceMode a ( '3:0' , ) which correlates to the log messages that the keyboard was set to Driver Mode, I assume the ( '0:0', ) for the mouse is also accurate. 

Still without any concrete solution after a closer investagion on the logs the following messages attracted my attention.  
 
```
[28561.000173 <    0.000000 >] kernel: razermouse 0003:1532:0099.0010: No inputs registered, leaving
[28561.000827 <    0.000000 >] kernel: razermouse 0003:1532:0099.0010: hidraw5: USB HID v1.11 Device [Razer Razer Basilisk V3] on usb-0000:00:14.0-1/input3
```
At first glance I thought the it's some race condition or the like and fixed by the subsequent message but a shy just look dont touch ls to the corresponding /sys/bus/hid/drivers/razermouse shows an input folder for the buses ...000D/E/F but none for the one ending 0010. I've little understanding of USB and this endpoint doesnt need an input channel but simple minded as I am I found it plausible if its dedicated to the mousewheel in a effort to prevent flooding and reduce latency the button channel while the third channel goes to the LEDs. 

Still, even  if im not completely wrong, I have no idea what caused this mismatch and made it persistent on the mouse itself.  I could also well imagine that some third party config, the firmware update / reset on windows or even some bug or misconfig in hyprland.  But i thought best chances are to ask here as the issue occures independent from the software / os environment.   

So far here are the journal entries for razor in reversed order.  If any further information needed like a pcap stream I'm glad to help. Experienced feedback or hints are also welcome.  

``` journalctl -r
[29909.679967 <    0.000000 >] systemd-logind[559]: Watching system buttons on /dev/input/event8 (Razer Razer BlackWidow V4 X)
[29909.672046 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1>
[29909.662560 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-13/1->
[29909.655179 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1>
[29909.648364 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-13/1->
[29909.644850 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9>
[29909.643331 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5>
[29909.641487 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1>
[29909.639792 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9>
[29909.634877 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1>
[29909.633294 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5>
[29909.631110 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5>
[29909.630024 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:35:28 | razer                          | DEBUG    | Device event [change]: /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5>
[28569.667002 <    0.000000 >] openrazer-daemon[1714]: 2024-11-23 06:13:08 | razer                          | DEBUG    | Writing persistence config
[... 105 lines cut ...]
[28561.039086 <    0.000000 >] razer_mount[23905]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0099.000E/
[28561.037792 <    0.000000 >] razer_mount[23904]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0099.0010/
[28561.032991 <    0.000000 >] razer_mount[23899]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0099.000F/
[28561.032613 <    0.000000 >] razer_mount[23898]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0099.000E/
[28561.032556 <    0.000000 >] razer_mount[23897]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0099.000D/
[28561.032196 <    0.000000 >] razer_mount[23896]: Driver razermouse
[28561.029680 <    0.000000 >] razer_mount[23892]: Driver razermouse
[28561.029241 <    0.000000 >] razer_mount[23891]: Driver razermouse
[... cut ...]
[28560.998801 <    0.000000 >] kernel: input: Razer Razer Basilisk V3 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:0099.000E/input/input32
[28560.998158 <    0.000000 >] kernel: razermouse 0003:1532:0099.000D: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Basilisk V3] on usb-0000:00:14.0-1/input0
[28560.998084 <    0.000000 >] kernel: input: Razer Razer Basilisk V3 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:0099.000D/input/input31
[28560.997576 <    0.000000 >] kernel: usb 1-1: Manufacturer: Razer
[28560.996985 <    0.000000 >] kernel: usb 1-1: Product: Razer Basilisk V3
```

lah7, 2024-11-23:
Some things to try:

1. Does replugging the mouse help? What about in different ports (especially any USB 2.0 vs 3.0)?
2. Are you able to use the mouse wheel in the BIOS/UEFI settings? 
   - _Independent from any OS -- depends if yours has a fancy UEFI with mouse support._
3. Try booting a live Linux environment (CD/ISO) and/or Windows Setup (DVD/USB) and see if it works as a generic mouse? 
   - _No driver interference, also suggest keeping unplugged until after boot so the mouse has completely power cycled._

It could be a hardware issue. I had a Razer Mamba TE (a year or so ago) that had erratic scrolling, and I opened it up to clean and re-solder a loose connection... **twice!** It was out of warranty, of course.

----

>  After that everything seemed to work fine until two days ago when suddenly the mousewheel stopped working.

Has there been a kernel update recently? As you're on Arch:

    cat /var/log/pacman.log | grep upgraded | grep linux | tail

There was an update to `libinput (1.26.2-1 -> 1.27.0-1)` 3 days ago:

    cat /var/log/pacman.log | grep upgraded | grep input | tail

You _could_ try [downgrading](https://wiki.archlinux.org/title/Arch_Linux_Archive) that package, or the kernel, or roll the system back a few days ago  **(if you know what you're doing in case of a system breakage)** - but testing a live environment with an older version should rule this out anyway.

Malaccoda, 2025-01-09:
Sorry for not answered faster. I dont think it has something to do with the os and it is certainly not a hardware defect. I captured the onnection now with wireguard and it confirms my assumption from my first description. When I plug in the mouse the scrollwheel, for a really short moment it submits something (and the scrollbar scrolls one tick) and then stops. I I attached i the according pcapng as base64 encoded txt (gh doesnt seem to allow pcapng as attachements) with the relevant part from second 95 to 108 (I tried the wheel several times after but nothing was submitted). 
[raverv3pcapng.txt](https://github.com/user-attachments/files/18358756/raverv3pcapng.txt)
To reconvert it to a pcapng file you can use
`` cat raverv3pcapng.txt | base64 -d > raverv3.pcapng ``

z3ntu, 2025-02-15:
Not sure if you tried this before, but if the problem still occurs after uninstalling OpenRazer, then it's not something we can help with unfortunately.

closed 2025-03-07 by Malaccoda

lah7, 2025-03-07:
It's all part of the troubleshooting process. From what you've given:

> About a month ago I bought the Basilisk V3. There were already from start some minor issues with a lagging mouse wheel and debugging showed a massive latency up to several seconds. 

Things already started off as a hardware/firmware issue.

> After I updated the mouse to the latest firmware 1.20 the lags didnt occure anymore.

Firmware issue fixed! Or, in the worst case, the update masked an underlying issue.

>  After that everything seemed to work fine until two days ago when suddenly the mousewheel stopped working.

Could be software, perhaps an update broke something. **OpenRazer didn't have a new release at the time** so it **could** have been due to kernel or input related packages like `libinput` (the latter was coincidently around the same time when you said it stopped working)

> So I plugged the mouse on the windows machine and was surprised to see that the wheel after "factory resetting" and running on windows didnt reactivate itself. 

Possibly firmware issue again. Razer should know their USB commands like a book.

Of course, it's possible (and extremely rare) for OpenRazer to cause _buggy_ Razer firmware to get itself into a weird state from a command it didn't expect or understand. This happened before with this Basilisk V3 when a very specific hardware effect "Off" was set (#1844, #1767), but nothing more then a replug or reset would fix. Once we knew that, we cared to stop it from happening again: 606ed01726be453b662f4980c211dc3d10f1eb94

Never have we seen firmware just become corrupt. We rely on contributors to thoroughly test their contributions. Many Razer hardware share identical protocols anyway... and OpenRazer isn't doing anything for the actual scroll event (similar to a left or right click), that's just generic mouse stuff.

If the broken scroll wheel was truly a bug caused by our code, many more would have complained their Basilisk V3 mouse wheel stops working after installing OpenRazer and we would need to know what specifically caused it.

> A test via libinput confirmed, that wayland doesnt receive any input from the mouse wheel. 
> 
> I dont think it has something to do with the os and it is certainly not a hardware defect. [...]  When I plug in the mouse the scrollwheel, for a really short moment it submits something (and the scrollbar scrolls one tick) and then stops.

To be honest, it sounds like a hardware issue. If it's under warranty, get it replaced.

If not, and you're comfortable with hardware, I'd carefully open it up, clean the wheel sensor and check/repair any wires or parts if they've become loose. You could also test it while opened up to see how the sensor works.

Seems you're not alone with this scroll problem:

* https://insider.razer.com/razer-support-45/brand-new-razer-basilisk-v3-scroll-wheel-not-working-60435
* https://www.reddit.com/r/razer/comments/x5mk9i/basilisk_v3_scroll_wheel_issue/
* https://www.ifixit.com/Wiki/Razer_Basilisk_V3_Troubleshooting#Section_Unresponsive_Scroll_Wheel

I've had a Mamba TE with a scroll wheel that went up/down erratically. Re-soldering loose wires for the scroll wheel fixed the problem... **twice**... because it happened again months later. I haven't bought a Razer mouse ever since.

Like @z3ntu said, if it happens outside of OpenRazer, then it's not a bug we can fix. Unless you can **prove** that something in OpenRazer driver (a mode, a command) started the problem.