openrazer/openrazer #2354: Support for Razer Deathadder V3 Hyperspeed

secretwolfie, 2024-12-08:
### Device Name

Razer Deathadder V3 Hyperspeed

### Product Number

RZ01-05140100

### Model Number

RZ01-0514

### VID:PID

1532:00c5

### USB Details

```shell
Bus 001 Device 002: ID 1532:00c5 Razer USA, Ltd Razer DeathAdder V3 HyperSpeed
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00c5 Razer DeathAdder V3 HyperSpeed
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V3 HyperSpeed
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-event-if01 -> ../event8
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-event-mouse -> ../event4
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-if01-event-kbd -> ../event6
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-if01-event-mouse -> ../event7
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-if02-event-kbd -> ../event9
lrwxrwxrwx. 1 root root   9 Dec  8 10:53 usb-Razer_Razer_DeathAdder_V3_HyperSpeed_000000000000-mouse -> ../mouse0
```

### USB HID Dump

```shell
001:002:002:DESCRIPTOR         1733675388.090662
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:002:001:DESCRIPTOR         1733675388.097659
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:002:000:DESCRIPTOR         1733675388.101661
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 01 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

z3ntu, 2024-12-09:
Duplicate of #2294

z3ntu, 2024-12-09:
Just a well-meant question, we have this section in the issue template:

> ## Before you continue...
> 
> [🔍 Search for existing issues and pull requests](https://github.com/openrazer/openrazer/issues?q=label%3A%22Device+Support%22) by searching for your USB PID. Please avoid creating duplicates.
> 
> You can get your USB PID by running `lsusb` in a terminal:
> 
> ```shell
> lsusb | grep 1532
> ```
> 
> For example, if you see `1532:0203`, your device PID is `0203`.
> 
> Your device may have different PIDs depending if it's wired or wireless, so check those too.
> 
> If the issue is closed, or the pull request is merged (purple), your device is supported in the next release. [Check the comparison](https://github.com/openrazer/openrazer/compare/stable...master) to see if that's the case!

Was this not clear, or in other words did you follow this and didn't find the other issue, and could we improve the text so that other people don't spend effort in filling out the template when it's just a duplicate?

closed 2024-12-09 by z3ntu