openrazer/openrazer #2358: Naga X "back" button not working in 3.9.0 - worked in 3.8.0

justinnichols, 2024-12-10:
### Affected Device / PID

Razer Naga X, 1532:0096

### Operating System / Version

Arch Linux

### Kernel Version

6.6.64-1-lts and 6.12.3.zen1-1

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

In a previous install of the same operating system I am using (Garuda Linux, an Arch Linux distribution), with openrazer 3.8.0, the "Back" button for the Razer Naga X (the one that sits just back from the mouse wheel) would register as an XF86Tools input and would indeed allow the use of navigating back in a browser, and allow remapping for other purposes.

I have since reinstalled Garuda Linux and between my last install and now, openrazer 3.9.0 has released.  Now the "back" button does nothing, no input is detected at all.

I have tried xev (on X11), evtest, and wev (on wayland) and none of them register anything when I press the "back" button.

I have performed all of the following troubleshooting steps:

* Tried the linux-lts kernel
* Verified my user is in the plugdev group (and openrazer-daemon runs just fine)
* Uninstalled and reinstalled openrazer-driver-dkms and openrazer-daemon
* Verified DKMS properly built for the two kernel versions I have installed (via troubleshooting page)
* Rebooted numerous times
* Plugged mouse directly into motherboard USB (instead of via a USB hub)
* Re-plugged mouse directly into motherboard USB
* Tried on X11
* Tried on Wayland

Since this is an Arch Linux distribution, it is difficult for me to attempt to downgrade to 3.8.0 without causing potential package conflicts, in order to verify the button works still with 3.8.0 at this present time.  I had been using the button just fine in my Windows install just prior to this reinstall of Garuda Linux (thus, the button is not broken).

Would love some insight here on how to possibly get this button to work again, if possible, or confirmation that it is indeed a bug with 3.9.0.  I like using it as my push-to-talk key in Discord.

lah7, 2024-12-10:
Looking at the changes [between v3.8.0 and v3.9.0](https://github.com/openrazer/openrazer/compare/v3.8.0...v3.9.0), I can't see anything specific to the Naga or mouse buttons. It could well be the kernel, especially since it's a fresh install and the previous installation potentially may have had some older configuration.

You can give v3.8.0 a try, as long as the versions for the OpenRazer components are matching. Since it's distributed as a DKMS module, it'll rebuild that version of the driver against the kernel you're running now.

Download old copies here:

* https://archive.archlinux.org/packages/o/openrazer-daemon/
* https://archive.archlinux.org/packages/o/openrazer-driver-dkms/
* https://archive.archlinux.org/packages/p/python-openrazer/

Then install them:

    sudo pacman -U *.pkg.zst

Reboot, and let's see if the bug persists. 

* If it **works** as expected again, we could try a `git bisect` to find out when it broke. 
* If it **doesn't** work in 3.8.0 either, it's just coincidence, but likely the kernel version or another package related to input. 
  * You can also downgrade kernels too for testing, as long as you also downgrade any driver packages built for that specific kernel, (e.g. `nvidia-lts`) but DKMS modules like  `nvidia-dkms` should be OK - but be prepared if you know how to recover/chroot and upgrade everything **if** it fails to boot.
  * Downgrading libraries like `libinput` may break the desktop, it would be safer to [roll back the system](https://wiki.archlinux.org/title/Arch_Linux_Archive#How_to_restore_all_packages_to_a_specific_date) to an earlier time if you want.

justinnichols, 2024-12-14:
@lah7 **Thank you for your assistance**.  I attempted to roll back to v3.8.0 the way you suggested, it rolled back just fine.  However, as you expected, it did not solve the problem.  So my guess at this point is that something in the kernel changed, or something in libinput changed.  I tried the LTS kernel (6.6.x) and it did not solve the issue.

I decided to work around the problem and just use a different mouse for now.  I'll keep checking back with the Razer Naga X from time to time to see if it somehow gets fixed with some kind of update.  The mouse I decided to use is also fully supported with ratbag and piper, so that will make managing it much easier anyway.  I adore my Naga X, and would love to continue using it, but I really want to have PTT in discord with the specific mouse button I am used to.

I will close this ticket since it is not at all the problem of openrazer (well, at least that I can tell at this time, anyway).

justinnichols, 2024-12-14:
Closing as not currently an issue specific to openrazer.

closed 2024-12-14 by justinnichols