openrazer/openrazer #2359: RuntimeError: "Should never be installed" when installing OpenRazer Python library

Dralexgon, 2024-12-10:
Hi! 👋

I'm trying to install the OpenRazer Python library as described in the [examples](https://github.com/openrazer/openrazer/tree/master/examples) using the command:

`pip install openrazer`

However, I encounter the following error:  
`RuntimeError: Should never be installed`

I couldn't find any documentation for the library other than these examples. Could you provide guidance or point me to the correct way to install and use the library?

Thank you in advance!

Dralexgon, 2024-12-10:
After ~4 hours of search I FOUND THE ISSUE
The library is installed with openrazer (that's why I wasn't able to found how to install it) BUT it isn't added to the venv
Use `python3 -m venv --system-site-packages path/to/venv`

Dralexgon, 2024-12-10:
[resolved]

After ~4 hours of search I FOUND THE ISSUE
The library is installed with openrazer (that's why I wasn't able to found how to install it) BUT it isn't added to the venv
Use `python3 -m venv --system-site-packages path/to/venv`

Dralexgon, 2024-12-10:
[Resolved]

To use openrazer lib in venv, you need to link your system package to your venv package.
I made this script if anyone has the same issue.

`
#Change these variables to match your system
VENV_PATH=venv
PYTHON_VERSION=3.12

# This is the path where the packages are installed
PACKAGES_PATH=$(python -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())")
VENV_PACKAGES_PATH="$VENV_PATH"/lib/python"$PYTHON_VERSION"/site-packages

python3 -m venv "$VENV_PATH"
source "$VENV_PATH"/bin/activate

ln -s "$PACKAGES_PATH"/* "$VENV_PACKAGES_PATH"/

pip install -r requirements.txt#optional
`

lah7, 2024-12-10:
Hi. I'm pretty sure the pip package published for `openrazer` and `openrazer-daemon` is just a stub to prevent the name from being squatted for malicious intent.

The examples were designed to "just work" with the system libraries, assuming `python3-openrazer` is installed for your distro. e.g.

    python3 ./examples/basic_effect.py

It may work in a venv by symlinking the modules from the system, but OpenRazer doesn't support this kind of isolated virtual environment. OpenRazer's architecture was designed to work within the host. Messages get sent to the daemon via D-Bus (more venv dependencies may need to be installed) in which the daemon processes/relays to the driver (via a sysfs filesystem) or the [fake driver](https://github.com/openrazer/openrazer/blob/master/scripts/setup_fake_devices.sh).