openrazer/openrazer #2361: RGB not turning off with computer

Matheos96, 2024-12-22:
### Affected Device / PID

Razer Ornata V2

### Operating System / Version

Fedora 41

### Kernel Version

6.12.5-200

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [X] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I am having trouble figuring this one out. I have openrazer installed with PolyChromatic and it seems to work great, except for one thing. When I turn off my PC (Fedora 41) the lights does not turn off on the keyboard at all. Not even after the computer has gone to sleep completely. This is also true for my mouse to be fair, which is a corsair one...

That being said, these do turn off when I shut down from Windows. I am not sure if Windows somehow tells them to turn off the lights or how it works. There is probably some BIOS setting for USB power, but I think it is handy to be able to charge my headset even with my PC off, so I would rather not disable that.

razer.log
```
2024-12-22 11:30:37 | razer                          | INFO     | Initialising Daemon (v3.9.0). Pid: 3199
2024-12-22 11:30:37 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-12-22 11:30:37 | razer                          | INFO     | Found device.0: 0003:1532:025D.000C
2024-12-22 11:30:37 | razer.device0                  | INFO     | Initialising device.0 RazerOrnataV2
2024-12-22 11:30:37 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-12-22 11:30:37 | razer                          | INFO     | Serving DBus
2024-12-22 11:43:11 | razer                          | INFO     | Stopping daemon on signal 15
2024-12-22 11:45:06 | razer                          | INFO     | Initialising Daemon (v3.9.0). Pid: 3211
2024-12-22 11:45:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-12-22 11:45:06 | razer                          | INFO     | Found device.0: 0003:1532:025D.000D
2024-12-22 11:45:06 | razer.device0                  | INFO     | Initialising device.0 RazerOrnataV2
2024-12-22 11:45:06 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-12-22 11:45:06 | razer                          | INFO     | Serving DBus

```

razer.conf

```
[General]
# Verbose logging (logs debug messages - lotsa spam)
verbose_logging = False

[Startup]
# Set the sync effects flag to true so any assignment of effects will work across devices
sync_effects_enabled = True

# Turn off the devices when the systems screensaver kicks in
devices_off_on_screensaver = True

# Battery notifier
battery_notifier = True

# Battery notification frequency [s] (0 to disable)
battery_notifier_freq = 600

# Battery notifications appear when device reaches this percentage
battery_notifier_percent = 33

# Apply effects saved to disk when daemon starts
restore_persistence = False

# Enable this if you dual boot with Windows and the effect isn't restored. Requires restore_persistence = True
persistence_dual_boot_quirk = False
```
persistence.conf

```
[SI2038F39704317]
backlight_active = True
backlight_brightness = 100
backlight_effect = static
backlight_colors = 255 0 0 0 255 255 0 0 255
backlight_speed = 1
backlight_wave_dir = 2
```

As you see, I only try to have a static color even.. Any ideas? Could ofc be Linux related in general as it happens for my mouse too. I do think Fedora shut down really shuts down though, not hibernate, so not sure where to look next...

lah7, 2024-12-22:
Seems very likely to be a BIOS/UEFI setting. Have a look in the DEL/F2 setup and poke around the power management, ACPI and USB options. S3-S4 modes are sleep, S5 is the shutdown state. There might be something to configure S5 behaviour.

What motherboard do you have? Might be worth confirming with the manual if specific ports are always on for charging (assuming a desktop).

My keyboard/mouse are plugged into the monitor (acting as a powered USB hub), and they always turn on/off following the system state. I have **ACPI → "USB Keyboard/Remote Power On"** enabled so I can wake/power the system with a keypress. You could see if that setting helps for your board.

Matheos96, 2024-12-22:
> s very likely to be a BIOS/UEFI setting. Have a look in the DEL/F2 setu

I will have a look! My mobo is a an Asus Prime X570-Pro board.
Manual here: https://dlcdnets.asus.com/pub/ASUS/mb/SocketAM4/PRIME_X570-PRO/E17432_PRIME_X570-PRO_UM_v3_web.pdf?model=PRIME%20X570-PRO

I did not exactly directly find anything useful in the manual... I found this 
[image]

which talks about a BIOS setting. This could help, but then it would probably also prevent my devices from charging when the computer is OFF (like my headset, as mentioned in OP). It would be interesting to understand how windows does this, or why it even makes a difference, am I turning off the PC from windows or Linux...

lah7, 2024-12-23:
You could also try turning on **ErP Ready**, it mentions about switching off some power at S4+S5 states, so you _might_ be able to keep _"USB power delivery in Soft Off state"_ on too. The manual doesn't mention **Deep Sleep** but it might be worth a try if you see this option. Also worth checking for any BIOS updates.

Out of curiosity, what happens if you press the power button (once) while in the DEL/F2 setup? All lights off (like Windows) or kept on (like Linux)?

In either case, it's _probably_ to do with ACPI. If you want to test something, add `acpi=off` to the kernel parameters and boot Linux, it'll temporarily turn off power management features. Mileage may vary since some things (like graphics) may not work properly. Observe what happens when shutting down with ACPI disabled (BIOS in control). An MSI board once gave me grief because of its buggy ACPI implementation (errors were seen in `sudo dmesg | grep ACPI`), so I gave it up for a different board instead :)

While not an OpenRazer issue, I imagine this topic will come up in future, so I'm keen to know what works for your scenario.

Matheos96, 2024-12-30:
Sorry about the delayed reply, I was away for a week on vacation.

I tried searching a bit in BIOS and noticed that the ErP **seemed** to help. Unfortunately though, for the reasons I feared... Turning on ErP Ready either with "Enable (S4+S5)" or "Enable (S5)" both **forces** _USB power delivery in Soft Off State (S5)_ to **disabled**. So yes, it helps, and technically the _USB power delivery in Soft Off State (S5)_ --> **disabled** is what helps (**ErP Ready** does not make a difference really), BUT this also of course prevents me from charging my headset when my PC is completely OFF (S5). I generally always turn it off completely over night, and that is when I would leave stuff to charge...

> Out of curiosity, what happens if you press the power button (once) while in the DEL/F2 setup? All lights off (like Windows) or kept on (like Linux)?

In BIOS, if I press the power button once, the computer turns off completely and the keyboard goes completely black (along with the mouse). In other words, it acts the same as windows.

I am yet to try kernel parameters but this is what I have got so far... Does the Windows-side driver somehow tell USB peripherals that they can turn off their lighting, or what is going on really. Or actually, since the same behaviour happens when turning off from BIOS, maybe Linux is overriding some default behaviour? Just throwing stupid theories around... :)

lah7, 2024-12-30:
> Does the Windows-side driver somehow tell USB peripherals that they can turn off their lighting

With Razer devices, there are [two modes of operation](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver#device_mode---set-device-mode): "Normal mode" and "Driver mode". Starting `openrazer-daemon` puts the device into driver mode (0x03). It should be back to normal mode (0x00) on graceful close. 

The BIOS will likely be in normal mode (generic keyboard). Quite possible the Windows driver put it back into "normal" mode too. This doesn't really answer for the Corsair mouse though.

Just in case when you shut down Fedora (GNOME/KDE?) and the `openrazer-daemon` process is **killed** at logout, you could try manually setting the mode and see if that makes a difference?

    echo -n -e "\x00\x00" > /sys/bus/hid/drivers/razerkbd/**/device_mode

So, stop `openrazer-daemon`, run above (to be sure) and then shut down to see if that works.

lah7, 2024-12-30:
Regarding [kernel parameters](https://www.kernel.org/doc/html/latest/admin-guide/kernel-parameters.html), try setting `acpi_os_name` to spoof as Windows. It is an unconfirmed theory, but I think ACPI has a role in this. The list of names can be found [in this source file.](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/acpi/acpica/utosi.c#n40)

So, _maybe_, the BIOS might behave differently if it thinks the OS was Windows (or earlier):

    acpi_os_name="Windows 2022"

Matheos96, 2024-12-30:
> ```
> echo -n -e "\x00\x00" > /sys/bus/hid/drivers/razerkbd/**/device_mode
> ```
> 
> So, stop `openrazer-daemon`, run above (to be sure) and then shut down to see if that makes a difference.

I tried shutting down the daemon followed by a shutdown. It did not help. Also, it seems like the device_mode file is not found at all for me (so I skipped that part)... Not sure why

```
echo -n -e "\x00\x00" > /sys/bus/hid/drivers/razerkbd/**/device_mode
zsh: no matches found: /sys/bus/hid/drivers/razerkbd/**/device_mode
```

Tried looking with grep...
```
sudo grep -r device_mode /sys/bus/hid/drivers/razerkbd
[sudo] password for matheos: 
grep: /sys/bus/hid/drivers/razerkbd/uevent: Permission denied
grep: /sys/bus/hid/drivers/razerkbd/bind: Permission denied
grep: /sys/bus/hid/drivers/razerkbd/new_id: Permission denied
grep: /sys/bus/hid/drivers/razerkbd/unbind: Permission denied
```
Either way, I noticed an error when running `openrazer-daemon -s` and checking `systemctl --user status openrazor-daemon.service`

This command actually killed by main enter button for some reason, as you see (empty enters), it seems like the enter button is infinitely pressed and I had to kill the terminal after this...
```
openrazer-daemon -s
Module psutil is missing, not checking for process name
Process 3206 stopped
❯ 
❯ 
❯ 
...
```
this indicates something went wrong...

```
systemctl --user status openrazer-daemon.service
○ openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead)
       Docs: man:openrazer-daemon(8)

Dec 30 22:58:42 fedora openrazer-daemon[3206]: 2024-12-30 22:58:42 | razer.device0                  | INFO     | Initialising device.0 RazerOrnataV2
Dec 30 22:58:42 fedora openrazer-daemon[3206]: 2024-12-30 22:58:42 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
Dec 30 22:58:42 fedora openrazer-daemon[3206]: 2024-12-30 22:58:42 | razer                          | INFO     | Serving DBus
Dec 30 23:02:57 fedora openrazer-daemon[3206]: 2024-12-30 23:02:57 | razer                          | INFO     | Stopping daemon on signal 15
Dec 30 23:02:57 fedora openrazer-daemon[3206]: Stopping daemon.
Dec 30 23:02:57 fedora openrazer-daemon[3206]: Exception ignored in atexit callback <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7f77be6f0050>>:
Dec 30 23:02:57 fedora openrazer-daemon[3206]: Traceback (most recent call last):
Dec 30 23:02:57 fedora openrazer-daemon[3206]:   File "/usr/lib/python3.13/site-packages/daemonize.py", line 72, in exit
Dec 30 23:02:57 fedora openrazer-daemon[3206]:     sys.exit(0)
Dec 30 23:02:57 fedora openrazer-daemon[3206]: SystemExit: 0
```

> So, _maybe_, the BIOS might behave differently if it thinks the OS was Windows (or earlier):
> 
> ```
> acpi_os_name="Windows 2022"
> ```

I also tried adding this kernel parameter temporarily on boot by editing in grub (using `e`) but it made no difference when shutting down right after login. Could it be that the daemon shut down is not graceful for me, and the device is left in driver mode...

lah7, 2024-12-31:
That's odd, the sysfs structure should be there. Can you manually check in the file browser?

    /sys/bus/hid/drivers/razerkbd/

There should be three directories with your keyboard's VID:PID, e.g.

    0003:1532:0214.0005
    0003:1532:0214.0006
    0003:1532:0214.0007

There are more then one ways in the sysfs (`/sys/`) to get to it, but I'd have thought Fedora's kernel isn't _that_ much different.

> Stopping daemon.

The `SystemExit` exception is harmless, although we might be able to catch that so it doesn't look broken.

> I also tried adding this kernel parameter temporarily on boot by editing in grub (using e) but it made no difference when shutting down right after login. 

Worth a try, thanks. ACPI is complicated, not worth going down that rabbit hole. You could try `acpi=off` just to see if that makes any different. When it comes to [ACPI](https://en.wikipedia.org/wiki/ACPI), it's either the OS or BIOS that's in control of power management.

> Could it be that the daemon shut down is not graceful for me, and the device is left in driver mode...

Not 100% if it's the cause, just wanted to check. You can find out with `cat device_mode` once you locate the file. In the case of my keyboard (Ultimate 2016) I could also try the macro key and if it blinks 4 times like it's saying 🚫 , that would be normal mode.

Another crude way is to try a live ISO - start with a powered off system and don't boot the regular system, so the device is definitely in "normal mode" without any interference or OpenRazer presence.

Matheos96, 2024-12-31:
Thanks for having the patience to debug this with me.

This is what is inside `/sys/bus/hid/drivers/razerkbd`  (so I actually have 4 folders for my Razer Ornata V2)
[image]

Inside the folders (or symlinked folders I believe?) I cannot see the device_mode file. I also could not find it in any subfolders...  
[image]

I will try the new kernel parameter next. I also have a gparted live USB which I could test with. May need to get a fedora live usb to get more apples to apples comparison, but that should not be a problem. Thanks again for the persistant support :)

Matheos96, 2024-12-31:
> You could try acpi=off just to see if that makes any difference.

I tried this and the result was that (probably unsurprisingly) the NVIDIA drivers could not be loaded and it defaulted to noveau. When I tried to perform shutdown though, nothing would shut down. Fedora and Linux kernel (I think) did shut down but the computer was left running (fan spinning) and my monitor was left on (black). The keyboard and mice also were left lit. I hit the power button once and it shut off, but mouse and keyboard still lit.

> Another crude way is to try a live ISO - start with a powered off system and don't boot the regular system, so the device is definitely in "normal mode" without any interference or OpenRazer presence.

I did this too, using a fresh fedora live USB. I first shut down my installed Fedora, booted into Windows and shut down from there (in order to shut down all the lights). I then performed a clean boot from the live USB and performed a shutdown from there. Still same issue (corsair mouse and Razer Ornata V2 keyboard still lit).

It is weird my openrazer driver folder structure seems to be missing the flag... But as you say, this also affects my mouse so it should be something specific to either linux kernel or fedora. I will now test the live USB steps but with gparted live, based on debian (I believe). It is also fairly recently made, so kernel version pretty much the same I think. I will check.

Matheos96, 2024-12-31:
> will now test the live USB steps but with gparted live

Same outcome... :( Neither peripheral's light are turned off, even when shutting down from gparted live. The common denominator between these two, apart from likely large parts of kernel, is GNOME. Not sure if that has anything to do with it...

Another theory is that it has something to do with drivers, but the issue is separate for both keyboard and mouse. I don't have anything special installed for the corsair mouse. Maybe I could look into that and see if that would at least resolve the mouse problem, then we would know that something similar is not happening on the keyboard side.

Matheos96, 2025-01-01:
Happy new year! 

Small addition: I realized that the keyboard and mouse **do** turn off for suspend. So that works at least...

lah7, 2025-01-01:
Happy new year to you too!

The only _workaround_ that I think will be practical is to instruct the Linux kernel to power off USB devices when you want to shut down the system. We can do this with the sysfs filesystem.

I don't know if there's a better way to do this, but here's a prototype of a script to execute at shutdown:

```bash
#!/bin/bash
# Script expected to run as root

# If the keyboard doesn't go off, maybe it needs the OpenRazer driver unloading first?
openrazer-daemon -s                        # or not? root's running this?
systemctl --user stop openrazer-daemon     # might not work either?
#echo -n -e "\x00\x00" > /sys/bus/hid/drivers/razerkbd/**/device_mode     # is it worth doing this?
modprobe -r razerkbd

# Power off every USB devices
for dev in /sys/bus/usb/devices/*
do
    echo 'suspend' | sudo tee "$dev/power/control"
done

# (For testing) Turn on USB devices after 3 seconds
#sleep 3
#for dev in /sys/bus/usb/devices/*
#do
#    echo 'on' | sudo tee "$dev/power/control" # or 'auto'
#done

# Actually do the shutdown
poweroff
```

The missing sysfiles on OpenRazer's side is strange, but there are others like `device_name` and `device_serial` that should exist at minimum, otherwise `openrazer-daemon` wouldn't work. The driver `razerkbd` should be loaded to that device:

    lsmod | grep razer

There might be another way, utility or 'rule' that does this. I hope this is a nudge towards a working solution!

Matheos96, 2025-01-02:
Thanks for the tips. I have tried playing around with this a bit but no luck as of yet.. :/
 
`openrazer-daemon` does run as user so it should not be killed by root I think.. At least `openrazer-daemon -s` without sudo works from CLI, though as explained earlier with the harmless python exception and weird "spamming of enter key" in console (and killing the main enter key).

The turning off of the USB devices does not seem to work at all. `tee` says _Invalid argument_ though I thnk it actually is because we cannot edit the `control` file just like "any file", as it is somehow [special kernel stuff](https://askubuntu.com/a/1048518/2083854)... 

If I could get something like this to work, I would probably add it as a service which runs on shutdown. One thought I had myself earlier was to use the openrazer python API to just turn off the lights before shutting down.. Though that would only affect they keyboard ofc, and I am not sure if that would then persist into reboot... Something like what you sugest would probably be best. May need to get chat gpt involved to try to get the script working..

Matheos96, 2025-01-02:
Finally some progress!

I was playing around, and realised that the `tee` error actually is from the kernel somehow, and is due to the device not accepting `suspend` as a value. `auto` and `on` both work, and I could successfully run the command to switch between those... Either way, that does not help at all.

When turning off the PC, I realised that my gamepad which is also connected via USB **does** turn off (lights) which made me think why it works for that but not for mouse and keyboard. It did not take me long to remember that both my mouse and keyboard are connected via a USB hub/switch to my PC (and to another PC). I use this hub as I work from home and use the same setup for my work computer and my personal, I just toggle between the PCs with a button attached to the USB switch! Connecting the keyboard directly to the PC, everything works as expected. The lights go out on shutdown.

No real solution though as of yet, as I don't want to route my keyboard and mouse directly to my PC, as that defeats the purpose of why I got the USB switch in the first place... Also, interesting how it works on windows but not here.. Could it be that I lack some drivers for the hub/switch itself? It is some generic "made in china" switch so hard to get any manufacturer/model info from the case itself. May need to dig into it using CLI

Matheos96, 2025-01-02:
I found the hub: `ID 05e3:0610 Genesys Logic, Inc. Hub`. I am not sure if I should blame the hub/driver for not propagating the suspend or whatever signal down to the devices connected, or if it is something in the kernel shutdown that should traverse the usb tree..(?)

Device 003 is the hub, my keyboard is device 004 which is now outside the "hub tree" when connected directly. The remaining three devices in the hub tree are a web cam, the mouse and a USB wireless headset receiver.
[image]

closed 2025-01-02 by lah7