openrazer/openrazer #2362: Running openrazer overrides hardware configuration, even though restore_persistence is set to false

steveyh25, 2024-12-22:
### Affected Device / PID

Razer Viper Ultimate

### Operating System / Version

Linux Mint 22

### Kernel Version

6.8.0-50-generic

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Running openrazer-daemon overrides the device's current DPI setting even though restore_persistence is set to false

My mouse has 600DPI saved as a hardware default, and when running the daemon it looks like this is being set to 1000. I then have to click the profile switcher on the bottom of the mouse in order to restore it to a 600DPI profile. Interestingly after starting the application, the light on the bottom of the mouse indicating which profile is selected turns itself off until I press the profile switcher button to restore 600DPI

Everything works fine and as expected when I do not launch the openrazer daemon. I only use the daemon for monitoring battery percentage

z3ntu, 2025-01-02:
This sounds like the device is switched to driver mode instead of device mode during startup, but to my knowledge this should only happen on some keyboards where we're using that mode.

Could you run `cat /sys/bus/hid/drivers/razermouse/0*/device_mode` before and after starting the daemon and seeing if that value changes?

steveyh25, 2025-01-03:
before starting the daemon that command output nothing, and also after starting the daemon that command output nothing

z3ntu, 2025-02-15:
And with `hexdump -C /sys/bus/hid/drivers/razermouse/0*/device_mode` ?

steveyh25, 2025-03-10:
00000000  00 00                                             |..|
00000002

both before and after

z3ntu, 2025-03-15:
Then we're not setting that, and with restore_persistence false we also shouldn't set anything else. Can you provide the log from the daemon when starting to check? `systemctl --user stop openrazer-daemon.service` and then `openrazer-daemon -Fv` should be good for this.

closed 2025-08-14 by github-actions[bot]