openrazer/openrazer #2364: Support for Razer Wolverine v3 Pro

SaltyCartharsis, 2024-12-25:
### Device Name

Razer Wolverine v3 Pro 2.4

### Product Number

RZ06-05200100

### Model Number

RZ06-0520

### VID:PID

1532:0a3f

### USB Details

```shell
Bus 001 Device 005: ID 1532:0a3f Razer USA, Ltd Razer Wolverine V3 Pro 2.4
Couldn't open device, some information will be missing
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 [unknown]
  bDeviceProtocol       208 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a3f Razer Wolverine V3 Pro 2.4
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Wolverine V3 Pro 2.4
  iSerial                 3 0000000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
total 0
drwxr-xr-x 2 root root 420 Dec 25 18:18 .
drwxr-xr-x 4 root root 900 Dec 25 18:18 ..
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-1532_Razer_Hammerhead_HyperSpeed_2.4-event-if03 -> ../event6
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-3Dconnexion_SpaceNavigator-event-if00 -> ../event13
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-HP__Inc_HyperX_Cloud_Alpha_S_000000000001-event-if05 -> ../event12
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-KBDCRAFT_07_Israfel_vial:f64c2b3c-event-if02 -> ../event17
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-KBDCRAFT_07_Israfel_vial:f64c2b3c-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-KBDCRAFT_07_Israfel_vial:f64c2b3c-if02-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-Nexight_INC_NexiGo_HelloCam_Composite_Device_000000000005-event-if02 -> ../event14
lrwxrwxrwx 1 root root  10 Dec 25 18:17 usb-Nexight_INC_NexiGo_HelloCam_N930W_Camera-event-if00 -> ../event31
lrwxrwxrwx 1 root root  11 Dec 25 18:17 usb-Nexight_INC_NexiGo_HelloCam_N930W_Camera-event-if02 -> ../event259
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-PloopyCo_Trackball-event-if02 -> ../event5
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-PloopyCo_Trackball-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-PloopyCo_Trackball-if02-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-PloopyCo_Trackball-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  11 Dec 25 18:18 usb-Razer_Razer_Wolverine_V3_Pro_2.4_0000000000000001-event-joystick -> ../event260
lrwxrwxrwx 1 root root   6 Dec 25 18:18 usb-Razer_Razer_Wolverine_V3_Pro_2.4_0000000000000001-joystick -> ../js0
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-Vaydeer_Multimedia_Knob_VAJP220130904383C4-event-if03 -> ../event9
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-Vaydeer_Multimedia_Knob_VAJP220130904383C4-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-Vaydeer_Multimedia_Knob_VAJP220130904383C4-if03-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Dec 25 18:17 usb-Vaydeer_Multimedia_Knob_VAJP220130904383C4-if03-mouse -> ../mouse1
```

### USB HID Dump

```shell
001:004:003:DESCRIPTOR         1735169854.637950
 06 00 FF 09 01 A1 01 85 02 15 00 26 FF 00 09 02
 75 08 95 3F 81 02 15 00 26 FF 00 09 03 75 08 95
 3F 91 02 C0 05 0C 09 01 A1 01 85 01 15 00 25 01
 09 CD 09 B5 09 B6 09 B7 75 01 95 04 81 02 15 00
 25 01 09 B0 09 B1 09 B3 09 B4 75 01 95 04 81 22
 15 00 25 01 09 E9 09 EA 09 E2 75 01 95 03 81 22
 75 05 95 01 81 01 C0
```

z3ntu, 2025-01-02:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

wesharper, 2025-01-26:
Built-in Linux support would be excellent, will look at the development link!

wesharper, 2025-01-26:
It seems like there are no controllers currently configured in the whole repo

z3ntu, 2025-02-15:
Depending what functionality Synapse can control on this device, it might fit well within the razeraccessory driver but without any extra info from you about what this device can do I can't suggest much.

wesharper, 2025-02-18:
Sorry, I'm a bit new to the ecosystem so I'm still getting my bearings. From what I've been able to gather, it seems like most of the support for Razer controllers is handled by the `xpad` kernel driver. I think that's likely a better option for the OP, as many of the problems are already solved there, and should be as straightforward as adding the vendor and product IDs to a specific xbox gamepad type mapping.

z3ntu, 2025-02-18:
Right, this device might need to be added to xpad.c like #2150  https://lore.kernel.org/linux-input/20231125-razer-wolverine-v2-v1-1-979fe9f9288e@z3ntu.xyz/

Can you applying this patch to your kernel and see if it improves anything?

```diff
diff --git a/drivers/input/joystick/xpad.c b/drivers/input/joystick/xpad.c
index 8fe2a51df649..575b1d2c9301 100644
--- a/drivers/input/joystick/xpad.c
+++ b/drivers/input/joystick/xpad.c
@@ -297,6 +297,7 @@ static const struct xpad_device {
 	{ 0x1532, 0x0a00, "Razer Atrox Arcade Stick", MAP_TRIGGERS_TO_BUTTONS, XTYPE_XBOXONE },
 	{ 0x1532, 0x0a03, "Razer Wildcat", 0, XTYPE_XBOXONE },
 	{ 0x1532, 0x0a29, "Razer Wolverine V2", 0, XTYPE_XBOXONE },
+	{ 0x1532, 0x0a3f, "Razer Wolverine V3 Pro", 0, XTYPE_XBOXONE },
 	{ 0x15e4, 0x3f00, "Power A Mini Pro Elite", 0, XTYPE_XBOX360 },
 	{ 0x15e4, 0x3f0a, "Xbox Airflo wired controller", 0, XTYPE_XBOX360 },
 	{ 0x15e4, 0x3f10, "Batarang Xbox 360 controller", 0, XTYPE_XBOX360 },

```

wesharper, 2025-02-19:
I've tried and had limited success, but I've also gone into the deep end as someone who's new to Linux by choosing NixOS, so I am still figuring out how to apply kernel patches properly and verify if my patches are being correctly applied.

Unfortunately, I'm not savvy enough yet to translate the pretty straightforward `dkms` patch approach to NixOS yet. Would appreciate someone verifying independently as I get my bearings.

wesharper, 2025-02-20:
I added a patch in my own kernel using the following:

```
diff --git a/drivers/input/joystick/xpad.c b/drivers/input/joystick/xpad.c
index afacca8..0291bce 100644
--- a/drivers/input/joystick/xpad.c
+++ b/drivers/input/joystick/xpad.c
@@ -325,6 +325,8 @@ static const struct xpad_device {
 	{ 0x1532, 0x0a00, "Razer Atrox Arcade Stick", MAP_TRIGGERS_TO_BUTTONS, XTYPE_XBOXONE },
 	{ 0x1532, 0x0a03, "Razer Wildcat", 0, XTYPE_XBOXONE },
 	{ 0x1532, 0x0a29, "Razer Wolverine V2", 0, XTYPE_XBOXONE },
+	{ 0x1532, 0x0a43, "Razer Wolverine V3 Tournament Edition", 0, XTYPE_XBOXONE },
+	{ 0x1532, 0x0a3f, "Razer Wolverine V3 Pro", 0, XTYPE_XBOXONE },
 	{ 0x15e4, 0x3f00, "Power A Mini Pro Elite", 0, XTYPE_XBOX360 },
 	{ 0x15e4, 0x3f0a, "Xbox Airflo wired controller", 0, XTYPE_XBOX360 },
 	{ 0x15e4, 0x3f10, "Batarang Xbox 360 controller", 0, XTYPE_XBOX360 },
```

The problem is, something in my kernel was already allowing both controllers to work with the exact same functionality pre-patch, so it's pretty hard to know if this would meaningfully resolve anything for others. The only thing that continues to not work is rumble support for the v3 pro wireless over dongle. I'll try `xpadneo` and see if that resolves the rumble support issue.

z3ntu, 2025-02-20:
So you're saying the controller already worked (apart from rumble) without any extra modifications?

wesharper, 2025-02-25:
It did on my machine, although it may have been due to something that Steam installed. I'll see if I can track down what steam may be doing.

prometheus-ex, 2025-09-19:
I haven't seen any movement on this, so I thought I'd mention that the Tournament Edition of the controller works perfectly fine as a standard Xbox layout. However, the additional buttons (2 trigger, 4 grip) are not functional.
Oddly, they _are_ detected in AntimicroX.

ZlordHUN, 2025-10-18:
Doesnt work on Pop OS 24. Support would be nice. Anything I can add to contribute to the implementation?