openrazer/openrazer #2367: How can I disable the remapping of the M1-M5 keys?

soleofthesea, 2024-12-29:
### Affected Device / PID

1532:028d Razer USA, Ltd Razer BlackWidow V4 Pro

### Operating System / Version

Arch Linux

### Kernel Version

6.12.4-arch1-1

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [X] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

Razer Synapse 4 now exposes the ability to rebind the macro keys and save them onto device memory, so it is no longer necessary to have the daemon running to use them.

I changed them to Ctrl+Alt+Shift+Win+1..5 so I can free up the F keys for something else but I don't see a way to tell Openrazer to stop converting the macros.

z3ntu, 2025-01-02:
So the mappings which you saved in memory start working again when you stop the daemon? We set the device to 'driver mode' (from default 'device mode') on daemon startup to handle some extra keys, but it's not user configurable to not do that.

I wonder now how Synapse gets those working in driver mode (which afaik they always use).

soleofthesea, 2025-01-04:
Actually, upon further experimentation I'd been passing the device into a Windows VM to use Synapse and it seems it's just that the daemon doesn't grab the device when detaching from QEMU. Unplugging and replugging still remaps the keys to the F keys.