openrazer/openrazer #2368: Support for Razer Hanbo Chroma AIO

firstkindgamer, 2024-12-31:
### Device Name

Razer Hanbo Chroma

### Product Number

RC21-01770100

### Model Number

RC21-017701

### VID:PID

1532:0f35

### USB Details

```shell
Bus 001 Device 008: ID 1532:0f35 Razer USA, Ltd Razer Hanbo AIO ARGB
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f35 Razer Hanbo AIO ARGB
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Hanbo AIO ARGB
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Device Input Paths

```shell
N/A
```

z3ntu, 2025-01-02:
Thanks for opening the issue! You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

dripsnek, 2025-04-16:
I've been working on adding Razer Hanbo support for [Liquidctl](https://github.com/liquidctl/liquidctl/pull/783), [Liquidtux](https://github.com/liquidctl/liquidtux/pull/73) and [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/2811).

I haven't looked at what it would take to adapt the Liquidtux driver to support OpenRazer, but what was apparent in writing the above was that the Hanbo runs a substantially different protocol to other Razer devices and may need to be treated as its own thing.