openrazer/openrazer #2376: Keyboard suddenly not lighting up

ZolaKluke, 2025-01-06:
### Affected Device / PID

Razer Blackwidow V4, 1532:0287

### Operating System / Version

Fedora 40 KDE

### Kernel Version

Linux 6.12.7-100.fc40.x86_64

### OpenRazer Version

3.9.0

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [X] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [X] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I'm unsure exactly when this problem began as it's been going on for 1-2 weeks but it may have been after a newer kernel however I have re-built the kernel modules for my current kernel and the problem is still persisting, also the lights for the numlock, caps lock, etc. also don't light up either

ZolaKluke, 2025-01-06:
Note: My mouse which is a Razer Deathadder V2 Pro (Wired or Wireless) does not have any issues with the lighting

lah7, 2025-01-06:
Other users reported issues with this keyboard. Although, there is no known solution besides re-plugging it in or [waiting  for other system components to be updated](https://github.com/openrazer/openrazer/issues/2291#issuecomment-2369910053) that might fix it. 

* https://github.com/openrazer/openrazer/issues/2291

Just for reference, what firmware version is your keyboard?

ZolaKluke, 2025-01-06:
Strange the issue only appeared in the last few weeks, it's been working fine for 3-4 months leading up to this

ZolaKluke, 2025-01-06:
Hmm checking the original post it seems to be possibly dualboot related yet I haven't booted into Windows since 2 months ago and it was still working fine then

8hodgsonkh, 2026-10-22:
sorry for necro but if anyone else comes accross this issue, enabling xhci hand off in bios fixes it