openrazer/openrazer #2377: Kubuntu 24.04 crashes upon input from Blackwidow V4 Pro Macro keys

danieljburt, 2025-01-06:
### Affected Device / PID

Razer Blackwidow V4 Pro, 1532:028D

### Operating System / Version

Kubuntu, 24.04

### Kernel Version

6.8.0-51-generic

### OpenRazer Version

3.9.0-1

### Support Checklist

- [X] My user was added to the plugdev group.
- [X] I have rebooted and the problem persists.
- [X] I looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page and it did not solve the problem.

### Symptoms

- [ ] Uninstalling OpenRazer fixes the problem
- [ ] Re-plugging the device fixes the problem
- [ ] It worked before a system/kernel update.
- [ ] Only a problem if the daemon is running.
- [ ] Only a problem from power up or a reboot.
- [ ] I'm dual booting with Windows.
- [ ] Intermittent or erratic problem.
- [ ] My device is missing a feature.

### Please describe the issue

I am using a KVM switch-docking station to allow the use of a Blackwidow V4 Pro between my gaming PC (Windows) and work laptop (Linux). 

When the keys on the left side of the Blackwidow V4 Pro keyboard or the unassigned M[1-5] macro keys are pressed, the Linux laptop crashes. There is no error report, the laptop freezes completely and accepts no further input even after disconnecting from the KVM switch-docking station.

The issue is repeatable and seems to be independent of whether Openrazer + Polychromatic are installed or not. 
I followed the suggestion on the Troubleshooting page and added the daily PPA which appeared to have no effect. 

Additionally, the audio scroll wheel and buttons on the right of the keyboard function correctly but the scroll "knob" on the top left does not operate in Linux. 

Is there some way to disable input from these keys using Openrazer/ Polychromatic or one of the other apps to at least prevent the crashes? Or map the macros to a harmless function that is recognised by my Linux system?

danieljburt, 2025-01-07:
Disabling "razerkbd" appears to prevent the system crashes when receiving input from the left-hand side macro keys. The brightness knob now changes the brightness of the keyboard LEDs (but does not change function when pressed but also doesn't cause a crash anymore either) and the Media macro keys and scrollbar on the top right of the keyboard still function correctly. 

Polychromatic is obviously no longer functional though.

danieljburt, 2025-01-20:
The issue persists when connecting the keyboard directly to the laptop, bypassing the KVM switch, yes.

When I initially experienced the issue, there was no configuration, I have since attempted to configure the keys as spacebar macros and disabling them entirely using the Razer officials apps in Windows with no impact on the issue reported above for Linux. 

Disabling "razerkbd" is the only way to avoid the crashes currently. As of writing this message, however, disabling "razerkbd" now also disables the correct functioning of the Media macro keys entirely, unlike my comment 2 weeks ago, where they continued to function.

rominail, 2025-01-24:
I have the same problem, it also freezes for me with one of the side button (the closest to the space bar)

rominail, 2025-02-13:
Thanks
Those 3 commands ran successfully
However I don't get what I'm supposed to do after make to run the daemon, I don't have it in the available services with systemctl
I removed the existing install I had
I tried `PYTHONPATH="pylib:daemon" python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf` and got `Unable to lock on the pidfile.`

This is the logs when I run make : 
``` [68396.943463] usbhid 3-3.2:1.4: Razer Device disconnected
[68396.943544] usbhid 3-3.2:1.3: Razer Device disconnected
[68396.970090] usbhid 3-3.2:1.2: Razer Device disconnected
[68397.048254] usbhid 3-3.2:1.1: Razer Device disconnected
[68397.122139] usbhid 3-3.2:1.0: Razer Device disconnected
[68397.122462] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.0/0003:1532:028D.0006/input/input34
[68397.175654] hid-generic 0003:1532:028D.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input0
[68397.176822] input: Razer Razer BlackWidow V4 Pro Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.1/0003:1532:028D.0007/input/input35
[68397.229571] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.1/0003:1532:028D.0007/input/input36
[68397.229857] hid-generic 0003:1532:028D.0007: input,hidraw6: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input1
[68397.230269] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:1532:028D.0008/input/input37
[68397.234867] hid-generic 0003:1532:028D.0008: input,hidraw7: USB HID v1.00 Mouse [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input2
[68397.235815] hid-generic 0003:1532:028D.0009: No inputs registered, leaving
[68397.238550] hid-generic 0003:1532:028D.0009: hidraw8: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input3
[68397.243627] hid-generic 0003:1532:028D.000A: hiddev0,hidraw9: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input4
[68397.346326] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.0/0003:1532:028D.0006/input/input38
[68397.399230] razerkbd 0003:1532:028D.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input0
[68397.509834] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.1/0003:1532:028D.0007/input/input39
[68397.561892] razerkbd 0003:1532:028D.0007: input,hidraw6: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input1
[68397.584445] input: Razer Razer BlackWidow V4 Pro as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:1532:028D.0008/input/input40
[68397.636419] razerkbd 0003:1532:028D.0008: input,hidraw7: USB HID v1.00 Mouse [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input2
[68397.636730] razerkbd 0003:1532:028D.0009: No inputs registered, leaving
[68397.636849] razerkbd 0003:1532:028D.0009: hidraw8: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input3
[68397.641289] razerkbd 0003:1532:028D.000A: hiddev0,hidraw9: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-3.2/input4```

Jacajack, 2025-02-13:
> I removed the existing install I had

Ah, I assumed you could try to use the previously installed daemon with the downgraded daemon, but okay. The pidfile error suggests that perhaps another daemon is running in the background. Can you make sure that no other openrazer daemon process is running? You can check the output of `ps aux | grep -i razer`.

If that doesn't help, could you please post the full output of the daemon start command? It should look somewhat like this:
```
Starting daemon.
2025-02-13 20:05:24 | razer                          | INFO     | Initialising Daemon (v3.6.1). Pid: 14410
2025-02-13 20:05:24 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2025-02-13 20:05:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2025-02-13 20:05:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2025-02-13 20:05:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
.....
```

rominail, 2025-02-14:
I can't get it to run properly
I reinstalled the daemon, tried all the commands and force the make with `-B` flag to bypass
```warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0```
I restarted the daemon
I tried installing `python3-openrazer` which was already installed
I did `sudo gpasswd -a $USER plugdev`
This is the troubleshooting from polychromatic

![Image](https://github.com/user-attachments/assets/5b612155-8c2c-4f8a-8cbf-ee634215e319)

Jacajack, 2025-02-14:
> But my laptop doesn't crash if I press the macros (I don't know if the drivers are loaded correctly or not though)

That's promising. You can verify if the driver is loaded and they keyboard is switched into driver mode by testing what keystrokes are emitted by the macro keys. You can use `xev -event keyboard` to see the keycodes. If I remember correctly, the macro keys should appear as F13-F24.

The nice way to ensure proper permissions for USB devices is via udev. Openrazer provides udev files to take care of that, but you don't have the package installed. You could manually install the `99-razer.rules` file and the permissions should be taken care of. I suppose you could try running `sudo make udev_install`:

https://github.com/openrazer/openrazer/blob/5e677b178be7da08fb72d7cd0791bf68a129fcc5/Makefile#L108-L112

> I did `sudo gpasswd -a $USER plugdev`

Just a note,  but this command will only take effect once you log into your account again.

> I can start the daemon. In polychomatic I still have the message "No backends loaded"

I've never used polychromatic, but I'm gonna guess that it can't find the Openrazer Python library to communicate with the daemon. Can you try running it similarly to the daemon, with the `PYTHONPATH` variable set? This way `pylib` dir will be added to the Python library search paths and Polychromatic should be able to fetch it.

```bash
# From the openrazer directory!
PYTHONPATH="pylib:daemon" polychromatic
```

markmandel, 2025-02-25:
Just wanted to say thank you for finding this! Was wondering what was going on!

markmandel, 2025-02-25:
From my digging, it looks like this is a kernel panic since my Lenovo Caps lock flashes as soon as it happens, and I can't even SSH into the box.

Last time I tried to get the Kernel logs I couldn't find anything, but with a halted kernel that can be tough.

z3ntu, 2025-02-26:
Can you not run `sudo dmesg -w` in a terminal before you 'trigger' the panic, and then see it on screen?

markmandel, 2025-02-26:
> Can you not run `sudo dmesg -w` in a terminal before you 'trigger' the panic, and then see it on screen?

Lemme give it a shot!

markmandel, 2025-02-27:
[image]

Unfortunately, I hit the button and everything immediately froze. There was no output on screen.

This is on:
* Debian Testing/Trixie
* Hyprland / Wayland DE
* open-razer 3.10.0

Maybe if I did it in a tty?

markmandel, 2025-02-27:
Got the logs in a TTY after a few attempts 🙌🏻 ! That was tough one to capture, because it often was just a message at the top of the screen.

[image]

For the life of me, I cannot find this text in `journalctl` though.

z3ntu, 2025-02-27:
It is saying `razer_raw_event` in the visible text though so unfortunately you're right and the openrazer code has a problem there

markmandel, 2025-02-27:
Anything else I can do to help narrow down what's going on?

rominail, 2025-03-03:
I was away for a while, anything I can do or was the problem identified?

markmandel, 2025-03-10:
Gently bumping this. I'm _pretty_ good at just not hitting buttons on my Blackwidow, but I'm not _perfect_. Would love to not have my keyboard cause a kernel panic.

Anyone got any decent workaround in the meantime?

Jacajack, 2025-03-10:
@markmandel I have a little suggestion, not sure if it will be very helpful or not. Could you please post output of `addr2line -e /path/to/razerkbd.ko -i 'razer_raw_event+0x34d/0x5b0'`? This could help to determine where in the code the fatal exception comes from.

markmandel, 2025-03-10:
Unfortunately, doesn't look like this was super useful. Not familiar with this tooling, so happy to take suggestions/suggestions on what I should be doing differently.

```
/tmp/test on ☁️   on ☸ agones
❯ cp /var/lib/dkms/openrazer-driver/3.10.0/6.12.17-amd64/x86_64/module/razerkbd.ko.xz .

/tmp/test on ☁️   on ☸ agones
❯ 7z x razerkbd.ko.xz

7-Zip 24.09 (x64) : Copyright (c) 1999-2024 Igor Pavlov : 2024-11-29
 64-bit locale=en_AU.UTF-8 Threads:20 OPEN_MAX:1024, ASM

Scanning the drive for archives:
1 file, 38172 bytes (38 KiB)

Extracting archive: razerkbd.ko.xz
--
Path = razerkbd.ko.xz
Type = xz
Physical Size = 38172
Method = LZMA2:20 CRC32
Streams = 1
Blocks = 1
Characteristics = BlockPackSize BlockUnpackSize

Everything is Ok

Size:       276917
Compressed: 38172

/tmp/test on ☁️   on ☸ agones
❯ l
total 312K
drwxrwxr-x  2 mark mark   80 Mar 10 11:54 .
drwxrwxrwt 23 root root  560 Mar 10 11:54 ..
-rw-rw-r--  1 mark mark 271K Mar 10 11:54 razerkbd.ko
-rw-r--r--  1 mark mark  38K Mar 10 11:54 razerkbd.ko.xz

/tmp/test on ☁️   on ☸ agones
❯ addr2line -e ./razerkbd.ko -i 'razer_raw_event+0x34d/0x5b0'
??:?
```

markmandel, 2025-03-10:
If I got the logs from the previous release, would that affect attempting to get the line number? (I.e should I try getting the crash details with the current version?)

I can't remember which version it was, and when the latest version landed on Debian Trixie.

Jacajack, 2025-03-10:
Yeah, that doesn't look too helpful.

> If I got the logs from the previous release, would that affect attempting to get the line number?

That is entirely possible. Ideally, you could try to attempt the following;
1. Build the kernel module yourself (`make driver`)
2. Unload the kernel module provided by Debian with `sudo rmmod razerkbd`
3. Load the self-compiled module with `sudo insmod driver/razerkbd.ko`
4. Try to reproduce the crash and take a picture of the log again
5. Then try `addr2line` again with the new address and the self-compiled module

This is my best idea at the moment. My experience with kernel development is quite limited, so maybe somebody better qualified will suggest a better approach.

z3ntu, 2025-03-15:
You could also just sprinkle some `printk(KERN_ERR "DBG %s:%d\n", __func__, __LINE__);` in various lines and see which line is printed last. Would be quite helpful to be able to narrow it down!

markmandel, 2025-04-06:
> 1. Build the kernel module yourself (`make driver`)
> 2. Unload the kernel module provided by Debian with `sudo rmmod razerkbd`
> 3. Load the self-compiled module with `sudo insmod driver/razerkbd.ko` [...]

Thanks for the instructions. Managed to at least get this setup! Was very easy to get going.

I did notice something interesting. If I `rmmod razerkbd` and then run `sudo insmod driver/razerkbd.ko`, I don't get the panic until I unplug and reconnect the keyboard. This might be something to do with how the kernel handles loading new hardware with custom drivers? Not sure.

Also, if anyone has suggestions on a reliable way to funnel kernel panics to a file, to make it far easy to actually see, that would be ace. `sudo dmesg --follow` seems useless here.

markmandel, 2025-04-13:
Eeeeyyyyyyyyy!

I finally got it done! Panic with the custom local build.

[image]

```
mark in 🌐 games-oss in openrazer on  master on ☁️  agones-mark-dev on ☸ agones
❯ addr2line -e ./driver/razerkbd.ko -i '0010:razer_raw_event+0x34d/0x5b0'
/home/mark/workspace/openrazer/driver/razerkbd_driver.c:2761

mark in 🌐 games-oss in openrazer on  master on ☁️  agones-mark-dev on ☸ agones
❯ addr2line -e ./driver/razerkbd.ko -i 'razer_raw_event+0x34d/0x5b0'
/usr/src/linux-headers-6.12.21-common/arch/x86/include/asm/bitops.h:60
/usr/src/linux-headers-6.12.21-common/include/asm-generic/bitops/instrumented-atomic.h:29
/home/mark/workspace/openrazer/driver/razerkbd_driver.c:3696
/home/mark/workspace/openrazer/driver/razerkbd_driver.c:3786
```

I wasn't sure which one was better - so here we go!

This was when hitting the macro M1 key.

markmandel, 2025-04-14:
PR is in! Seems to fix the problem 🤞🏻

lmf-mx, 2025-04-17:
Same behavior with 1532:0293 Blackwidow v4 X.
Thanks for this. Can't wait to see this get merged in and packaged.

markmandel, 2025-04-17:
If you wish to test (on Debian at least), I just dropped some deb files on the PR: https://github.com/openrazer/openrazer/pull/2439#issuecomment-2813618591

lmf-mx, 2025-04-17:
@markmandel Installed the package from #2439, no issues. Everything behaves properly and no issue with Polychromatic using the daemon.

z3ntu, 2025-04-19:
Thanks for working on this!

z3ntu, 2025-04-24:
I'm thinking this change might also fix your issue, could someone try please? #2452

markmandel, 2025-04-29:
Debs are now listed on https://github.com/openrazer/openrazer/pull/2452#issuecomment-2840029815 of anyone wants to test, otherwise I'll try and get to it as soon as I can 👍🏻

z3ntu, 2025-05-06:
Fixed by #2452 - hopefully!

closed 2025-05-06 by z3ntu